xoseperez/espurna #1630: Wiijoo Smart Table Lamp

filipemurias, 2019-03-11:
Hello!

Does anyone know if the Wiijoo Smart Table Lamp is compatible with Espurna?

Examples:
https://www.youtube.com/watch?v=S1pRHTL5D-E
https://www.amazon.com/Wiijoo-Children-Compatible-Controlled-Smartphone/product-reviews/B07BLPC8QL
https://www.aliexpress.com/item/WiFi-Smart-Table-Lamp-Compatible-with-Alexa-Google-Home-Voice-Control-Hand-Touch-Light-LED-Ambient/32911627149.html?spm=a2g0s.9042311.0.0.3da24c4dQV9bms

I tested several Espurna binaries (espurna-1.13.5-iwoole-led-table-lamp.bin and some arilux), but nothing happens when I change colors. Not even ON/OFF works.

Thank you!

Filipe Múrias

ColinShorts, 2019-03-11:
I quite liked these devices but ended up settling for the iWoole's and 3 or 4 attempts to figure out the correct pins.

Maybe it would be worth creating a generic ROM with led's on every usable gpio pin so that we can more easily and quickly identify the correct pins for new devices? I guess the same could be done for relays/switches/buttons too.

However, it's possible that there might be a more complicated setup - like the ai thinker with a separate chip to drive the channels, in which case you'll likely need to open it up to figure it out.

filipemurias, 2019-03-12:
Thank you, @ColinShorts !

I opened the lamp and there is in deed a STC8F2K16S2 chip (on the left side).

[image]

Do you think that there is a way to make this work with Espurna? Is there any lamp similar to this one?

Filipe Múrias

ColinShorts, 2019-03-12:
All things are possible, but I'm way out of my depth for this 🙂.

I'd imagine (and could be way off the mark) that the esp will be talking to the STC chip using serial, so unless you have a copy of the original firmware (reflash and snoop on the serial lines) it's likely going to be pretty difficult to figure out.

Unless there is a similar chip/light already supported by espurna, or another firmware that's documented the work?

closed 2019-05-18 by stale[bot]