xoseperez/espurna #1631: Access point not appear on Wifi list

chryss44, 2019-03-11:
*Before creating a new issue please check that you have:*

* *searched the existing [issues](https://github.com/xoseperez/espurna/issues) (both open and closed)*
* *searched the [wiki](https://github.com/xoseperez/espurna/wiki)*
* *asked for help in the [chat](https://gitter.im/tinkerman-cat/espurna)*
* *done the previous things again :)*

*Fulfilling this template will help developers and contributors to address the issue. Try to be as specific and extensive as possible. If the information provided is not enough the issue will likely be closed.*

*You can now remove this line and the above ones. Text in italic is meant to be replaced by your own words. If any of the sections below are not relevant to the issue (for instance, the screenshots) then you can delete them.*

**Bug description**
*A clear and concise description of what the bug is.*

**Steps to reproduce**
*Steps to reproduce the behavior.*

**Expected behavior**
*A clear and concise description of what you expected to happen.*

**Screenshots**
*If applicable, add screenshots to help explain your problem.*

**Device information**
*Copy-paste here the information as it is outputted by the device. You can get this information by typing `info` via serial, terminal or in the debug tab in the web UI. The relevant information is that surrounded by the scissors-cut lines (`---8<-------`).*
*If you cannot get this info from the device, please answer this questions:*
* *Arduino Core version*
* *ESPurna version*
* *Flash mode*
* *Device brand, model and version*

**Tools used**
* *Desktop operating system*
* *Browser & version*
* *IDE & version*
* *Compiler & version (if not embedded in IDE)*

**Additional context**
*Add any other context about the problem here.*

chryss44, 2019-03-11:
I have try to compile 1.13.3 and program it on Sonoff Basic with success (esptool).
I have just modified Hardware on Sonoff Basic to put another button.
When i reboot, everything is perfect on serial log except "[MDNS] Fail" and AP doesn't appear on Wifi list :-(
Boot log 
`[000161] [MAIN] ESPURNA 1.13.3
[000161] [MAIN] xose.perez@gmail.com
[000161] [MAIN] http://tinkerman.cat

[000162] [MAIN] CPU chip ID: 0x8B0103
[000164] [MAIN] CPU frequency: 80 MHz
[000167] [MAIN] SDK version: 3.0.0-dev(c0f7b44)
[000171] [MAIN] Core version: 2.5.0
[000174] [MAIN] Core revision: 2501570554
[000178] 
[000179] [MAIN] Flash chip ID: 0x14405E
[000182] [MAIN] Flash speed: 40000000 Hz
[000186] [MAIN] Flash mode: DOUT
[000189] 
[000190] [MAIN] Flash size (CHIP)   :  1048576 bytes /  256 sectors (   0 to  255)
[000197] [MAIN] Flash size (SDK)    :  1048576 bytes /  256 sectors (   0 to  255)
[000204] [MAIN] Reserved            :     4096 bytes /    1 sectors (   0 to    0)
[000211] [MAIN] Firmware size       :   528048 bytes /  129 sectors (   1 to  129)
[000218] [MAIN] Max OTA size        :   495616 bytes /  121 sectors ( 130 to  250)
[000226] [MAIN] EEPROM size         :     4096 bytes /    1 sectors ( 251 to  251)
[000233] [MAIN] Reserved            :    16384 bytes /    4 sectors ( 252 to  255)
[000240] 
[000241] [MAIN] EEPROM sectors: 251, 250
[000244] [MAIN] EEPROM current: 251
[000247] 
[000248] [MAIN] EEPROM:  4096 bytes initially |   180 bytes used ( 4) |  3916 bytes free (95)
[000258] [MAIN] Heap  : 43088 bytes initially |  5552 bytes used (12) | 37536 bytes free (87)
[000265] [MAIN] Stack :  4096 bytes initially |  1124 bytes used (27) |  2972 bytes free (72)
[000273] 
[000274] [MAIN] Boot version: 31
[000277] [MAIN] Boot mode: 1
[000279] [MAIN] Last reset reason: Power on
[000283] [MAIN] Last reset info: flag: 0
[000286] 
[000288] [MAIN] Board: ITEAD_SONOFF_BASIC_2
[000291] [MAIN] Support: ALEXA API BROKER BUTTON DEBUG_SERIAL DEBUG_TELNET DEBUG_WEB DOMOTICZ HOMEASSISTANT LED MDNS_SERVER MQTT NTP SCHEDULER TELNET TERMINAL THINGSPEAK WEB 
[000306] [MAIN] WebUI image: SMALL
[000309] 
[000549] [MAIN] Firmware MD5: f4258c5dd6b170dfc84ba547f22f8a4b
[000550] [MAIN] Power: 4466 mV
[000551] [MAIN] Power saving delay value: 10 ms
[000551] 

---8<-------

[000563] [TELNET] Listening on port 23
[000565] [WEBSERVER] Webserver running on port 80
[000569] [RELAY] Retrieving mask: 255
[000570] [RELAY] Relay #0 boot mode 0
[000571] [RELAY] Relay #1 boot mode 0
[000572] [RELAY] #0 set to OFF
[000574] [RELAY] #1 set to OFF
[000578] [RELAY] Number of relays: 2
[000580] [BUTTON] Number of buttons: 2
[000583] [LED] Number of leds: 1
[000586] [MQTT] Async ENABLED, SSL DISABLED, Autoconnect ENABLED
[000600] [THINGSPEAK] Async ENABLED, SSL DISABLED
[000614] [MAIN] Uptime: 0 seconds
[000615] [MAIN] Heap  : 43088 bytes initially | 11936 bytes used (27) | 31152 bytes free (72)
[000617] [MAIN] Power: 4466 mV
[000638] [WIFI] Creating access point
[000700] [WIFI] Captive portal enabled
[000702] [WIFI] -------------------------------------- MODE AP
[000703] [WIFI] SSID  ESPURNA-8B0103
[000703] [WIFI] PASS  fibonacci
[000705] [WIFI] IP    192.168.4.1
[... 138 lines cut ...]
[000543] [TELNET] Listening on port 23
[000545] [WEBSERVER] Webserver running on port 80
[000551] [RELAY] Retrieving mask: 255
[000552] [RELAY] Relay #0 boot mode 0
[000553] [RELAY] Relay #1 boot mode 0
[000554] [RELAY] #0 set to OFF
[000555] [RELAY] #1 set to OFF
[000558] [RELAY] Number of relays: 2
[000560] [BUTTON] Number of buttons: 2
[000564] [LED] Number of leds: 1
[000566] [MQTT] Async ENABLED, SSL DISABLED, Autoconnect ENABLED
[000582] [THINGSPEAK] Async ENABLED, SSL DISABLED
[000634] [WIFI] Creating access point
[000697] [WIFI] Captive portal enabled
[000698] [WIFI] -------------------------------------- MODE AP
[000699] [WIFI] SSID  ESPURNA-8B0103
[000700] [WIFI] PASS  fibonacci
[000702] [WIFI] IP    192.168.4.1
[000705] [WIFI] MAC   6A:C6:3A:8B:01:03
[000708] [WIFI] ----------------------------------------------
[000714] [MDNS] FAIL
[001557] [RELAY] Setting relay mask: 0`

I have try with 1.13.2 and CORE 2.4.2...same issue.
Serious problem on news librairies, no ?

chryss44, 2019-03-12:
I have try this evening and with 2.3 Core everything is OK on 1.13.2, but no access point appear with 1.13.4 and 1.13.6.
To bex

mcspr, 2019-03-13:
Do you mean that you cannot connect to device's access point?
See #1574 , softAP does not work on Basic when running 1.13.4. I have no problems running 1.13.5 or current dev branch

chryss44, 2019-03-13:
Access point not appear on wifi list (nothing) with Core 2.5 and ESPurna 1.13.5 and DEV 1.13.6.
Sometimes, they appears and connection's impossible.
If i have time this week, i can try with another Core, what test do you want i try ?

bakman2, 2019-03-17:
@mcspr I have the same issues with 1.13.5 and a Sonoff Basic. The thing is not connecting after OTA update, can't easily test the actual problem, have to remove it from my fan first.

chryss44, 2019-03-17:
I have compiled 1.13.2 with CORE 2.3 on 1.8.5 Arduino and everything is OK.
I'll try CORE 2.4 this week if i have time.

mcspr, 2019-03-17:
Have you tried changing loop delay value?
One Basic not like the other... I have tried upgrading one of the old "ceiling" devices - it too gets stuck connecting to AP / softAP works 10% of the time. It is very hard to connect to, but it is possible to start a telnet session and do a `set loopDelay 10` (make sure it says `+OK`)

bakman2, 2019-03-18:
@mcspr very strange behaviour. I setup a node-red flow to notify me when it is connected to my network, that happened only once an hour. telnet was disabled, quickly enabled it and set the loopdelay and downgraded to 1.13.3 which works fine. thanks!

zukoy1, 2019-03-22:
Hi all,

Same bahviour with a sonoff basic, I cannot connect to the AP.

Release 1.13.6 downloaded today.

[000830] [WIFI] Scanning
[001662] [RELAY] Setting relay mask: 0
[002966] [WIFI] No networks found
[002967] [WIFI] Creating access point
[003055] [WIFI] Captive portal enabled
[003055] [WIFI] -------------------------------------- MODE AP
[003056] [WIFI] SSID  ESPURNA-79F1A1
[003056] [WIFI] PASS  fibonacci
[003059] [WIFI] IP    192.168.4.1
[003062] [WIFI] MAC   5E:CF:7F:79:F1:A1
[003065] [WIFI] ----------------------------------------------
[003071] [MDNS] FAIL
[060001] [MAIN] System OK

Have you any workaround?

Thanks!

chryss44, 2019-03-23:
I have try with Arduino 1.8.8 and 1.8.5, library up to date and Core 2.3, 2.4, 2.4.2, 2.5 with 1.13.6 dev, 1.13.5, 1.13.3, 1.13.2 on Sonoff Basic.
The conclusion is 
With all version 1.13.3, 1.13.5, 1.13.6 on all core -> same like zukoy1, i can see on serial monitor "Creating AP" but AP doesn't appear on all computer (MAC or PC) and it's impossible to connect on it.
I have try to custom general.h with my House Access Point and on serial monitor i have seen Scanning but just with my printer and nothing else :-(
```
[000839] [WIFI] Scanning
[001666] [RELAY] Setting relay mask: 0
[002973] [WIFI]     BSSID: 30:E1:71:DE:0E:89 CH: 11 RSSI: -57 SEC: WPA2 SSID: DIRECT-88-HP ENVY 5640 series
[002974] [WIFI] No known networks found
```
The only version full OK is 1.13.2 with Core 1.3 and Arduino 1.8.5 with all library up to date.

If somebody have old version 1.13.2/Arduino 1.8.5  (compile OK) with old library, is it possible to zip it and give a link ?

mcspr, 2019-03-25:
@chryss44 Have you tried simply changing loop delay value?
> [000642] [MAIN] Power saving delay value: 1 ms

either `set loopDelay 10` in the console and `reset` / powercycle
or add `#define LOOP_DELAY 10` to hardware defines, so it is the default

afaik that is the only difference major difference between versions. given some discussions from the esp8266/Arduino repo, I think what needs to be changed is place loop delay between each loop callback instead of a single one in the `systemLoop`, so that the device is switching context more frequently.
my test unit is RF model circa 2017, which is Basic under disguise. setting loop delay with current `dev` makes AP visible and connectable. idk about the new models with esp8285, but will check with some M3 boards

btw, from previous logs:
>  [MAIN] Power: 4466 mV
>  [MAIN] Power: 4094 mV

are you using 5v by any chance? sonoff power input is 3v3

chryss44, 2019-03-25:
Hello Max, 
Thnaks
I'll try when have time
yes, 5v, probably my programmer but since i use it, no problem, Sonoff Basic are resistant :-)
I''l change it too on the next try.

chryss44, 2019-03-25:
I have try 1.13.6-dev Core 2.5 and 2.4.2, Arduino 1.8.5 and loop_delay 10 => No success AP activ on serial but not see in wifi list :-(

The serial monitor 
```
 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld
[000124] 

---8<-------

[000125] [MAIN] ESPURNA 1.13.6-dev
[000125] [MAIN] xose.perez@gmail.com
[000125] [MAIN] http://tinkerman.cat

[000125] [MAIN] CPU chip ID: 0x8B0103
[000128] [MAIN] CPU frequency: 80 MHz
[000132] [MAIN] SDK version: 2.2.1(cfd48f3)
[000136] [MAIN] Core version: 2.4.2
[000139] [MAIN] Core revision: 3140015267
[000142] 
[000143] [MAIN] Flash chip ID: 0x14405E
[000147] [MAIN] Flash speed: 40000000 Hz
[000151] [MAIN] Flash mode: DOUT
[000153] 
[000154] [MAIN] Flash size (CHIP)   :  1048576 bytes /  256 sectors (   0 to  255)
[000161] [MAIN] Flash size (SDK)    :  1048576 bytes /  256 sectors (   0 to  255)
[000168] [MAIN] Reserved            :     4096 bytes /    1 sectors (   0 to    0)
[000176] [MAIN] Firmware size       :   508576 bytes /  125 sectors (   1 to  125)
[000183] [MAIN] Max OTA size        :   512000 bytes /  125 sectors ( 126 to  250)
[000190] [MAIN] EEPROM size         :     4096 bytes /    1 sectors ( 251 to  251)
[000197] [MAIN] Reserved            :    16384 bytes /    4 sectors ( 252 to  255)
[000204] 
[000205] [MAIN] EEPROM sectors: 251, 250
[000209] [MAIN] EEPROM current: 250
[000212] 
[000213] [MAIN] EEPROM:  4096 bytes initially |   225 bytes used ( 5%) |  3871 bytes free (94%)
[000222] [MAIN] Heap  : 40680 bytes initially |  5624 bytes used (13%) | 35056 bytes free (86%)
[000230] [MAIN] Stack :  4096 bytes initially |   876 bytes used (21%) |  3220 bytes free (78%)
[000238] 
[000239] [MAIN] Boot version: 31
[000241] [MAIN] Boot mode: 1
[000244] [MAIN] Last reset reason: Power on
[000248] [MAIN] Last reset info: flag: 0
[000252] 
[000252] [MAIN] Board: ITEAD_SONOFF_BASIC
[000256] [MAIN] Support: ALEXA API BROKER BUTTON DEBUG_SERIAL DEBUG_TELNET DEBUG_WEB DOMOTICZ HOMEASSISTANT LED MDNS_SERVER MQTT NTP SCHEDULER TELNET TERMINAL THINGSPEAK WEB 
[000271] [MAIN] WebUI image: SMALL
[000274] 
[000506] [MAIN] Firmware MD5: e0bab641db3e964b59e985549be24f3f
[000507] [MAIN] Power: 4466 mV
[000507] [MAIN] Power saving delay value: 10 ms
[000507] [MAIN] WiFi Sleep Mode: MODEM
[000510] 

---8<-------

[000521] [TELNET] Listening on port 23
[000522] [WEBSERVER] Webserver running on port 80
[000525] [RELAY] Retrieving mask: 255
[000525] [RELAY] Relay #0 boot mode 0
[000527] [RELAY] #0 set to OFF
[000531] [RELAY] Number of relays: 1
[000533] [BUTTON] Number of buttons: 2
[000537] [LED] Number of leds: 1
[000539] [MQTT] Async ENABLED, SSL DISABLED, Autoconnect ENABLED
[000552] [THINGSPEAK] Async ENABLED, SSL DISABLED
[000618] [WIFI] Creating access point
[000683] [WIFI] Captive portal enabled
[000684] [WIFI] -------------------------------------- MODE AP
[000684] [WIFI] SSID  ESPURNA-8B0103
[000685] [WIFI] PASS  fibonacci
[000687] [WIFI] IP    192.168.4.1
[000690] [WIFI] MAC   6A:C6:3A:8B:01:03
[000694] [WIFI] ----------------------------------------------
[000699] [MDNS] OK
[001530] [RELAY] Setting relay mask: 0
```
I have try 3.3 on my programmer, to program no problem but relay don't work on 3.3, only on 5v.
No difference.

lamptah, 2019-04-05:
I am having similar issue on my Sonoff Basic. However, I found that the behavior is different depending on the power source used.

Using my 3.3V programmer as power source, the Sonoff Basic on 1.13.5 works properly independent of any loopDelay setting. I tried all values between 1ms and 10ms. Power level as reported over serial is 2975mV. I could not find any problems with the device while operating of programmer power.

Using the mains as supply, when having a loopDelay of 1ms up to 4ms produces non-functioning wifi. Setting a loopDelay of 5ms up to 10ms always results in proper function. Power level as reported is 3225mV. 

I also observed that after having set the loopDelay to 3ms while on 3.3V programmer power, and switching to mains power, it successfully started and connected to wifi. After issuing a reset command via serial terminal, subsequent restart yielded the same failure again:

```
[000232] [MAIN] ESPURNA 1.13.5-lamptah
[000233] [MAIN] xose.perez@gmail.com
[000233] [MAIN] http://tinkerman.cat

[000233] [MAIN] CPU chip ID: 0xB01A56
[000237] [MAIN] CPU frequency: 80 MHz
[000240] [MAIN] SDK version: 1.5.3(aec24ac9)
[000244] [MAIN] Core version: 2.3.0
[000247] [MAIN] Core revision: 159542381
[000250]
[000252] [MAIN] Flash chip ID: 0x14405E
[000255] [MAIN] Flash speed: 40000000 Hz
[000258] [MAIN] Flash mode: DOUT
[000261]
[000262] [MAIN] Flash size (CHIP)   :  1048576 bytes /  256 sectors (   0 to  255)
[000270] [MAIN] Flash size (SDK)    :  1048576 bytes /  256 sectors (   0 to  255)
[000276] [MAIN] Reserved            :     4096 bytes /    1 sectors (   0 to    0)
[000284] [MAIN] Firmware size       :   473504 bytes /  116 sectors (   1 to  116)
[000291] [MAIN] Max OTA size        :   548864 bytes /  134 sectors ( 117 to  250)
[000298] [MAIN] EEPROM size         :     4096 bytes /    1 sectors ( 251 to  251)
[000305] [MAIN] Reserved            :    16384 bytes /    4 sectors ( 252 to  255)
[000312]
[000314] [MAIN] EEPROM sectors: 251, 250
[000317] [MAIN] EEPROM current: 251
[000320]
[000321] [MAIN] EEPROM:  4096 bytes initially |   973 bytes used (23%) |  3123 bytes free (76%)
[000330] [MAIN] Heap  : 36152 bytes initially |  5328 bytes used (14%) | 30824 bytes free (85%)
[000338] [MAIN] Stack :  4096 bytes initially |   768 bytes used (18%) |  3328 bytes free (81%)
[000346]
[000347] [MAIN] Boot version: 6
[000349] [MAIN] Boot mode: 1
[000352] [MAIN] Last reset reason: Power on
[000356] [MAIN] Last reset info: flag: 0
[000359]
[000361] [MAIN] Board: ITEAD_SONOFF_BASIC
[000364] [MAIN] Support: ALEXA API BROKER BUTTON DEBUG_SERIAL DEBUG_TELNET DEBUG_WEB DOMOTICZ HOMEASSISTANT LED MDNS_SERVER MQTT NTP SCH
EDULER TELNET TERMINAL THINGSPEAK WEB
[000379] [MAIN] WebUI image: SMALL
[000382]
[000597] [MAIN] Firmware MD5: e51ea3ce89cb154a18dd56d59f156e33
[000598] [MAIN] Power: 3226 mV
[000598] [MAIN] Power saving delay value: 3 ms
[000599] [MAIN] WiFi Sleep Mode: MODEM
[000602]

---8<-------

[000617] [TELNET] Listening on port 23
[000619] [WEBSERVER] Webserver running on port 80
[000622] [RELAY] Retrieving mask: 0
[000622] [RELAY] Relay #0 boot mode 0
[000622] [RELAY] #0 set to OFF
[000624] [RELAY] Number of relays: 1
[000627] [BUTTON] Number of buttons: 2
[000630] [LED] Number of leds: 1
[000633] [MQTT] Async ENABLED, SSL DISABLED, Autoconnect ENABLED
[000646] [THINGSPEAK] Async ENABLED, SSL DISABLED
[000920] [WIFI] Scanning
[001623] [RELAY] Setting relay mask: 0
[003047] [WIFI] --> BSSID: 08:02:8E:E7:E9:44 CH:  3 RSSI: -50 SEC: WPA2 SSID: ferd2.4
[003047] [WIFI]     BSSID: A0:04:60:D0:69:44 CH: 11 RSSI: -39 SEC: WPA2 SSID: fern2.4
[... 82 lines cut ...]
[000608] [WEBSERVER] Webserver running on port 80
[000611] [RELAY] Retrieving mask: 0
[000611] [RELAY] Relay #0 boot mode 0
[000612] [RELAY] #0 set to OFF
[000613] [RELAY] Number of relays: 1
[000616] [BUTTON] Number of buttons: 2
[000619] [LED] Number of leds: 1
[000622] [MQTT] Async ENABLED, SSL DISABLED, Autoconnect ENABLED
[000636] [THINGSPEAK] Async ENABLED, SSL DISABLED
[000909] [WIFI] Scanning
[001613] [RELAY] Setting relay mask: 0
[003035] [WIFI]     BSSID: 88:DE:A9:07:7F:44 CH: 11 RSSI: -70 SEC: WPA2 SSID:
[003035] [WIFI]     BSSID: A0:04:60:D0:69:44 CH: 11 RSSI: -41 SEC: WPA2 SSID: fern2.4
[003039] [WIFI] No known networks found
[003046] [WIFI] Creating access point
[003203] [WIFI] Captive portal enabled
[003205] [WIFI] -------------------------------------- MODE AP
[003205] [WIFI] SSID  ESPURNA-B01A56
[003206] [WIFI] PASS  **********
[003208] [WIFI] IP    192.168.4.1
[003210] [WIFI] MAC   86:F3:EB:B0:1A:56
[003214] [WIFI] ----------------------------------------------
[003220] [MDNS] FAIL
[060001] [MAIN] System OK
```

zukoy1, 2019-04-06:
Hi all,

I tried adding "#define LOOP_DELAY_TIME 10" into hardware.h. But the problem still persist.

Following comments from @lamptah, I was using USB to power on the sonoff basic. I changed to 3.3V power supply and now it is possible to connect to AP, but sonoff basic, cannot connect to my WLAN. I'm really sure about I'm using the correct SSID and password... 😄 

We are uploading the firmware with PlatformIO in a Mac. 
Core version --> 2.3.0
Firmware version --> 1.13.6-dev

This is the serial information:

[002987] [WIFI] Connecting to BSSID: 1C:B7:2C:82:EB:C0 CH: 11, RSSI: -30, SEC: WPA2, SSID: WLAN
[060001] [MAIN] System OK
[062999] [WIFI] **Could not connect to WLAN**
[063002] [WIFI] Creating access point
[063007] [WIFI] Captive portal enabled
[063008] [WIFI] -------------------------------------- MODE AP
[063008] [WIFI] SSID  ESPURNA-79F1A1
[063009] [WIFI] PASS  **********
[063011] [WIFI] IP    192.168.4.1
[063014] [WIFI] MAC   5E:CF:7F:79:F1:A1
[063017] [WIFI] ----------------------------------------------
[063024] [MDNS] OK

Not sure, but in other Espurna releases I could upload the firmware with the USB, then use the AP with the same USB connection and configure Espurna without problems.

We will try to do in different ways to see if something changes, but I'm lost. 

Thanks!

zukoy1, 2019-04-19:
Hi @mcspr !!!

I applied your changes from "**[Safer loop delay default #1699]**" and now it is working!!! 

Thanks so much Max!!!!
zukoy

chryss44, 2019-04-19:
Problem have been result, if i use esptools like that :
`esptool.py --port /dev/cu.usbserial-00000000 write_flash -fs 4MB -fm dout 0x00000 espurna-1.13.5-itead-sonoff-basic.bin`
It's Ok, but if i use 
`esptool.py --port /dev/cu.usbserial-00000000 write_flash -fs 1MB -fm dout 0x00000 espurna-1.13.5-itead-sonoff-basic.bin`
I have all this problem :-( 
I don't understand

closed 2019-09-30 by xoseperez