xoseperez/espurna #1632: Sonoff basic (Sonoff RF R2 POWER V1.0) external switch causes programming mode.

davebuk, 2019-03-12:
The new type of board (https://github.com/arendst/Sonoff-Tasmota/issues/4515) works with the standard Sonoff Basic espurna firmware.
I have a few of these boards and I am trying to set one up with an external toggle switch. I have soldered wires to IO2 and GND and changed 'hardware.h' on the 1.13.6-dev branch from:

#define BUTTON2_PIN         14
#define BUTTON2_MODE        BUTTON_SWITCH | BUTTON_SET_PULLUP
#define BUTTON2_RELAY       1

to

#define BUTTON2_PIN         2
#define BUTTON2_MODE        BUTTON_SWITCH | BUTTON_SET_PULLUP
#define BUTTON2_RELAY       1

The Sonoff works correctly with the toggle switch, the problem is that if the switch is in the closed (made) position and the Sonoff is powered off then on, it powers up in the programming mode.

Is there another change I need to make to stop this happening?

Thanks.

davebuk, 2019-03-13:
To get around my issue for now, I have re compiled using:

`#define BUTTON2_PIN         2
#define BUTTON2_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH | BUTTON_SET_PULLUP
#define BUTTON2_RELAY       1` 

I have wired a door bell across the two wires to act as a manual override. This works well but it would be nice to be able to use a standard wall ON/OFF toggle switch.

davebuk, 2019-03-17:
@xoseperez 

After some more research, it seems the IO2 pin (GPIO2) has the same issue as the main GPIO and puts the Sonoff into program mode at boot if held low. It seems that the Rx pin GPIO3 can be used. I have compiled using:
```
#define BUTTON2_PIN         3
#define BUTTON2_MODE     BUTTON_SWITCH | BUTTON_SET_PULLUP | BUTTON_DEFAULT_HIGH
#define BUTTON2_RELAY     1
```
This seems to work with an external toggle switch and allows normal boot regardless of the switch position. It is also a lot easier to solder a wire to the Rx pin rather than the IO2 pad on the PCB. I will look to install the device into the circuit I want to control soon and test it fully.

If there is a reason to not use this configuration for some reason, please let me know. :-)

davebuk, 2019-03-22:
I have been running this setup now for a few days with my original light switch connected between RX (GPIO3) and ground. The setup seems to be working fine and is controlled correctly from MQTT/ Openhab and the external switch.

closed 2019-03-22 by davebuk

pabloalcantara, 2019-04-27:
@xoseperez The use of gpio 3 on Sonoff Basic R2  could be the default on the next release? Or At least two versions one with it and the other not?