xoseperez/espurna #1633: Error reading data from PZEM004T @ HwSerial (error: 3)

trafoman2004, 2019-03-12:
I'm not getting success with PZEM004T sensor on Espurna 1.13.5,  it return only 0's for all magnitudes.
Debug window shows: [SENSOR] Error reading data from PZEM004T @ HwSerial (error: 3).
I have tested the hardware with PZEM_ESP8266HwSerial.ino and it's working as it should.

mcspr, 2019-03-13:
https://github.com/olehs/PZEM004T/blob/ecb01be2a6bfbb6b33f9fe5b5c70efb2c9ee8843/examples/PZEM_ESP8266HwSerial/PZEM_ESP8266HwSerial.ino#L23
example you mention swaps UART to GPIO13 and GPIO15.
I think the related setting here is `PZEM004T_HW_PORT` which should be set to `Serial1`

trafoman2004, 2019-03-13:
It was my mistake, sorry. I connected as per the wiki instructions but it doesn't mention that pins 13 & 15 are only for soft serial, it also says that Serial1 is default, but in sensors.h is configured as: #define PZEM004T_HW_PORT  Serial.

mcspr, 2019-03-13:
Hold on. I'm sorry too, `Serial1` is a different thing!
gpio 13 & 15 are still for `Seria`l (aka UART0), but we don't actually call `Serial.swap()` anywhere to make the switch.

It should be called *somewhere* before the `Serial.begin(...baud...)`. In case of PZEM, somewhere here:
https://github.com/xoseperez/espurna/blob/ec4883a4fc0033873c5db89de25e1d993517ae94/code/espurna/sensor.ino#L1346

closed 2019-05-20 by stale[bot]