xoseperez/espurna #1634: After compiling firmware for millet lamp

1043717432, 2019-03-13:
Hi, thanks for sharing. After compiling firmware for millet lamp, the function realized is only to click on and off, only channel 0 channel is opened, channel 1 channel can not be opened, and the encoder rotation is all brightness adjustment. When the encoder is rotated quickly, the web is not responding, and the system runs out!

mcspr, 2019-03-13:
Have you tried Encoder lib from https://github.com/xoseperez/espurna/issues/1447#issuecomment-450510619 ?
Are you building with current `dev`? I think the Web part was fixed.

1043717432, 2019-03-18:
> 你从[＃1447（评论）](https://github.com/xoseperez/espurna/issues/1447#issuecomment-450510619)尝试过Encoder lib 吗？
> 你在建设当前`dev`吗？我认为Web部分是固定的。

hi,this is the configuration file
`- name: diytaideng
  platform: mqtt
  state_topic: ESPURNA-1C184A/relay/0
  command_topic: ESPURNA-1C184A/relay/0/set
  brightness_state_topic: "ESPURNA-1C184A/brightness"
  brightness_command_topic: "ESPURNA-1C184A/brightness/set"
  brightness_scale: 255
  color_temp_state_topic: "ESPURNA-1C184A/channel/0"
  color_temp_command_topic: "ESPURNA-1C184A/channel/0/set"
  color_scale: 255
  payload_on: 1
  payload_off: 0
  availability_topic: ESPURNA-1C184A/status
  payload_available: 1
  payload_not_available: 0
  retain: true`
Unable to adjust brightness

Unable to change color temperature

Only open and close

1043717432, 2019-03-18:
hi,this is the configuration file
Unable to adjust brightness

Unable to change color temperature

Only open and close
```- name: diytaideng
  platform: mqtt
  state_topic: ESPURNA-1C184A/relay/0
  command_topic: ESPURNA-1C184A/relay/0/set
  brightness_state_topic: "ESPURNA-1C184A/brightness"
  brightness_command_topic: "ESPURNA-1C184A/brightness/set"
  brightness_scale: 255
  color_temp_state_topic: "ESPURNA-1C184A/channel/0"
  color_temp_command_topic: "ESPURNA-1C184A/channel/0/set"
  color_scale: 255
  payload_on: 1
  payload_off: 0
  availability_topic: ESPURNA-1C184A/status
  payload_available: 1
  payload_not_available: 0
  retain: true```

1043717432, 2019-03-19:
> 你从[＃1447（评论）](https://github.com/xoseperez/espurna/issues/1447#issuecomment-450510619)尝试过Encoder lib 吗？
> 你在建设当前`dev`吗？我认为Web部分是固定的。
[image]
This is my DIY desk lamp, which is compiled with the firmware of millet desk lamp. I want to control the brightness and color temperature in the family assistant. At present, I can't control it, only turn it on and off.

mcspr, 2019-03-19:
> This is my DIY desk lamp, which is compiled with the firmware of millet desk lamp. I want to control the brightness and color temperature in the family assistant. At present, I can't control it, only turn it on and off.

Cannot control it using MQTT? Home Assistant light mqtt component only has a single White Light control, but we need 2-channel control. You can replace brightness topics with `ESPURNA-1C184A/channel/1/set` & `ESPURNA-1C184A/channel/1`. Brightness MQTT topic does not work right now, code needs changing.

> When the encoder is rotated quickly, the web is not responding, and the system runs out!

What about the encoder problem? What kind of encoder is used? What serial debug say about delta values or it reboots even before that? Have you tried this [Encoder library](https://github.com/mcspr/Encoder/tree/esp8266/isr-in-ram)?

1043717432, 2019-03-25:
> > 这是我的DIY台灯，用小米台灯的固件编译。我想控制家庭助理的亮度和色温。目前，我无法控制它，只能打开和关闭它。
> 
> 无法使用MQTT控制它？Home Assistant light mqtt组件只有一个White Light控件，但我们需要2通道控制。您可以用`ESPURNA-1C184A/channel/1/set`＆替换亮度主题`ESPURNA-1C184A/channel/1`。Brightness MQTT主题现在不起作用，代码需要改变。
> 
> > 当编码器快速旋转时，Web没有响应，系统用完了！
> 
> 编码器问题怎么样？使用什么样的编码器？什么串行调试说delta值或者在此之前它重新启动？你试过这个[编码器库](https://github.com/mcspr/Encoder/tree/esp8266/isr-in-ram)吗？

Thank you for your reply!

At present, the use of millet lamp firmware, encoder rotation to adjust brightness and adjust cold color to warm color can not be displayed in the family assistant, how can I write yaml?

I want to be able to adjust brightness and color temperature in a family assistant.

1043717432, 2019-03-27:
> Brightness MQTT topic does not work right now, code needs changing.

Where does the code need to be changed before the brightness theme can be used?

mcspr, 2019-03-27:
> Where does the code need to be changed before the brightness theme can be used?

https://github.com/xoseperez/espurna/blob/21de702bccc8301a2a938c4b035cc87ff2b84ab2/code/espurna/light.ino#L503
this line needs to be outside the `if { ... }` scope

1043717432, 2019-03-27:
> .. }
What do you mean，

I don't understand. Can you elaborate? Thank you!

closed 2019-06-02 by stale[bot]