xoseperez/espurna #1635: Magic Home ZJ-WFMN-A 1.1 - White always on

Kerwood, 2019-03-14:
I flashed my Magic Home RGBW LED Controller with `espurna-1.13.5-magichome-zj-wfmn-a-11.bin` pre-built binary, from the releases page.
I am not able to control White, I'm guessing that's channel 3. It is just constantly on.

[image]
[image]

Any ideas ?

ColinShorts, 2019-03-14:
Try 1.13.5-magichome-zj-wfmn-b-11.bin.

There are a couple of revisions of hardware for some devices, this one has white on a different channel.

Kerwood, 2019-03-14:
Thanks.. Will give it a shot

Kerwood, 2019-04-01:
Nope.. the "b" version did not do the trick.

ColinShorts, 2019-04-01:
I guess you'll need to figure out which gpio pin is connected to the white channel. If you've got access to a multimeter you'll probably be able to figure out which pin on the chip is connected to the transistor that's driving the white channel.

Failing that, you might be able to use an alternative firmware that allows you to change the gpio on the fly, you already know which pins drive RGB, so not too much trial and error required.

The last scenario is a hardware fault, but hopefully not that.

MonsieurBibo, 2019-04-20:
hey, I found that the gpio 15 turns the white LEDs on/off.

MonsieurBibo, 2019-04-20:
> hey, I found that the gpio 15 turns the white LEDs on/off.

Unfornately, I think we can just on/off the white LEDs.

Kerwood, 2019-04-21:
How exactly ?

MonsieurBibo, 2019-04-21:
> How exactly ?

if you use the debug terminal with the command "gpio 15 0" or "gpio 15 1" you can turn on/off the white LEDs.

closed 2019-06-27 by stale[bot]