xoseperez/espurna #1652: mqtt LWT

CrispinP, 2019-03-24:
Could you please implement LWT for mqtt?

A separate topic and message (both static) config world work.

I use sonoff to tell me if a circuit is energised. Having a lwt allows quicker response than scheduled ping. 

Thanks.

jackwilsdon, 2019-03-24:
The devices publishes a LWT on the `/status` topic, with a `1` being published by the device when it is online and a LWT of `0`.

CrispinP, 2019-03-24:
I'm seeing the 1 but when I drop the power to sonoff it does not change to zero.
I had a look at the $sys topic as well and see that it does not change when the sonoff drops. Why might that be? Obviously the LWT will not get sent it Mosqiutto does not see it drop.

[image]

None of the values there change when the sonoff is on or off (unclean shutdown)

flashy02, 2019-03-25:
I'm using the LWT for monitoring all devices and can say, it works fine. Additionally I control with the LWT the automated garagedoor with no problems. You can config the LWT in the UI. Since some versions of espurna, the time constant is 300s, that means at least after 300s being offline the broker sends the LWT. I'm using normally 30s or 10s for the garagedooropener.

xoseperez, 2019-03-25:
@CrispinP: As @flashy02 says the keep alive (`MQTT_KEEPALIVE`) time by default is 300s. Please allow 5 minutes to receive the LWT message or change the keep-alive value for a faster response.

CrispinP, 2019-03-25:
Thanks both. I realised I was not waiting very long. Did not know
until @flashy02 mentioned the 300s that there was a delay.

Could I change the topic then to make the time shorter? 300s is an eternity
:)

Regards

flashy02, 2019-03-25:
You have to change the time in the UI. The topic is still to find under ".../status".
[image]

CrispinP, 2019-03-25:
I'm getting old. I did not check :(

CrispinP, 2019-04-11:
Thanks both, this works as expected :)

closed 2019-04-11 by CrispinP