xoseperez/espurna #1659: webinterface bug when connecting through internet

hamses33, 2019-03-26:
Hello,

when I'am accessing my espurnas locally the webinterface works fine but when I'am connecting through the internet the login prompt screen appears and I can login but on mobile  phone the website wont show anything and on PC it show this:  

08uÊuÿV×Í„ ¦k±BF‹’’ÖQJW¨•®ÄX!«IŒÉV
³Âª¢uˆ6SZHkN�8ˆ¡e­3–t}Õ4€½&¾5Ló6Õ f¶‰A–8+ÿÔ ‡NDÙºQI4»(:¦®¥R›"±ýDõhKÜÊ~j¥{§¤‰Z%¤ÆHÓïsPÁ™Hz¼kÓ3]pÏGfªëµ}¤íèªU]i¦Saç0ºa†€HáJíâÔ	hNp¥õŠóÀ“Íåd³åÌ¥zýÅ:{!Íjý~Æ$ON¬ºšhÄ$c½79ž±1¨°âVß
Â¨Òº£^ÒºX±~¤Ïd‘
\#4£zPòk'Èsª„¥Pk-›Ck•nœ"$žX9)='b…	ÀžYg43Kn";n~±1°²anŠ°Þä%M@Bnœ›(Œ$yIËùe~BQ‡¦A~MŒÅ	Â)_•$g
;|hþÖëåZ˜v³”%çY·ÛM-++“Ëö]­ä×ž+n$Všê,5¹¢D’êø¦�¢çÄüÿQ×‹§–²QU›˜w5WGH¼ä×UœN“uv²6eïÎ8#Ÿ¾Åé|§KCla5¡‹³HÛgaœÜ³RaŠÏHž�œîÅ©4éSÉBÜN~”@@±éµFs£ÏœvB*ÊÐ„4²ÓwG³zÂ-ò)	çJ½†�B5
¿²IKûF•¤ªrWl•6Å
é]Û½(¤ùÙyˆîïÎ¢Ñþ|™ïÈÆçÀì<|ÌÎ#æÂìLj>ÌÎ&æÄìL8/îÊsãŽ,w#çÈì4Or²ð¹’“ ç‹&ýæ¨Yëð*ÞºN¯Ù[Î‘nSÓ	jí¡v™"›  PIyP0µc!!I<¾xñ"5#hþÁsËf/—id“ÌC&§˜ÛXn62N7Vœ•õ9I¦±|†2+›ÎXp‚Á$¡»Mv7™ ß,ñÂÔÑäÛ©¡OG‘;Á½}[Of9§lt`S.Ð*Cz‹¬®t‚Ô~X]l…©šõ-1Ýá@�ÎVñ¯”åååvë”«Ë VÅ
b{í¤¹
Åmžîëe9¬•›#OôÏË‘PŠ2ä¨³²ÎÌ”T
òrÎÊÔ!òr¤t‰¼,"/SJ·ÈË¢ëùéš®‘›á.ŒéºG^zBQˆü3_“Šú¬4•ü5¾™Xä¹è¥à}ÈF“
—›­t2wÔHMA ?KTÂ˜…¡3`¢ò¹|“’ÊH¦êë¬/¯KÂ›IØSsUc…eçÉ…Xíº7ˆ¯Ôuë-éòE¶ZÁf6*7Áî ™†Æ·ª8T0&b¾,ÅÖÓ©)»JSçRÂð¨×Ì»®xjoºfšP™Vã•6ìä'ö²ÒÐ|°­g*r˜j‚‘’ 3À	ú;à¨Ç.[Ë…ÕÄ6£Y•e-K±kàÿHj9BØg×¦•æk>Ç
¥éÓóB0W«…-áŠ~'[¬4óšf?Éœ–“ˆ"ãÆuvS3;^Éd}7-'¹Qi&÷þÒÉËj·279µ*“ó¸iŠ
¡bmm-ßloT×Qh‘Ó‚ì’\Ö˜Í5E{y
å· Ý<hï&ìŒyÜÝ#v³Yu=óZöS¸.qr,_p¶U¬N+‘ÿË¾ù÷Û7“[‘0{›b©áÓ’Òã—IR˜›’ÁÂ<ù©ñf’âAÌ$i£˜IKc&UÇLšÈœ1’Ù”ÂŽÉ±Ì$Ð`¦Væ\©n–8S¨Âh’òš<f‘eN~x—œ˜áU:3ä†ZPëÖZ±Atžˆ×ª,[é»— 1Õ<¹é+­dõvBr 2nÄÆ–|¹at‘ZnÉøðU§-½ŒºZOÔ-µbÚ3ž‘ƒ	à‹Šw–ÂcÞsŽÿ@\¥rf¼Îñ2HRFª€‘èrÞ7«Pº/9YhÏIÙÄnúÁ=tÍqBµ‘h0å½(“H
R[«m"£‚Q‹‰Ë9
ýÑ4b…ô•Ó7Õž T,Jî|H„¿ªïc¿J—VØ#Sw/àØ™¬):Ê�ë„ªÐ+[N?bÁµty-}k®n­—´™Mñ)1G¨i…TrŸÆvKª*nÐ[Ý˜¥ÏsåRDR´Õ—«Ï—S,Ÿ¸‹¨zÞòwLŽèÑÇIÁ«qå8¯Ò¨R^äO»Ã¶Öm¢­Ó
	±>“ÈB¨¹¼Ó·S'HÌ‘yˆ<;£­ìÏ#5³c#VH¸anV³q²4-‹JM¥7Áú&—ë@j:ñÚ¤ÕSg€R|_][í­­´S¡,ÓÉ„]'Ô×àùIëurçmU6Ý¦xMÉ3$Š_+”ÕLš‘˜§—¥lØXÃâ	°bÍÿÎêÑêÏjá½bäÖœ›JFèð¢…‹Aº‘0€Ÿâ_m›áŽˆ‰6O¢«J—Fa+úªrC„j
‹Ð(Šy3ÖWÿUkNÚç_™Th*g
ãê#yñ<”ê=_ÇŠúÃ‰Wyci=¯ÕNjå¢$$iŒ„*ÏèBÎ:­¤Þ¼Ñ¯ã_-Ö/#ó¦#öH<Ö»“d)cÅïõtœ‹Ü	š2ƒh&¼å¦Wi˜¹u>ë7ño\ï.V0cÛ‚O^!ŒU}mÇœWéŠ2XÔ›Þýóð.`²´"Xg¶®ûÀø²ÆãzEx¯0`$l ÜûÏ
“ñ¦ð@zãSò¹©½$žW´—eíYÿ¾ª=¯iÏëñs=QS²Ád+ÍÄÛrâm%ñ¶šx[K¼½H¼­'Þ6ô·fòeY{Ž»ÕL€ßL�ÜLÖ�¸™�x9ù¢?¯jÏ1Ö¥W´L«:þhXMäZÓžãj˜z¡Wõ"™I ð?*"è1/Z$MÃE1#)ªÌI‘qaü'ýt‰Ü(’ÔÄÉ‹%ƒN’šƒ…A%)*Ê-ážŽ.ÉLÕÂ9qsÃLÒl$/ØDÎîÜx“Ìd/Œ:áÓ'x’a¹á'¶Q„"9JnJ†ÁäD£H¶“Â'w:&%ÃvŠ"S2)7>%Ãv‹£T$Ë
T)\D’¼.´’^²èôˆ¤¼îŽÀ$Â‰ï…èA?#àõðYÏèÌˆj, S°	$BKRÒ
ªSSlÌÌtv²<(cFžipegÓuþœdþ*Wêïµ°®LºÊ5é¦sj8QÙo«Â,ŒoW¬RòÈ«nµNz£ÚýÀ¶”Ç ~€ÇÆtÿ‘Œ¨I’o³H.çÕ‰£,PýÔC²—ëõxƒŸ+·²XóZÓ…óËT¸ÕI³&Ê ÷pzJò³¨b}}½¨YÄˆtMÜ~–0ÁÚôs7õÆz[²Û‚|ôV¡qÖŒ4rí³11(SZõ3Ìpb'!Ñ¯X
Yãæä5é*Ì	ÕÏJ¼Ó‘~zîóª³:’«ÊzJ0žÐÒË©I…èÔ$“ªkäbÛÊIH«»­l
J£O~Í	á€é¯÷zk·ù9
yDrÛÐ2è¡O
Í z«ÍP­i=Æã…Y¸qI.:ƒL2˜]¨ñò©=ƒXÐ%"¦
$iR(6(Ý_e0@n+4–Æ»ÉüVê[/3_OYßXÛÀr€„]þ>5+s©QÒ¹C%‰¿Øä›W÷µV¦PÊ4Mle41Ñš%yhbÐ¼TD¸²ÛYëgÿÉ•A§®<Ë<òóYÌµ~Ä]RKïR²ðuN#ŠÒr@õÏ_·?T.¾º½‹{¨¶8[@ñÌqœÙq˜«!œPôé´j*«èz½ÐÝFmJH±S5~
ªHbñFŸrmå¦â9$tÔ\r&‰l`ÈØˆ‹4É¯jGv%S6Zmž¨ÊE`­–°	spo[<éiÄcL×›k"áÁë]=vƒ6¬XÐ�™•ˆ,°BhÀ”ž¸TÖë)–¦x“Fµ±Ê™¾NnY:¾$£JŒˆ¶É\ï!Ðì£Ë
hÙÂú¦¥˜¯Ò*\¼ù¥s.ÐmBêubš8±Ç±ÞvÎ4 Üif| ëÐ+‚s5^ž*k¹saÅ\B·ˆq\hå
ð`³¢†çÖ]Ï¦è˜ïö&Õè¢Æ®[Évó’8‘hž\	êÉ6Ònù—y$³¢"çdT™½ùšÑfI!ÕÌž1ñî\§I �£4…ŒdÑÂ	vç'ÑN}'‰®IÑ²ªø=-çbIÆ›¶ÔÄêŒ&Cç°p€µ>
FåüÃyÝ/[ï?]Ô~=ðñdÇƒÃÏÃÝÏxh$ÖÛùØùH‡>Ž>|­»ø°5Úz÷e÷KçãîV­¿¼ÁœŸ>Oñûö/[û_Á¼áüóv÷b·3ž\ÐÉµ[ËŸ~:úüùÍòÇ_?mí~ÞèýÜ{GmŒêŸ¾ë›Ñé×/Ó^gk§;þ2ì½þr¶ÿ¦7é½>ÕGŸ÷ØÖÎ°‹U}:üüiëKçÏ_~­ýùáb¿óå×Oý‹­¥ó‹ÚåÚù»£T¹õÓ§Ï«»ÁÙOƒÁÀ¶Mƒ¯†TÊ8®|Á;ÓjKJÎLà>©‘ˆŠÆþ©;bi -Z™9qè0B#>F‹X9±V3rÐJh'%¦ä‹ù3Ž5HïXŠ“9SÛÉòH†A‘S“ÕHE–¶³ªyÑàiÚµùU2–d%KÂOÌ›áô\~¥›qòv"”°¸Ü$ù—‹½¯2ÇT&¬d3=>y‚bÄmÍ•‹o¤sÙ+í‹€. šHÎû2t½(c‡\Æ,W Q1R§2‚xí÷¦#fUåy âÐ;:«u:Á'Ø‰ÅÇNäM-"‰²øÖi{µóÙ—!Ð¨“I2Dˆð4›jô™ôï3”%(a÷‘J@6™Z#<ø4_aGY]ÃvS°éÃ2á’j0 ®Ôé¢"TÈä¤àCb¦|€ù�4+‡"DÊG[}Î=’!Þúì×ç”lûÓÀNwÀ.rOïDP2äY-8ÀaEŒBK`ÁTMºä(‹ñCxåQrLþ�è¨3!•"b2Ï¢¦èÒF=Š–çy0<=xÈˆdõù³Š(2uGx®J…VòcßAsé¯ÂÇ1þÜ;ð$Koxò9]hß<ÉžYy�ªC¿«×€ú~¤½={´#¿ôO‘«Á‚£>rAÛ!D7òñ4‘ ¦sB5íÖë´ÉC`z½ìø-
»3qÏØ•Ž„î¸§w
é¯�vß$ FœyY ¸é$û=`!t†iAŸâÔkEò…tŸ+§²”Ù§›îúˆÎðÈ
&…nÔÆ•­©bæ¼ì€Î >fšA¬„ÀgÕ‡·ïâ	Ø^¢µ~]‡83ô¾'2¯4O×{½Û‚q{òŸ1òûE£¡Û‡qVòeBÈéé%x¥HÓo&=#Ÿ/IëÍ¥ÎÐ_àìüÁ`Ä¾r-!‡é;t—´eÍP«l;u¨³æ-ª{ˆ%Ú·’oläêHê9†‰~"ÊªM·û‹¤{;Ýé©Û­œ²ï°F”«++«xf”U]…ÿ7VÍ„M€màß½G…6ðï\G¥%•Â¼}õ™æ’9ò_ 9*¤±T¼	#éÃñ£;±§ä¼å|ãÛýÛ
Fó6¶²í¼/ql	š¼†È»:Y**÷~o¯”ÄLœöëîaIh:wÔš*øþ@ÕùÇC“G,Hb˜´zBùò²\7gí>ÞƒBùÚ•¥PCD±ÍCŒùÛ•ºU"ü‡Ò&Zœ”þÈQÜ¢ã�X/ƒ^µ/“ŸM­6yÖ‚,…U÷½5`Ôf;gõKÚU²£Äòšc×
N‚J9+†7AÝîmuäs­6s.[Ê2pûÒò©>zý~g4Üët>w~ÚÝé¼ûþîûÛ÷ßÃåÞMS»zSÛ®¯î}Ù;Øùøekï»rÑ¹ØE[ÃpéãÇ½íAg§#.)¢::í?]}ø«;=Ø®¶?ÿäÜòëƒýËÉ/ï»Ñ¯oœ“_ÏÜ7ýwÍîç`ûà¨îý¬L§½ð{m{ÕsjÍwÃ•7Ó¥7ŸÖ~Ýr&£O;‰¶vñÆ¤Nç5¶µ7¹èuN¶×·.ýÁ§•µË¯¿½ÕGë¿ýú²î÷Ïv·>úxiG¸
ù;Ÿ_½4ìvßý<jNß\¾ÞÞ>¿ÄŽÆn8ìnoÿúç×Ýßö‰6ß¡Ehk@lï]^\0‡G+?¯|Ù]Ö;îûÞêaß=Xý‚9Þ¬nu>ÿŒ—„¼í°ÑŸ±]€ûWog½Ã~þ«ûfÿâbøÚ‹v¿¾={¿¼ÝÝZ9uv÷÷’e?b?¡M´&íõv'7ÎÊë7¸ýæôuïÔx[½Ã7“a
GÐä�ûù3Á»zQÿëg¥¿·öógÿËÇ«ŸÎ‹­pùŠüÚ9ûšls-`¯aÐæ/Þöç¦û®ï|ÿ:¿›4Æ~ÇsÕßÿuÿ/†yaÛfBTfhbÂº`Nš~o¹Õô}k±)†3*/ª±ü’km<©çéûB™b†Ÿçžjå32Q TšŽ%çÆý´zÞ‹¸@ÂÃ—•ÕdnÇÑ5ëÕ¼*²úß3)DÊ+Sro:i¦Ä¾´EâÈvò	 Fþ”22©=$Ã“õLÊëü}¶ÃÈÍ˜ÛdôÚQH5=t¶¹ÆôÅÿO,œŽ¢d?Ó}Kv•P~>Ñ€Ÿû_ExÄŸxÜe=’sÐ'í³(üƒ+)‹ÿ»Ñwåîr÷ö¬ # =Ÿšh´ÌµÈ	•´»sæd¼1òÃªi†ŸÐ [>vÇþŸîáÕøÔ…Ö!øÌø¼oðºÃí¯gãñ„—ÊrŠnÐ`_g&òÕz¦iéÿùßþçÿ(Ýæ…§.ePÀ9„5$ÿ²`"[Õeó6mµ,*Iæ•µ5«±¾b5›
<Èö¶êùÂéã!-gÒ„\ÑÂZ¸)Dû €K–ÌÃð!©Ë†þ¨³³n·'óå¦vcÃvßfHnç/ÃnÏ.Iì”:ôß.	Cb‰gÅÁGn†‚øð!]˜›ŽKd“êúã	h1Ì.ùý~²:â^ÑiI@ìLÁ’ÈÌ­ÇðmØxu¸»ýùÓþÑ¯Ð—|h¾Ú"áÍ˜†0§hè†¿

mcspr, 2019-03-27:
See #1610 and https://github.com/xoseperez/espurna/issues/1610#issuecomment-473794750

mcspr, 2019-05-19:
Closing via #1723

closed 2019-05-19 by mcspr