xoseperez/espurna #1661: Configuring TX/RX pins for sensor use.

davebuk, 2019-03-27:
I have been reading various posts that suggest I can compile a Sonoff Basic V2 to use the TX or RX pins for temperature sensors.

What I'm not sure about, some posts say using them will disable the use of serial data. Does that stop the Sonoff being reflashed using the header pins or do you just disconnect the sensor to allow flashing again.

mcspr, 2019-03-28:
Right. You change `DEBUG_SERIAL_SUPPORT` to `0`, so ESPurna does not initialize serial logging (`Serial.begin(...)`) and pins can be used as general GPIOs

And it is a good idea to detach the sensor when flashing again.

davebuk, 2019-03-28:
Thanks. I am using PlatformIO on Windows 10 to build the modified binaries for flashing. Do I add `#define DEBUG_SERIAL_SUPPORT    0` to the hardware.h file under the Sonoff Basic entry?

mcspr, 2019-03-29:
Either way. If you do not mind modifying original files.

We also support optional definitions file at `code/espurna/config/custom.h`. It will be enabled if you supply `-DUSE_CUSTOM_H` in either platformio `build_flags` or using system env variable `PLATFORMIO_BUILD_FLAGS`. See https://docs.platformio.org/en/latest/envvars.html#building

davebuk, 2019-03-29:
Great. I've been looking at custom.h files. From what I understand, entries in custom.h override any other pre defined settings. E.G. change GPIO pin associations. Is that correct?

I had a play disabling some of the build options to reduce the size of a binary for a RGB magic home controller. It was coming out at 501KB and wouldn't load via web OTA. It's now around 470KB and loads fine. I did this by adding various entries in the hardware.h file. I'll look at using custom.h instead.

I'm trying to work out if it's worth adding my own environment entries to platformio.ini files which then have -DUSE_CUSTOM_H in the build entry?

mcspr, 2019-03-30:
default settings will not be applied if you define settings beforehand (see .h #ifndef #define ... #endif constructs). custom is at the top of all.h, so it is loaded first. if something is missing #ifndef, compiler will show warning about redefinitions.

it depends. if you use system environment to set platformio build flag defaults, all envs will use them and load default wifi, mqtt etc. settings

davebuk, 2019-03-30:
@mcspr Thanks for your help. The first set of temp probes have arrived so I'll see what I can get working over the next few weeks. I guess when V2 of espurna is ready, some of these configurations will all be done from the web gui?

closed 2019-03-30 by davebuk

mcspr, 2019-04-04:
> I guess when V2 of espurna is ready, some of these configurations will all be done from the web gui?

Right. Which reminded me (once again) to actually do it. #1680 for basic support, without handling edge case like this when we use serial pins and need to disable debugging from runtime. I just wonder what kind of safeguard to put there to start up serial debugging back. Maybe new button action...

davebuk, 2019-04-05:
Like you said, I guess a button would work. Maybe a reboot button option that disables the TX/RX GPIO mapping and re-enabling debugging on restart. The toggle would allow one or the other option to be enabled but not both.

I guess it creates a problem with the firmware if serial debugging is enabled and something else is connected to one of the serial GPIOs?

davebuk, 2019-04-05:
I managed to get the custom.h file working. I've added a DALLAS 1 wire temp sensor to a SWA1 WiFi switch to one of the serial pins.

I have also got a BME280 environmental sensor connected to a Sonoff Basic V2 across GPIO 1 and 3 (TX/RX). It is working but with some issues. See #1672