xoseperez/espurna #1671: Blank web interface, problem with a <style> element?

wsw70, 2019-04-02:
# Bug description

In some conditions (over a VPN, mostly - but not only) the web interface is blank. 

# Steps to reproduce

## My setup
- a 1Gbps switch to which are connected:
  - a server ( called `srv` later)
  - an Ubiquity WiFi access point to which are connected 
    - Sonoff devices (mostly Sonoff Basic but also RF Bridge) (called `espurna` later)
    - a laptop with Windows 10, latest Chrome (called `local-laptop` later)
  - a VPN endpoint (Wireguard) to which is connected a Windows 10 + latest Chrome (called `remote-laptop` later)

### What works
- `espurna` connects to the Wi-Fi and works perfectly (MQTT-driven)
- access from `local-laptop` to `espurna` is fine, I see the web interface.

### What does not work 
- access from `remote-laptop` to `espurna`: the web interface is blank. It is connected  but is blank (details below)

## What I thought the problem was

Initially I thought that the problem was due to some MTU issues over the Wi-Fi connection - that in some conditions the content of the web interface is loaded, and in others not. There was a case open for a similar problem.

## Why is it different
I analyzed the output of the web interface, collected from `remote-laptop` and `srv` via a 

    curl -v switch-9 --output switch.html.gz

The file is correctly gzipped and unzips to `switch.html` which is a correct HTML file. Opening this in a browser results in a blank screen. Analyzing the output in DevTools, I found out that **removing the fourth (last) `<style>` element in `<head>` brings back the output**. This is not a proper page as all elements are visible at once:

[image]

When doing the same action (removing the `<style>` element) on a live connection, I get the same output.

**The thing is that over the WiFi connection at home I get a correct output.** 

I cannot make sense of that. The only difference between the accesses is that one (`laptop-local`) is directly over the same WiFi `espurna` is on, and the other one is via VPN.
The file loaded on `srv` (not over the WiFi but not over the VPN either) is incorrect too.

I have some vague idea that, maybe, some kind of special connectivity is required in the JavaScript of the page which breaks one case and not the other ?

**Device information**
* *ESPurna version* 1.13.5

**Tools used**
* *Desktop operating system* Windows 10
* *Browser & version* Chrome 73

mcspr, 2019-04-02:
Problem in #1610 affects websocket, not base page load, and what you see is
just a result of removing display:none rule

Running packet capture on the AP device, you can actually see ESP tries to
send malformed websocket data back. No way around that besides either
patching network lib (in case of Core 2.3.0), or working around it by
reducing single data packet size.

>

mcspr, 2019-05-19:
Closing via #1723

closed 2019-05-19 by mcspr

excalq, 2019-07-16:
Curious if there's a workaround for this, as I have an irrigation system deployed remotely (four hours drive away) and am unable to access it via HTTP to change other settings. Changing my OpenVPN client MTU isn't helping so far.

wsw70, 2019-07-16:
@excalq I did not find any. I moved from openVPN to Wireguard but this did not change anything.

mcspr, 2019-07-16:
Telnet interface should still work though? If would expect `/config` endpoint to work too, but I have not tested that

Any MTU lower than the 1500 breaks this, which is unavoidable with referenced vpn software and that's the general problem :/
Current nightlies build with the fix btw. So can any environment that runs `pio run -t build-lwip` before building anything else.

I would look at this again for 1.13.6, since doing large payloads in ws also causes crashes when not careful.