xoseperez/espurna #1672: BME280/BMP280 sensor shows 'Wrong ID' in values section.

davebuk, 2019-04-02:
I have setup a Sonoff basic to use Pins 1 and 3 (Tx/Rx) for a BMP280, 4 pin sensor.  I have the following in custom.h
```
#define DEBUG_SERIAL_SUPPORT 0
#define BMX280_SUPPORT 1
#define BMX280_ADDRESS 0x00
#define I2C_SDA_PIN 1  //TX PIN
#define I2C_SCL_PIN 3  //RX PIN
```
Once built I get the following:

[image]

and at Debug I get: `[SENSOR] Error reading data from BMP280 @ I2C (0x00) (error: 4)`

I have also tried setting 0x58, 0x77 and 0x76 but get the same status page as above. I guess espurna is not setting the address, or do I need to install pull up resistors on the data lines?

davebuk, 2019-04-04:
After a reboot this morning, the sensor seems to be working. The main status page is reporting correct values against BME280 @ I2C (0x00) but am still getting the following error 6 in the debug output.

```
[SENSOR] Initializing BMP280 @ I2C (0x00)
[SENSOR]  -> ERROR 6
[SENSOR] Initializing BMP280 @ I2C (0x00)
[SENSOR]  -> ERROR 6
[SENSOR] Initializing BMP280 @ I2C (0x00)
[SENSOR]  -> ERROR 6
[SENSOR] Error reading data from BMP280 @ I2C (0x00) (error: 4)
[SENSOR] Initializing BMP280 @ I2C (0x00)
[SENSOR]  -> ERROR 6
[SENSOR] Initializing BMP280 @ I2C (0x00)
[SENSOR]  -> ERROR 6
[SENSOR] Initializing BMP280 @ I2C (0x00)
[SENSOR]  -> ERROR 6
[SENSOR] Error reading data from BMP280 @ I2C (0x00) (error: 4)
[MQTT] Sending SonoffB005/temperature => 18.120 (PID 1)
[MQTT] Sending SonoffB005/pressure => 991.2406 (PID 1)
[MQTT] Sending SonoffB005/humidity => 51.59 (PID 1)
[SENSOR] Initializing BMP280 @ I2C (0x00)
[SENSOR]  -> ERROR 6
```

mcspr, 2019-04-06:
My bet would be on I2C not being able to work properly over (mildly) long cable.
I've have a same sensor attached to a Basic, with ~30cm cables. No problems there.

Error code should probably be stringified... ref about 6 meaning:
https://github.com/xoseperez/espurna/blob/903ceb1e844586a6b954c294d7a4438c75a08467/code/espurna/sensors/BaseSensor.h#L13-L22

davebuk, 2019-04-07:
I rewired with just a very short cable, 2-3cm and it still didn't work. After a re compile of the binary and a re flash it started working but still have the error 6 messages against BMP280 rather than the BME280 (BMP280 not fitted). I've tried again using a screened data pair approx 30cm long and the BME280 data is working, but I get a constant error 6 message every 10 seconds.

mcspr, 2019-04-08:
Two POI. Inside _read, when data returns `0xffff`:
https://github.com/xoseperez/espurna/blob/903ceb1e844586a6b954c294d7a4438c75a08467/code/espurna/sensors/BMX280Sensor.h#L325
Inside _pre, when it tries to read chip register to determine if it is BMP or BME:
https://github.com/xoseperez/espurna/blob/903ceb1e844586a6b954c294d7a4438c75a08467/code/espurna/sensors/BMX280Sensor.h#L121
If we can be sure that communication is ok, maybe something there is amiss? My personal question is where does `0xffff` magic come from. [BME Datasheet](https://ae-bst.resource.bosch.com/media/_tech/media/datasheets/BST-BME280-DS002.pdf) only mentiones `0x80000` if value is "skipped" 

You can also try Adafruit libraries. Note, that they have separate ones for BME and BMP:
https://github.com/adafruit/Adafruit_BME280_Library
https://github.com/adafruit/Adafruit_BMP280_Library
They expect ArduinoIDE to build things, their `examples/` dir will auto populate IDE's menu. Or you can manually do a `pio init -b esp01_1m` and copy source there.

davebuk, 2019-04-08:
So I guess the temperature readout is giving 0xffff at some point giving the error? Maybe there is a problem with the chip I have and is outputting the incorrect values?

I'm new to all this building and compiling binaries so trying other libraries is beyond my current knowledge! 😁 I'll leave the firmware running as it is for now while it's working and have another look when I have more time to spend with the system. Thanks.

davebuk, 2019-04-10:
Think I have found the issue. After making my own error message in BaseSensor.h, the error 4 is from the 'Pre-read hook' section of BMX280Sensor.h. My guess is that espurna is looking for another sensor and is trying to get a reading and failing.

I changed the `#define BMX280_NUMBER` in sensor.h from 2 to 1 and compiled a new binary. There are now no more error 4 or 6 messages in the debug section of espurna web UI and the BME280 is working correctly still.

mcspr, 2019-04-10:
Ok, that wasn't like this before. Updated detection & defaults based on your comment: #1690

davebuk, 2019-04-10:
When the code has been added to the dev branch I'll test and let you know. Thanks.

closed 2019-04-11 by xoseperez

davebuk, 2019-04-11:
I'll test this weekend.

PR #1690 (2019-04-11): Update BMX280Sensor
resolves #1672
- default to one sensor
- dynamic number and address setting
- do not return i2c-specific error for wrong values
(unless it is some value returned by Wire.read() and not sensor?)