xoseperez/espurna #1675: Enable NoFUSS by default

skorokithakis, 2019-04-03:
Is there a reason that NoFUSS is disabled by default (maybe most devices cannot be updated?).

I have many, many devices around the house, and it would be great if they could autoupdate themselves with NoFUSS. It's currently a bit hard to install NoFUSS due to the many dependencies, but I could implement support for the protocol in my [espota-server](https://gitlab.com/stavros/espota-server), which is a single Rust binary.

Would that be useful? I'd very much like it if I could just define a URL and have all devices update whenever I dropped a binary in.

schweini, 2019-04-05:
Just FYI: I am implementing a NOFUSS 'server' in a perl CGI right now. It really seems quite trivial to get the basic functionality working. No idea why the official server has so many dependencies.

You get some parameters in the HTTP headers (or ENV, if running under CGI), like the MAC address, sensor type and running version.

If you want to update (in my case, if there is a file at `firmwares/AB:CD:EF:GH:12:34/update.bin` ), you simply return a JSON with a field pointing to that file, and you are kind of done. I simply move the update.bin file to another filename after it has been sent, in order not to get stuck in an update loop.

NOFUSS also seems to send a parameter "CHIPSIZE", but my SONOFF POW R2 sends the value of 4194304, which seems like 4MB, and hence excessive. I'd love to have some way of knowing if the update succeeded,and if there's enough space free on the device.

skorokithakis, 2019-04-05:
What is the device type for you? The way I do it with espota-server is I have files called `sonoff-basic-0.4.3.bin` which gets served if the device is under 0.4.3. Then I don't need to move the files at all, devices just automatically do the right thing depending on their type. I just don't know what they report as their type yet.

@xoseperez would you enable NOFUSS by default if the server were easy to deploy? I'm wondering what the reason to have it disabled is.

schweini, 2019-04-09:
Mi problem with versioning is that I re-compile quite often, with different
build options, so 'simple' versioning' doesn't always cut it.

But my point was that programming a NOFUSS 'server' from scratch should
really not be considered a hindrance, because it is so simple.

skorokithakis, 2019-04-12:
Ah, yes, certainly. I've added support to my server. It would be great if NoFUSS supported a header that would tell you the free space remaining, so you could at least throw an error if the new binary is larger.

skorokithakis, 2019-04-12:
I think the reason NoFUSS is not enabled by default is that it's 12 KB. I wonder if it could be trimmed down a bit and enabled by default, as it seems very very useful.

mcspr, 2019-05-06:
> I think the reason NoFUSS is not enabled by default is that it's 12 KB. I wonder if it could be trimmed down a bit and enabled by default, as it seems very very useful.

Right. The issue of code size IMO is mainly from using Core's HTTP client library and things that come with it. Most of networking is done via asynctcp library. No HTTP client code though, besides simulating it via raw text (thingspeak, influxdb)

Anothing thing I've recently "remembered" about MQTT/mDNS discovery - you don't even have to have hard-coded nofuss server value. Device can query something like _nofuss._tcp to find it at boot / initial setup phase / via some external trigger / etc.

BTW, I do see that NoFUSS is now supported by the espota-server: https://gitlab.com/stavros/espota-server/commit/9d1b2393ebfc0020e45cf5399368303b03da7554

skorokithakis, 2019-05-06:
@mcspr Yes, I've added NoFUSS support to it, as I said above. It's very convenient when you have a house full of devices, just drop the new firmware in a directory and they all auto-update.

mcspr, 2019-05-06:
Oops. Yes. Right.

OTA mechanism using async libraries is already there, removing esp8266otaserver dependency:
https://github.com/xoseperez/espurna/blob/dev/code/espurna/ota.ino

Only part missing is to add nofuss client part, which will remove nofuss httpclient dependency.

skorokithakis, 2019-06-08:
@xoseperez Is it at least possible to do a separate build/release with NoFUSS enabled? It's a very useful feature when one has a lot of devices...

skorokithakis, 2019-06-08:
Oh, I think I disabled a bunch of stuff for my custom NoFUSS builds, so maybe the separate build would still not be under 512 kb... I guess I'll have to make it easier for me to custom-build the firmware.

closed 2019-08-15 by stale[bot]