xoseperez/espurna #1683: Default thing "Description" in UI and heartbeat

PieBru, 2019-04-05:
Hi all,
how about initializing the UI General/Description default value as the MANUFACTURER+DEVICE values already #defined in hardware.h, i.e. "Itead S20" for platform.ini item [env:itead-s20] ?
It would enable easier devices identification in the UI and in MQTT management software such as HASS.
Thanks,
Piero

mcspr, 2019-04-11:
MQTT heartbeat does send `<root>/board` by default with `MANUFACTURER`_`DEVICE`. But you need to subscribe to that topic and then do something with it.

HomeAssistant does use either hostname or description for device naming (entity registry thing):
https://github.com/xoseperez/espurna/blob/f86a3c6ad0e16f1b7d38643f6016c3fb540572bc/code/espurna/homeassistant.ino#L255
Maybe default should be board name instead of hostname?

mcspr, 2019-11-09:
Closing per comment above. Description is a user setting and we will get duplicate with board otherwise.
ofc if `HEARTBEAT_REPORT_BOARD` is off / `hbReport` bit `(1 << 15)` is `0`, then maybe it would be useful.

closed 2019-11-09 by mcspr

PieBru, 2019-11-09:
Thanks @mcspr, sorry I forgot this issue :(
My proposal was about initializing the UI General/Description default value, now it's empty.
Thank you,
Piero

mcspr, 2019-11-09:
Hm. I guess we can send `boardName` when `desc` is not available (settings key values).
But with current webui approach, it will then save this setting to the flash as new `desc`

PieBru, 2019-11-09:
OK @mcspr , it was only aesthetics and as you pointed out it may cause more problems than the "blank field anxiety" it solves :)))
Don't worry, thanks.
Piero