xoseperez/espurna #1686: sonoff pow ota update

theonedemon, 2019-04-06:
**Bug description**
upgrade from 1.13.4 to 1.13.5 via web ( admin Upgrade ) failed in 40% upload ( progress bar )
device restared ( i hear relay click  ) and device restarted ( uptime 5sec  )

**Steps to reproduce**
open device web interface in browser
open tab *Admin* chose espurna-1.13.5-itead-sonoff-pow.bin and click *upgrade* buttton

**Expected behavior**
firmware is updated

**Screenshots**
no screen device not upgradable 

**Device information**
```
[485725] [WEBSOCKET] Requested action: dbgcmd
[485728] 

---8<-------

[485730] [MAIN] ESPURNA 1.13.4 (2aa5c59)
[485731] [MAIN] xose.perez@gmail.com
[485734] [MAIN] http://tinkerman.cat

[485735] [MAIN] CPU chip ID: 0xA03951
[485737] [MAIN] CPU frequency: 80 MHz
[485739] [MAIN] SDK version: 1.5.3(aec24ac9)
[485745] [MAIN] Core version: 2.3.0
[485749] [MAIN] Core revision: 159542381
[485753] 
[485757] [MAIN] Flash chip ID: 0x1640E0
[485762] [MAIN] Flash speed: 40000000 Hz
[485766] [MAIN] Flash mode: DOUT
[485769] 
[485771] [MAIN] Flash size (CHIP)   :  4194304 bytes / 1024 sectors (   0 to 1023)
[485777] [MAIN] Flash size (SDK)    :  1048576 bytes /  256 sectors (   0 to  255)
[485781] [MAIN] Reserved            :     4096 bytes /    1 sectors (   0 to    0)
[485785] [MAIN] Firmware size       :   491216 bytes /  120 sectors (   1 to  120)
[485793] [MAIN] Max OTA size        :   532480 bytes /  130 sectors ( 121 to  250)
[485800] [MAIN] EEPROM size         :     4096 bytes /    1 sectors ( 251 to  251)
[485807] [MAIN] Reserved            :    16384 bytes /    4 sectors ( 252 to  255)
[485814] 
[485816] [MAIN] EEPROM sectors: 251, 250
[485819] [MAIN] EEPROM current: 251
[485821] 
[485822] [MAIN] EEPROM:  4096 bytes initially |  1636 bytes used (39%) |  2460 bytes free (60%)
[485831] [MAIN] Heap  : 36264 bytes initially | 19672 bytes used (54%) | 16592 bytes free (45%)
[485839] [MAIN] Stack :  4096 bytes initially |  1508 bytes used (36%) |  2588 bytes free (63%)
[485849] 
[485853] [MAIN] Boot version: 31
[485856] [MAIN] Boot mode: 1
[485860] [MAIN] Last reset reason: Exception
[485863] [MAIN] Last reset info: Fatal exception:29 flag:2 (EXCEPTION) epc1:0x4021f413 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000
[485871] 
[485875] [MAIN] Board: ITEAD_SONOFF_POW
[485879] [MAIN] Support: ALEXA API BROKER BUTTON DEBUG_SERIAL DEBUG_TELNET DEBUG_WEB DOMOTICZ HOMEASSISTANT LED MDNS_SERVER MQTT NTP SCHEDULER SENSOR TELNET TERMINAL THINGSPEAK WEB 
[485892] [MAIN] Sensors: HLW8012 
[485893] [MAIN] WebUI image: SENSOR
[485897] 
[485899] [MAIN] Firmware MD5: 40418b33f66f515966c7024165850292
[485904] [MAIN] Power: 3134 mV
[485906] 

---8<-------

[485907] +OK
```

coatbrigand, 2019-04-06:
I had the same thing on various devices. I found I could update over the air from the Arduino IDE.
1.13.4 is broken.

mcspr, 2019-04-10:
#1574 is tangentially about it. https://github.com/xoseperez/espurna/commit/92e10943bb88ccc331c5af39f6504c799257819e fixes the webui bug, so 1.13.5 should be ok. And 1.13.4 release .bins were deleted to avoid this problem :)

closing as resolved

closed 2019-04-10 by mcspr