xoseperez/espurna #1687: Tuen relay OFF before OTA firmware flash?

schweini, 2019-04-06:
OTA firmware flashes (using the web interface) quite often, and seemingly at random, hang at around 80% upload progress, on my SONOFF BASICs and POW and POW R2 units.

I can't be 100% sure, but to me it seems that turning the relay OFF before sending the OTA incresses the chances of the upload working considerably. 
If true, I am guessing this is a power supply issue?

If anyone can replicate this behaviour, I would recomend that we should turn the relay off automatcally before accepting an OTA flash.

mcspr, 2019-06-06:
If you do remember the circumstances, which OTA method did you use? WebUI, ArduinoIDE upload / espota.py or using curl http://.../update?
I can think of only two reasons for that.
1. too low loopDelay value that hardware-resets the device, which is changed back to 10ms from the original 1ms (1.13.5). webui is likely to fail with this, but i think it should work with current -dev version
2. interrupt frequency depending on load from [hlw8012 library](https://tinkerman.cat/post/hlw8012-ic-new-sonoff-pow)? so we may need to disable them at the start of upgrade. using detachInterrupt, because just turning off all isr (nointerrupts()) will turn them back on when first flash write competes. but only relevant to cores <2.5.0 - https://github.com/esp8266/Arduino/commit/9515f466846f665fa68a7169465ee89aaa97b67e#diff-2162e3da48a10d5899c93c262576c28e
(ping @xoseperez)

schweini, 2019-06-06:
>    esp8266/Arduino@9515f46#diff-2162e3da48a10d5899c93c262576c28e

This happened quite often using webUI OTA.
 I saw the loopDelay in the changelog, and have recompiled, and will keep
an eye on it when I'm updating the units.

On the other hand - why would the relay's state influence the loopDelay
thing?

My observation that this might have something to do with the relay's state
wasn't very scientific, to be honest. But it sure seems to have helped
(i.e.: OTA failed 3 times with relay on, but succeeded on the the first try
after turning it off).

mcspr, 2019-06-06:
Maybe it is not related.
Basic was behaving pretty bad in general with low loopDelay values. Not really sure is that is related to power problems, wireless performance issues or something else.
POW & POW r2 still send out sensor data via debug log, that's another possibility of somewhat slowing things down to the point of reset. `info` and `crash` will show recent reset reason and crash data.

closed 2019-08-12 by stale[bot]