xoseperez/espurna #1688: espurna-nodemcu-pzem004t binary

rradar, 2019-04-08:
Hello!

Just to have it clear: this one here is sibling of feature request #1583 :grin: 

I wonder if it would be possible - as there are already a couple of custom binaries (more than 100?) and some for the nodemcu:

- espurna-1.13.5-nodemcu-geiger.bin
- espurna-1.13.5-nodemcu-lolin.bin

to release one with support for the pzem004t included.

I know it's possible because this sensor is supported. And it fit's pretty perfectly together :1st_place_medal: 

[image]

Thank's a lot!

forast3r, 2019-04-17:
First of all, sorry for the offtopic... 
@rradar have you modified the PZEM004T to connect to the nodemcu? If so, could you share your mod? Thanks!

rradar, 2019-04-25:
@forast3r no modifications. pzem004t want's 5V serial and esp8266 (as the nodemcu) is 5V tolerant (designed for 3.3V) on the logic pins (the CEO of espressif stated) :rocket:

So just a espurna binary is missing for success! :1st_place_medal:

mcspr, 2019-04-25:
Minimal config to make it work with HW serial (aka RX TX pins)
2021e6c6c6b4735ebb1cab0e8e81ba95d131a2f4

Maybe something like this would be nice (i.e runtime settings. see getSetting(...) keys)
d5300c42e27ff1f9d3079dc174c3b08a3aa31137
To support both SW and HW serial. Not yet handling swapped pins - 13 & 15 can be used when program calls `Serial.swap()`. Or whatever GPIOs, using SoftwareSerial. Not really sure if NodeMCU board will read any serial data via HW serial, because RX & TX are connected to the USB converter, so that may be a hard requirement.

rradar, 2019-04-26:
Just to be clear (once again). The picture above in the post is not my setting!

But I'm driving a setup since yesterday with a wemos d1 mini and software serial (D1&D2) and using the 5V rail to the pzem004t. The wemos is powerd via usb. That's it and it works. I don't use espurna for this right now because there is no binary. Instead I use Tasmota 6.5 right now..

mcspr, 2019-04-28:
Despine naming, it'd still be compatible. My main point is that some level runtime configuration is required for it to be useful (like it is useful to have SoftwareSerial support built in, so you don't have to use HW serial pins)

I'll add the proposed patch then. Nightly builds will fire up either today to tomorrow, depending how fast I do it.

closed 2019-05-01 by mcspr

rradar, 2019-05-01:
Very nice @mcspr thank's a lot for your work! I will test it and report my results. Could you just point me to the binary - I'm blind and can't locate it here https://github.com/mcspr/espurna-nightly-builder/releases :confused:

mcspr, 2019-05-07:
@rradar any update? 👀

rradar, 2019-05-09:
Sorry, can't give any update right now because I don't have access to the device. Will try asap and give feedback (nodemcu is already flashed :wink:)

rradar, 2019-05-18:
hey @mcspr
I just made a quick test with the image you provided. I only tried the hardware serial (as a software serial is not yet included?) without luck.

I switched RX - TX to be also sure there is no miss match but I still/always get this error in the debug panel:

`-> [SENSOR] Error reading data from PZEM004T @ HwSerial (error: 3)`

Anything I could do/test/debug?

The exact same hardware (pzem004t) works with a wemos d1 mini and tasmota (v.6.x) and software serial so the unit has no defect as I can tell

mcspr, 2019-05-18:
@rradar SW serial is supported in that build.

ESPurna settings are backed by key value storage. If you look at PR changes, you can bunch of getSetting(...) calls with keys and DEFAULT HEADER VALUES that supposed to configure PZEM. We need 3: `pzemSoft`, `pzemRX`, `pzemTX` and we can use [terminal commands](https://github.com/xoseperez/espurna/wiki/Terminal) `get <key>` and `set <key> <value>` to do that.

`pzemSoft` needs to be `1`, set RX and TX accordingly.
This can be in WebUI, but some other things need to be modified first :)

Regarding HW serial: I suspect this is nodemcu problem and we can't use either RX or TX properly (can't remember which 🤔) because it is connected to the USB UART chip onboard.

rradar, 2019-05-18:
Wow, that sounds complicated @mcspr :D

I'm not blessed with skills to debug in here anyfurther without some step-by-step tutorial. Sorry for that!

It could be nodemcu problem with usb uart chip. But it also would apply for the wemos d1 mini (even shares the same uart chip I think). I also didn't had luck in the first run with the d1 mini, tasmota and hardware serial so I switched over to the software version on pins d1 and d2.

Would it possible to create another build which has software serial activated? Or/and another for the d1 mini (to cross check things)?

mcspr, 2019-05-18:
@rradar I outline again - **SoftwareSerial** mode is supported 😉 It's OFF by default, that's all.

I assume RX is D1 (GPIO5) and TX is D2 (GPIO4).
Debug panel has a terminal input at the top, where the "Send" button is. Just type there and press enter / press "Send":
`set pzemSoft 1`
`set pzemRX 5`
`set pzemTX 4`
`reset`

Wait until it reboots. Now, on the Status page, sensor values must have `
PZEM004T @ SwSerial(5,4)(0/192.168.1.1)` labels to the right of them

rradar, 2019-05-18:
@mcspr sorry I was blind. Thank's for having patience with me :D

And I can tell, **it works** with software serial! :fireworks:

I chanced my wires to D1 and D2. It didn't worked with your example because I switched RX and TX... as I was to lazy to rewire I just went with the debug panel again and entered:

```
set pzemRX 4
set pzemTX 5
reset
```

and profit!

I would really like to see the same binary for a wemos d1. But I just will go with the nodemcu for now!
Thank's for your effort and for the upcoming release! :+1:

PR #1712 (2019-05-01): PZEM004T: settings & dev board
resolve #1688 

- platformio `env:nodemcu-pzem004t`
- port settings setup from v2

No support for swapped `Serial`, just SoftwareSerial. PZEM library does `Serial.begin()` right in the constructor, so `Serial.swap()` call can go right after local sensor's `.begin()` if HW port is used.

SoftwareSerial support is always built in. PZEM library requires global define `PZEM004T_NO_SWSERIAL` an sensor here needs to conditionally depend on it, because it will strip tx+rx constructor.
545856 - 541896 = **3960** bytes of difference between SWS and no-SWS versions.
Not ad [... cut]