xoseperez/espurna #1696: Print mDNS support on the Status page

goatchurchprime, 2019-04-15:
In issue #129 it says that mDNS is available, but is turned off by default; you have to build it from source if you want it.  It took us quite a few hours to discover this feature.

Now we have the new binaries compiled and are updating them in our sonoff devices where we can find them, it is impossible to tell which is which.  

Can we have a line
 
> **mDNS support:** [yes/no]

on the web Status page please?  Then it will be quicker for people to discover that this is a feature.

mcspr, 2019-04-26:
Just adding mDNS support: yes is a tricky thing here
when MDNS_SERVER=1, it will query _mqtt._tcp service and update settings (this is the default)
when MDNS_CLIENT=1, .local resolution will work (this is optional)

_SERVER also does device's own service announcements: _telnet._tcp, _http._tcp, _arduino._tcp (OTA discovery)
Terminal `info` banner has a string listing those define tokens, so that can be reused. But it should probably name things more clearly.

mcspr, 2019-11-09:
*Update:*

Using current `dev` branch with Core version 2.6.0 and up will allow to query .local domains without any setting. ATM platformio is not yet updated to the 2.6.0, but one can use `platform = ${common.arduino_core_git}`.
Note that this depends on LWIP2 option to be enabled, which is also the current default.
ref: https://github.com/xoseperez/espurna/pull/1907/files#diff-63bb1c277c0b8dc961f6f4c48bf19c23R87

closed 2019-11-09 by mcspr