xoseperez/espurna #1697: Disable the switch option

goatchurchprime, 2019-04-15:
We use the Sonoff POWs to measure power on devices **that we never want to turn off**.  

Can we have an option in the Switches menu that is "I really mean Always ON!" instead of just being on at boot mode.  I don't want it ever to go off if someone taps the switch or sends an MQTT relay/off message to this device.  

We already have other ways of turning our machines off, like kill switches and a separate power unit that turns a whole lot of other machines off at the same time.  

Having this easy off switch in the Sonoff is a liability.  But I want it to be easy to read the power and do the other config features.  

Adding this lockin feature into the Boot mode option would be the minimum to achieve this.  But you could add a new thing on the API that locks or unlocks the switch remotely.  

But the really main thing is we'd like to easily disable that button as a switch.

mcspr, 2019-04-26:
> But the really main thing is we'd like to easily disable that button as a switch.

I remember button control was proposed as https://github.com/xoseperez/espurna/pull/1322
But i think this should be resolved via [v2 branch](https://github.com/xoseperez/espurna/tree/v2) / current #1680 , since we will be able to remove relayID<->button association at runtime

xoseperez, 2019-05-22:
Even after removing the association between buttons (physical or virtual) and relays there will be ways to toggle the relay (terminal, MQTT, API...). For me the proposed lock option in #1705 makes sense...

markusrudolf, 2019-06-13:
I also want to have an option to disable the physical switch via web interface since our triplets are starting to exploring our flat and pressing all buttons that are in range :-)

PR #1705 (2019-08-12): Lock relay status on boot
resolves #1697 

- `relay_t::lock` flag to control which target status we want. after relay changes to that status, we can no longer change it
- boot mode setting to lock relay in either OFF or ON state

cc @goatchurchprime @xoseperez 
this *might* interfere with planned motor changes from #1510 ?
closed 2019-08-12 by mcspr