xoseperez/espurna #1704: Sonoff Dual R2: read buttons by MQTT not working

skyynet, 2019-04-21:
I finally managed to install a MQTT Broker on my Synology and the relays report just fine.

On the Sonoff Dual R2 there are headers for two additional buttons named Button0 and Button1. When I connected one of them with GND I got the information via MQTT but also one relay toggled.

I changed the configuration in hardware.h to
`#define BUTTON1_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
#define BUTTON2_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH`

and in defaults.h to
`#ifndef BUTTON1_PRESS
#define BUTTON1_PRESS       BUTTON_MODE_NONE
#endif
#ifndef BUTTON2_PRESS
#define BUTTON2_PRESS       BUTTON_MODE_NONE
#endif`

Now connecting the headers don't switch the relays but also don't send information via MQTT. Where is my misconception?

mcspr, 2019-04-22:
You can set relay IDs here to 0. Button event will still go through, if the goal is to avoid changing relay state
https://github.com/xoseperez/espurna/blob/8e95632daf3ec7c88079c9e577e1b331a35c7146/code/espurna/config/hardware.h#L496-L497

skyynet, 2019-04-25:
Thanks a lot, @mcspr 
This solved my question and now I'm able to make use of the two buttons.

I'll add two switches to get information about my robot mower's garage which is already opened and closed by Sonoff Dual R2. With two button stats I can check if the mower is inside and if the garage door is closed to further automate opening and closing of the garage door.

You made my day!

closed 2019-04-25 by skyynet