xoseperez/espurna #1706: Thingspeak Sonoff Pow R2

serestre, 2019-04-26:
Hola, agradecería mucho algo de ayuda

No he podido leer la información del Sonoff Pow R2 en Thingspeak. Tengo la siguiente configuración en el canal de Thingspeak:

[image]
[image]

----------------------------------------------------------------------------------------------------------
El Sonoff esta flasheado con ESPURNA 1.13.5

[image]
----------------------------------------------------------------------------------------------------------
La configuración del sensor es la siguiente

[image]
----------------------------------------------------------------------------------------------------------
asi esta configurada la conexion a Thingspeak

[image]
----------------------------------------------------------------------------------------------------------
Debug Log

[image]
----------------------------------------------------------------------------------------------------------

serestre, 2019-04-30:
Solucionado, el problema estaba en la configuracion que habia establecido en la red WiFi. No conozco mucho ese mundo del router

Inicialmente lo había configurado con una IP fija. 
Luego le quite todo eso y ya envía datos a ThingSpeak

Antes & Despues
[image]

mcspr, 2019-05-01:
No hablo español, sólo para que lo sepa. 🤐

So you changed from static IP config to DHCP, and now it works? You can close the issue if that is correct.

closed 2019-05-01 by serestre