xoseperez/espurna #1707: Magic Home LED Controler Board v2.3 WAS CORRECT!

lazylazyllama, 2019-04-26:
After issue #512 the magic home v2.3 image was removed... 

But in my case this image had the correct configuration.

I have a magic home device and the 2.0 image is not working very well.
Red and green are switchted....

But now I flashed the very old 2.3 image and everything works perfectly...

So I would really like if you could re-add this image.. so that I can flash the current espurna version with the configuration I need.

lazylazyllama, 2019-06-26:
:(

mcspr, 2019-06-27:
Looking at https://github.com/xoseperez/espurna/commit/53469c1738cb3a20a991969aa070718c3c945f27#diff-64ccb24d1cc0509f64794258b04da3ecL917
There is an almost exact copy named ARILUX_AL_LC02 / arilux-al-lc02
https://github.com/xoseperez/espurna/blob/e445cf8c7003827680d63f33b391bc95faec5de3/code/espurna/config/hardware.h#L1635-L1654
which you i think you can use if don't care about the ir and led present :)

Given that there are a lot of hardware for these devices, dynamic configuration led / buttons / relays / channel pins etc. makes more sense as a solution, see [tree/v2](https://github.com/xoseperez/espurna/tree/v2) / #1680. Most of the backlog was cleared out, so that is something to finally merge into the tree instead of leaving it as a half-done pull request.

closed 2019-09-02 by stale[bot]

larianum, 2020-10-20:
@lazylazyllama Hey. have you managed to fix the issue? I have similar problem - some version before 1.13 was working just fine (all channels were right), but after update they are shifted on +1 (channel #0 should be Red, but it's empty, or white - not sure, don't have white, channel #1 should be green, but it is Red, channel #2 should be Blue, but it's Green. and Blue is on channel #3, which must be used as white, or 'alpha' I don't know). tried to build without white (3 channels), turn off 'use white' , not helped. I can't use my remote, because it is using first three channels (0-2 index) for colors. Maybe I'm missing something.

lazylazyllama, 2020-10-20:
> @lazylazyllama Hey. have you managed to fix the issue? I have similar problem - some version before 1.13 was working just fine (all channels were right), but after update they are shifted on +1 [...]

Sorry, but I don't use espurna anymore.

mcspr, 2020-10-21:
Repeating my previous comment, the different light+ir binaries simply replace the default settings for channel<->pins, which can be changed at runtime as well to fix slightly different boards:
https://github.com/xoseperez/espurna/wiki/Lights#dimmer

If you have any ideas for the correct name and actually have the board in question (I don't) so we don't guess the configuration based on the old comments, please submit a PR

larianum, 2020-10-21:
@mcspr yes, I have this board. but not sure if I build it right when changing pins in config.

mcspr, 2020-10-21:
Can you share the config you are using?

larianum, 2020-10-21:
@mcspr I build it with this command: **env ESPURNA_IP=<ipaddress of device> ESPURNA_AUTH=<device password> platformio run -e magichome-led-controller -t upload**
instead of magichome-led-controller I tried to use different devices, like arilux, but still no luck.

pin config defined in hardware.h:
    ```
// Light
    #define LIGHT_CHANNELS      4
    #define LIGHT_CH2_PIN       14      // RED 14
    #define LIGHT_CH3_PIN       5       // GREEN 5
    #define LIGHT_CH4_PIN       12      // BLUE 12
    #define LIGHT_CH1_PIN       13      // WHITE 13
```

Then I tried this:
```
// Light
    #define LIGHT_CHANNELS      3
    #define LIGHT_CH1_PIN       14      // RED
    #define LIGHT_CH2_PIN       5       // GREEN
    #define LIGHT_CH3_PIN       12      // BLUE
```

and as well I tried to change pin numbers in these configs, and still no luck. 
I can't force firmware to use channel #0 as color channel.

mcspr, 2020-10-21:
Have you tried these pins instead?
```cpp
    #define LIGHT_CHANNELS      4
    #define LIGHT_CH1_PIN       12      // RED
    #define LIGHT_CH2_PIN       5       // GREEN
    #define LIGHT_CH3_PIN       13      // BLUE
    #define LIGHT_CH4_PIN       15      // WHITE
```

larianum, 2020-10-21:
@mcspr Will try tomorrow and provide feedback. Can you, please, confirm that I use build command correctly? or I should add some params to apply these changes?

mcspr, 2020-10-21:
No issue there, magichome-led-controller env uses `-DMAGICHOME_LED_CONTROLLER`, so make sure you are modifying the correct entry in the hardware.h

Note that I just used the `MAGICHOME_LED_CONTROLLER_20` pins, but swapped the 1st and the 2nd channel pins.

larianum, 2020-10-22:
@mcspr tried your config, but it won't work. tried to swap red and green to check if my config changes are applied at all - and nothing changed. so the issue is that my changes are ignored. is there a chance that these changes are reverted before build (some kind of git feature)?

mcspr, 2020-10-22:
Have you tried doing `set ltDimmerGPIO0 12`, `set ltDimmerGPIO1 5` ... (build ch1 is gpio0 setting, ch2 is gpio1 and etc.) and then rebooting?

If `git diff` still shows your changes, those are still there on the filesystem. You can try adding build flags in the command as well,  `env ESPURNA_FLAGS="-DLIGHT_CH1_PIN=12`, it will also pin-point the exact line it uses in the header files

larianum, 2020-10-22:
@mcspr sending commands manually helped. So seems that these changes are ignored on build phase (stashed, then revert, build, and get back from stash or smth like that)