xoseperez/espurna #1709: Unknow local plugs

darkoddio, 2019-04-29:
Last year I bought couple of these plugs from a local vendor that developed their own plug/cloud.
Now these plugs doesn't work anymore with the original firmware and cloud.
Do you think I can try to flash them with Espurna? 
In case I try it, which release I need to use?
Is it fine to use the 4 top right contacts (3V3, TX, RG, GDN) that you see in the second image?
Thanks in advance

[image]
[image]

mcspr, 2019-05-04:
Should be ok. Even GP**IO0** is already there

darkoddio, 2019-05-06:
Not really sure about your statement and GPIO0
Is there a build ready that I can use to flash it or do I need to build a custom one?

ColinShorts, 2019-05-06:
GPIO0 needs to be grounded in order to put the device in the programming mode.

You'll need to identify which gpio pin is attached to the relay and any indicator led's, you might need to check whether or not those pins need to be inverted.

You might get lucky and find an existing binary that will work, or you can do some trial and error if you want to compile your own images. Perhaps the best method would to be using the gpio command in the terminal (see https://github.com/xoseperez/espurna/wiki/Terminal), or over serial //IF// it is safe to do so - in most cases it is not safe to have the device powered by mains and be connected via serial.

darkoddio, 2019-05-25:
thanks for clarifying
I managed to flash it, started with a generic relay espurna-1.13.5-generic-esp01s-relay-40.bin
but the relay does not respond, neither pressing the button on the plug.
How to find the correct relay and how to change its value?
Also power meter doesn't work obviously.
Any suggestion will be appreciated :)

darkoddio, 2019-05-31:
hi all, any advise? how to find which GPIO are the power meter and the button?

closed 2019-08-07 by stale[bot]

mcspr, 2019-08-07:
Sadly, can't find any info on the name Freelux 16a besides this:
http://www.freelux.com.vn/page/smartplug

I know it's may be late for this... but looking at the board, it looks like esp12f:
https://fccid.io/2AHMR-ESP12F/User-Manual/User-Manual-3564272
ref "figure 7" with the pinout
And on the very first photo we can see that GPIO0, GPIO2 and GPIO4 are soldered on top (bottom row). I would expect IO4 to be the relay switch there, IO2 as LED and IO0 as external button (or just for the flash purposes)
On the second photo, 4 pins are supposed to be used by the SPI flash interface and are seemingly going to the secondary board via 10pin header which we might want to look underneath of.
IDK why `gpio` command would not work though. e.g. `gpio 2 1` or `gpio 2 0` should light up the built in led on wemos d1 clone boards that use esp12f module

darkoddio, 2019-08-07:
thanks for the follow-up
Once back from vacation I'll try again those gpio commands and let you know.
After the info you found, do you think there is a way to have a working espurna on it (and how)?

mcspr, 2019-08-07:
To switch things, likely. Still no idea about the power meter chip, which is somewhere on the other board. 

It would need proper pin numbers for RELAY1_PIN & BUTTON1_PIN
For example, GENERIC_ESP01S_RELAY_V40 uses io0 for relay on / off and io2 for led (which we guess is a led here too)
https://github.com/xoseperez/espurna/blob/274f71d7d9047bb2bdf6b5c2cca7a6ff9d63142b/code/espurna/config/hardware.h#L2413-L2420