xoseperez/espurna #1710: Sonoff Basic R2 - Button with GPIO 3 (RX Pin)

pabloalcantara, 2019-04-30:
We already Know that the Sonoff Basic R2 does not expose the GPIO 14 and that the GPIO 2 (dificult to solder) can´t be pressed on boot, 
So It´s possible to create a version for the Basic R2 that uses the GPIO 3 (RX pin) as a button? It could be an alternative compiled version, or just an option inside the basic version to select the gpio (14 for the original Basic, 3 for the R2).
I have compiled using the base of the Basic just changing the gpio from 14 to 3 and is working perfectly.
Thanks

mcspr, 2019-05-01:
IMO right now it is better to do the second option, "select the gpio", by using this approach: #1680 
You can try building from that PR and see how it goes. `btnGPIO0` setting will control the pin, configured at boot (and not exposed in UI yet).

If you have any idea about where to put configuration presets (generated from hardware.h) to change from WebUI / Terminal, please share :)

pabloalcantara, 2019-05-01:
The option tô select is the best option. Because we can use just one firmware.

davebuk, 2019-05-16:
[image]
I have found that the soldered through holes, mainly 'K_P' are connected to GPIO2. This gives an easier way of attaching wires or header pins WRT ground. You still have the boot failure if held low at start so no good for a toggle switch, but push button works fine. As stated above you also need to set GPIO 2 rather than 14.

davebuk, 2019-06-12:
The article here: https://www.forward.com.au/pfod/ESP8266/GPIOpins/index.html
describes a way of using GPIO 2 with GPIO 0 and a switch and does not effect boot as it sets GPIO 0 as an output after boot.

Can this, or does this option appear in espurna for the basic R2?