xoseperez/espurna #1713: Wiki: Flash Tuya based device without opening/soldering

hyteoo, 2019-05-03:
In my search for better and cheaper devices on amazon, I came across smart sockets that can not be opened without destroying the device.
They are all using the **Smart Life** app made by Tuya but distributed under different brands

There is a project that fakes the OTA of the original firmware and lets you flash your own custom firmware: https://github.com/ct-Open-Source/tuya-convert

Tested it on 2 types of devices, flashed espurna directly and works like a charm!

Maybe it would be a good idea to reference this in the Wiki

AndreasSchutz, 2019-05-07:
What firmware version/environment did you use? I've been struggling to find a firmware that doesn't make the tuya module stack dump continuously.

hyteoo, 2019-05-07:
Latest master branch / PlatformIO. I just created a new device without any buttons/relays/leds defined.
Just to have espurna on the device and then OTA trial and error until I got the pins right.

Tuya convert was installed on a Pi3.

jsponz, 2019-05-07:
@hyteoo Which devices you tested?

AndreasSchutz, 2019-05-07:
Okey, thanks for getting back to me... sounds so easy when you summon it up like that! ;-) 
I just got [one of theese.](https://www.aliexpress.com/item/Wireless-Wifi-Smart-Home-Socket-Switch-Remote-Control-Socket-WiFi-Smart-work-with-Amazon-Alexa-Google/32961722743.html?spm=a2g0s.9042311.0.0.d1b04c4dWk8vMV), opened it up and soldered the connections for flashing. The ESP8266ex module inside was similar [to this](https://github.com/arendst/Sonoff-Tasmota/wiki/Hyleton-313-Smart-Plug).
Do you mind sharing your button/relay settings so that I can compare it to my findings?

At first I had some success but it was unstable, then I did a factory reset from serial and after that all I get are stack dumps.

AndreasSchutz, 2019-05-09:
@mcspr
All my "bugs" went away when i reassembled the device and gave it decent power. I also had a problem where the device couldn't do OTA and only downloaded half the file before crashing and that went away too. Right amount of juice really is crucial, now it is running like a charm!

hyteoo, 2019-05-09:
@AndreasSchutz  glad you figured that out.
@jsponz tested devices were these:
https://www.amazon.de/gp/product/B07GF2XLMN
https://www.amazon.de/gp/product/B07CDCYLQ6

closed 2019-07-15 by stale[bot]

mcspr, 2019-07-15:
I tagged another issue in the https://github.com/xoseperez/espurna/pull/1774, this is the relevant one.

https://github.com/xoseperez/espurna/wiki/OTA-TUYA was added
https://github.com/xoseperez/espurna/wiki/Hardware-JINVOO-VALVE-SM-AW713 has a mention that it can use such method
Sidebar now has a link near the SonOTA