xoseperez/espurna #1717: espurna MQTT use with thingsboard

PirSoft, 2019-05-15:
hi dear;
i would like use espurna with thingsboard, but not get telemetry date from espurna. 
connection is ok. but cat't send telemetry data to thingsboard.
**settings:**
MQTT Root Topic : v1/devices/me/telemetry

**DEBUG**
[557691] [MQTT] Sending v1/devices/me/telemetry/data => {"temperature":26,"humidity":51,"time":"2019-05-15 23:31:30","mac":"18:FE:34:9D:59:75","host":"ESPURNA-9D5975","ip":"192.168.0.134","id":8758} (PID 1)

what is problem.? 
thanks.

mcspr, 2019-05-16:
Based on [Thingsboard documentation](https://thingsboard.io/docs/reference/mqtt-api/#telemetry-upload-api), ` v1/devices/me/telemetry` is the topic ESPurna needs to send data to. But, JSON option appends `/data` to the root topic, as you can see in the debug log. Thingsboard is not subscribed to that, so it cannot see the data.

My guess is there must be an option in thingsboard that allows to use different topic.

PirSoft, 2019-05-16:
have you any solutions? need espure firmware modification?
thanks.

mcspr, 2019-05-16:
Like I said. If thingsboard allows it - configure a different topic for telemetry data. Or we can make espurna JSON topic configurable instead

PirSoft, 2019-05-16:
if possible pls you can make espurna JSON topic configurable option. 
thanks.

mcspr, 2019-05-16:
Sure, but how configurable exactly?
Right now only `data` part is configurable via build time settings. Instead, JSON topic setting can be a full topic, with optional placeholder values:
`{root}/data` (default)
`data/{hostname}`
`just/json/data/no/placeholders`

In this case this seems more appropriate than thinking about how root topic and `data` combine (or whatever it is set to)

PirSoft, 2019-05-17:
I think JSON  /data should be selectable. or no placeholders.  if settings option for JSON format change by user. Espurna MQTT suport will bey generic useful..
thanks.

mcspr, 2019-05-18:
If you like, you can try building the firmware yourself from this tree and then change `mqttTopicJson` setting to the one you need:
https://github.com/xoseperez/espurna/compare/dev...mcspr:mqtt/json-topic

I think this *can* go into the 1.13.6 (minor things is to update webui), because that changes internal DEFINES and not something that was previously configurable.

mcspr, 2019-05-18:
also cc @ElderJoy about https://github.com/xoseperez/espurna/issues/1717#issuecomment-493694095, because thermostat module references MQTT_TOPIC_JSON for remote sensor. Is it safe replace that with `+ String('/data')` and detach JSON topic from the base one?
https://github.com/xoseperez/espurna/blob/55b9f2a52f9fc5a225ae4880d15893d15161209f/code/espurna/thermostat.ino#L263

ElderJoy, 2019-05-18:
It is allowed for most cases now, but...
Thermostat use standard mqtt data output to obtain remote sensor data. From this point of view topic "/telemetry" would be also appropriate to obtain needed data. Using just topic "/data' would be limitation that wouldn't allow to obtain remote temperature from devices, that translate temperature using topic "/telemetry". My proposition as for thermostat is to subscribe to all possible topics that can translate standard telemetry data to have possibility use such devices as remote sensors.

closed 2019-09-22 by stale[bot]