xoseperez/espurna #1720: Conflict between Sonoff Pow and Thingspeak

serestre, 2019-05-17:
Hello, I have a conflict between Sonoff Pow R2 and Thingspeak with espurna 1.13.5

When I do not enable them to send information with thingspeak, the equipment reads current, voltage, power, etc. and works well indefinitely.

The problem begins when I enable information to be sent to a Thingspeak; randomly after a while (minutes or hours) one of the two sonoff or both stop sending information to Thingspeak, the red LED and the blue LED go out but when this happens I can not access them through the web interface and they appear in the WiFi network the sonoff with problems.

Some help ?

serestre, 2019-05-17:
I already resolved it. The reading interval of the sensor had it in 1 second ... I passed it to 6 seconds and I have not had any problems

mcspr, 2019-07-07:
Please try with the current `dev`, I think we fixed in the #1806

closed 2019-07-07 by mcspr