xoseperez/espurna #1722: RF bridge raw commands from Telnet

vvrhf, 2019-05-18:
Good day

1. Could You pls explain, is there a way to send raw RF commands from Telnet (without  MQTT). 
If Yes, could You give an example of syntax of that commands  
2. I read the RF_RAW option was removed in March, should I enable any other option to use RAW commands ? 

Thanks

mcspr, 2019-05-18:
Hey,

No such command yet. But something can be added, since we are already have hex data parser for MQTT raw data
(and probably increasing terminal buffer for RFBRIDGE builds, so we can fit large commands)

RF_RAW option was removed by the #1693 (but only 10 days ago), because it is no longer optional and is always included.

vvrhf, 2019-05-19:
Thank You for the answer. 
Is there and could You advise any existing direct (without MQTT) way to send RF codes over a network ?

thanks

mcspr, 2019-05-19:
You can use HTTP API (https://github.com/xoseperez/espurna/wiki/RESTAPI)
```
$ curl -XPUT "http://192.168.1.150/api/rfraw" -d "apikey=<...>&value=AAC003E855"
# if you disable restful api setting, this can be GET instead of PUT
curl "http://192.168.1.150/api/rfraw?apikey=<...>&value=AAC003E855"
```

mcspr, 2020-05-20:
https://github.com/xoseperez/espurna/commit/01f5afff804674890fcd91e4d9f7c0ce65467df0 adds `rfb.write` terminal command for this exact use-case

closed 2020-05-20 by mcspr