xoseperez/espurna #1729: Firmware for new device. WiFi in line dimmer.

davebuk, 2019-05-20:
Hi all. How can I look to build a firmware for the following device:
https://www.amazon.co.uk/OurLeeme-Dimmer-Control-Function-Compatible/dp/B07RHCFKQ2

I have got one coming this week so I won't know what chip setup is inside but will post when I do. I wonder if the Sonoff wall dimmer will work as base code?

davebuk, 2019-05-29:
The device has arrived. Is it as simple as using something like the 'itead-bn-sz01' and changing the GPIO config to suit, or does a separate environment need to be created?

mcspr, 2019-05-29:
Yes. Since you are just testing, anything with LIGHT dimmer would do. Just make sure to use 1MB environment

davebuk, 2019-05-30:
Great, I'll have a go at the weekend. Is there a way to find out the GPIO assignments? Looking at the soldered board, GPIO 0, 14 and 15 are connected. I guess the on board button connects to 'RST' and maybe 0 for BOOT.

Other than that, I also guess 14 or 15 control the two colour LED and the opposite one controls the dimmer output?

mcspr, 2019-05-30:
Use multimeter? Button thing is interesting, because it kind-of does different things - GPIO0 for flash / some SW button control, but RST will reset the chip itself. No software interaction there.
TX RX  are going to the secondary MCU then? (or I am missing some turn)

davebuk, 2019-05-30:
The only measurement I got was 9K ohms from one of the ESP pins to one of the LED legs. Other resistances were around 2M ohm.

I need to work out which of the two header pins are TX and RX respectively, but 3v3 and ground are obvious. I guess the through hole header pin pads would be the ones to use for flashing, or should I take wires directly from the ESP chip Tx/ Rx connections?

Thanks.

mcspr, 2019-05-30:
Either way. I sure does look like tx rx are right there.

About dimming, it looks like manufacturer used Tuya SDK to set up the secondary controller. Tasmota repo already has an example of what it sends / receives:
https://docs.tuya.com/en/mcu/mcu-protocol.html
https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/xdrv_16_tuyadimmer.ino

davebuk, 2019-05-30:
How do the dimming signals work for an espurna LED dimmer. Do they send signals using the Tx/Rx lines to a MCU which then controls the output drivers?

mcspr, 2019-05-30:
What I mean is we don't have LIGHT_PROVIDER for Tuya SDK dimmer variant (yet), if this thing uses serial communication with secondary MCU.
Most common LIGHT_PROVIDER_DIMMER generates PWM signal on gpio pin, which translates into brightness.

davebuk, 2019-05-31:
The two header solder pads that probably are TX/ Rx are not directly connected to the TX/ Rx of the ESP, they are connected to the 8500 chip below. I was unsure how the dimming was controlled using a single GPIO. 

How long would it take to bring serial dimming into espurna? I may go and have a little play with tasmota for this device.

mcspr, 2019-05-31:
It would be useful to know if it is really using that serial protocol.
Something like this is already done for rfbridge. Basic idea that I got from the Tuya docs is that ESP should send out heartbeat messages (3s initially flagged 0x00, 10s normal operation flagged 0x01) when it boots and during normal operation, then the second controller will accept commands / send out current state.

Another thing to do while it is wired is to download the original firmware .bin via [esptool.py](https://github.com/espressif/esptool#read-flash-contents-read_flash) (0x100000 as size).

davebuk, 2019-05-31:
Trying to upload tasmota using pins on the board or pins wired directly to the TYWE3S fails. I don't get communication. Tried with button held at boot, button still held after boot, GPIO 2 to GND at boot and held to GND after boot with no success. The device is not in normal operation so it's doing something different at boot, plus the status LED flashes green after a few seconds then stays very dimly lit green.

I've not used esptool other than the espeasy GUI version to flash. I'll have to look at this over the next few days and see if I can first establish communication then get the current firmware extracted.

mcspr, 2019-05-31:
Is it the same with GPIO0? Not aware of GPIO2 operation, this wiki states that it needs to be high:
https://github.com/esp8266/esp8266-wiki/wiki/Boot-Process#esp-boot-modes

One thing ESP should always do when resetting is output lines like these on GPIO1 (TX), at 74880 speed:

pressing RST button on d1 mini when GPIO0 is pulled LOW
```
 ets Jan  8 2013,rst cause:2, boot mode:(1,7)
```
pressing RST without anything connected
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v8b899c12 < this is eboot from Arduino, after this line things are sw-specific 
~ld
```

davebuk, 2019-05-31:
Sorry, meant GPIO 0 not 2, post above edited. Apologies!

davebuk, 2019-06-03:
I'm am having issues trying to confirm I have connection, but connected to ESP chip Tx/Rx and carrying out the above using putty I get:

```
ets Jan  8 2013,rst cause:2, boot mode:(1,2)

 ets Jan  8 2013,rst cause:2, boot mode:(1,2)

 ets Jan  8 2013,rst cause:2, boot mode:(1,2)
```

 ```
ets Jan  8 2013,rst cause:2, boot mode:(3,2)

load 0x40100000, len 2592, room 16
tail 0
chksum 0xf3
load 0x3ffe8000, len 764, room 8
tail 4
chksum 0x92
load 0x3ffe82fc, len 676, room 4
tail 0
chksum 0x22
csum 0x22

2nd boot version : 1.7(5d6f877)
SPI Speed : 40MHz
SPI Mode : QIO
SPI Flash Size & Map: 8Mbit(512KB+512KB)
jump to run user1 @ 1000

OS SDK ver: 2.0.0(e8c5810) compiled @ Jan 25 2019 14:26:04
phy ver: 1055_9, pp ver: 10.7

rf cal sector: 121
idle_task_hdl : 40107a90,prio:0, stack:384
tim_task_hdl : 40107bd8, prio:2,stack:512
```

mcspr, 2019-06-03:
But it works. Second log is non-Arduino bootloader, logging that it loaded the manufacturer's firmware. Which now can be safely backed up by esptoolpy and something else uploaded instead

davebuk, 2019-06-03:
I have tried

`esptool.py -p COM3 -b 1155200 chip_id` 
and
`esptool.py -p COM3 -b 115200 read_flash 0 0x100000 flash_contents.bin`
after setting GPIO 0 to GND during boot and neither work. Constant time out errors. One time I had, `invalid head of packet 0xff` but can't get esptool.py chip_id to output anything.

mcspr, 2019-06-03:
Does it work without the `-b ...` param?

davebuk, 2019-06-03:
```
c:\Python27\Scripts>esptool.py -p COM3 read_flash 0 0x100000 flash_contents.bin
esptool.py v2.6
Serial port COM3
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header

c:\Python27\Scripts>esptool.py -p COM3 chip_id
esptool.py v2.6
Serial port COM3
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header

c:\Python27\Scripts>
```

mcspr, 2019-06-03:
Maybe slower? `-b 9600` If it fails seemingly at random, maybe wiring is causing problems.
Putty can show boot mode, (just to make sure it works, we want a single line with (1,...)). Just make sure to disconnect putty before running esptool itself

davebuk, 2019-06-03:
No different at 9600. I'll do some more testing over the next few days and see what I can up with.

davebuk, 2019-06-03:
Got it. Using info from [(https://github.com/arendst/Sonoff-Tasmota/wiki/Tuya-Dimmer)], I connected GND to the forth header pin while booting and holding GPIO 0 to ground. I now have:

```
c:\Python27\Scripts>esptool.py chip_id
esptool.py v2.6
Found 1 serial ports
Serial port COM3
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: bc:dd:c2:a6:cc:bd
Uploading stub...
Running stub...
Stub running...
Chip ID: 0x00a6ccbd
Hard resetting via RTS pin...
```

davebuk, 2019-06-03:
Downloaded file is here:
[flash_contents.zip](https://github.com/xoseperez/espurna/files/3249401/flash_contents.zip)

Thanks for your help. Am I ok to try Tasmota on this device now?

mcspr, 2019-06-04:
Yes. Thank you for the file! I just tried it with 1M dev board and see
the heartbeat on hw serial, as expected.

If you don't mind going slightly offtopic, can you try uploading some
basic (literally sonoff-basic should be ok) espurna image first? I was
not able to reproduce the issue from
https://github.com/xoseperez/espurna/issues/1663 and I wonder if
recent dev builds solved it or I am missing some other bug because of
espota.py usage. I tried upgrading via Web too, no problems :/

davebuk, 2019-06-04:
No problem. Just to be sure, you would like espurna dev uploaded via esptool then once initialised, try tasmota using espurna web OTA?

mcspr, 2019-06-04:
Correct. The test is if Tasmota sonoff.bin or sonoff-basic.bin work after updating from ESPurna via Web OTA

davebuk, 2019-06-04:
Working correctly for me. Uploaded a self built version of Sonoff Basic 1.13.6dev of espurna using esptool.py. Espurna boots correctly and I connected to my WiFi. Re-booted then used espurna OTA to flash Tasmota 'sonoff.bin'. Rebooted and device is now showing as sonoff device in wireless networks. Connected and setup Tasmota WiFi credentials and device is now connected to my WiFi and accessible using Tasmota web GUI.

Hope that is of some help?

davebuk, 2019-06-04:
The dimmer works with Tasmota using device configuration 'Tuya Dimmer 54'. ON/OFF toggle and dimming works using web GUI.

mcspr, 2019-06-04:
Great. Thanks for checking the OTA process!

Now the only thing left is to implement tuya dimmer module.

davebuk, 2019-06-04:
Thanks for your help. I've learnt alot over the past few days!
I'm sure it'll only take you a few days to implement the dimmer code for espurna! Let me know if I can help with any testing.

davebuk, 2019-06-05:
In addition to dimming, would the ability to dim using external push button or switch be viable. Ref #1331 ?

mcspr, 2019-06-08:
Button dimming is interesting in this context. idk why that pr never got split up. It got an important addition too - different scheduling method for transitions to make them more consistent.
Although, based on the documentation (and tuyadimmer tasmota code, again), that button is intended to reset settings (wifi or all)

Some things I've noticed after drafting a module for this at https://github.com/xoseperez/espurna/compare/dev...mcspr:experimental/tuya (**WARNING!** will crash right now, guaranteed :)
- light update happens in ticker context and crashes instantly when doing serial writes. that PR code pretty much describes what we need - add loop function with provider update
- default 10ms per transition between brightness levels is too fast for 9600 speed serial communication. so we either need to anticipate that and increase it proportionally, or add runtime setting to increase it, or turn it off  completely for tuya.
- mcu can send gpio mapping and we need to update existing configuration. likely a speed-up for #1680

davebuk, 2019-06-08:
The 'button' option was to use spare GPIOs on the ESP and espurna software to allow extra external buttons for dimming control. I'm in the UK and here we don't normally have a 240V neutral at the light switches. I'd plan to fit one of these devices in the ceiling and have dimmer control using the two wires that come from the wall switch. They'd have to perhaps be configured so double press to brighten up and long press to dim down, with single press to toggle on/off.

From what I understand of the tuya docs and your knowledge, the ESP is only there to provide the interface to the outside world. The MCU does all the control via serial from the ESP, not GPIO pins. Like you said, even the physical button on the device must connect to the MCU rather than the ESP because holding it down at boot wouldn't let me flash it. I had to connect the ESP GPIO 0 to GND to get it to work (while the third header pin was also at GND throughout the flash).

davebuk, 2019-06-08:
There is some info here: https://www.letscontrolit.com/forum/viewtopic.php?t=5431

Not sure if it's of any use?

mcspr, 2019-06-08:
Oh. I was reading the https://docs.tuya.com/en/mcu/mcu-protocol.html#3query-the-working-mode-of-the-set-wi-fi-module, and secondary chip can sometimes send real pin numbers. Just for compatibility sake, but we are not dealing with one of those here.
Long-press from the PR is a bit weird though. I think that was meant to go to the library code instead, probably need to add that there first... Double,triple,long click methods should work right now though.

2nd link is interesting. Per the same doc latest section, functional protocol that sort-of matches length 8 and 4 byte value is "temperature", but the forum post got dpID 03 instead of 02.
https://www.letscontrolit.com/forum/viewtopic.php?t=5431#p31388
(which is later used as 2 for some reason, and back again)
>  55AA 00 06 0008 -> 03 <- 02 0004 00000038 4E

this hints as to why tasmota code remembers byte in that position from the data mcu sends out initially (assuming documentation protocols are just examples)

Other documentation on the tuya website is in chinese and "mcu simulator" app I got excited about does need some kind of protocol description .json

mcspr, 2019-06-15:
I have updated https://github.com/mcspr/espurna/tree/experimental/tuya with the **tuya-generic-dimmer** platformio env to build the firmware .bin
I think the communication is ok now, debugging rx/tx is on by default (tuyaDebug setting, 0/1).
Having re-read the documentation, I think `55aa0304000006` should be the packet for the button (aka wifi reset)? No handler for that yet, only DP config and switch / brightness commands from the ESP to the MCU.

davebuk, 2019-06-16:
Great work. I'll try and find some time later today to build and test.

davebuk, 2019-06-16:
What's the web GUI login? admin / fibonacci doesn't seem to work. There doesn't seem to be a password required for the initial WiFi access point either.

davebuk, 2019-06-16:
I have built and uploaded using Tasmota minimal .bin then espurna firmare.bin via ota. The device first showed as an espurna-(chip ID) but with no security for the WiFi AP. After a power cycle, I was able to connect and login to the web GUI. Not sure if some Tasmota config was still partially loaded as my WiFi SSID was already there, just no password.

I can access the device via my WiFi but do not have any control of the output. Using 'info' in the debug section I get:

```
[571338] [WEBSOCKET] #5 connected, ip: 192.168.1.121, url: /ws
[574884] [TYUA] OUT: 55aa00000000ff
[574908] [TYUA] IN: 55aa000000010101
[574911] [TYUA] OUT: 55aa000300010407
[574935] [TYUA] IN: 55aa0003000002
[580406] [WEBSOCKET] Requested action: dbgcmd
[580415] 

---8<-------

[580417] [MAIN] ESPURNA 1.13.6-dev
[580418] [MAIN] xose.perez@gmail.com
[580419] [MAIN] http://tinkerman.cat

[580422] [MAIN] CPU chip ID: 0xA6CCBD
[580424] [MAIN] CPU frequency: 80 MHz
[580424] [MAIN] SDK version: 1.5.3(aec24ac9)
[580425] [MAIN] Core version: 2.3.0
[580426] [MAIN] Core revision: 159542381
[580429] [MAIN] Build time: 1560706427
[580431] 
[580432] [MAIN] Flash chip ID: 0x1440C8
[580437] [MAIN] Flash speed: 40000000 Hz
[580439] [MAIN] Flash mode: DOUT
[580441] 
[580443] [MAIN] Flash size (CHIP)   :  1048576 bytes /  256 sectors (   0 to  255)
[580445] [MAIN] Flash size (SDK)    :  1048576 bytes /  256 sectors (   0 to  255)
[580447] [MAIN] Reserved            :     4096 bytes /    1 sectors (   0 to    0)
[580449] [MAIN] Firmware size       :   504320 bytes /  124 sectors (   1 to  124)
[580454] [MAIN] Max OTA size        :   253952 bytes /   62 sectors ( 125 to  186)
[580456] [MAIN] SPIFFS size         :   262144 bytes /   64 sectors ( 187 to  250)
[580458] [MAIN] EEPROM size         :     4096 bytes /    1 sectors ( 251 to  251)
[580460] [MAIN] Reserved            :    16384 bytes /    4 sectors ( 252 to  255)
[580463] 
[580465] [MAIN] EEPROM sectors: 251, 250
[580466] [MAIN] EEPROM current: 250
[580467] 
[580468] [MAIN] EEPROM:  4096 bytes initially |   996 bytes used (24%) |  3100 bytes free (75%)
[580469] [MAIN] Heap  : 33056 bytes initially | 18952 bytes used (57%) | 14104 bytes free (42%)
[580473] [MAIN] Stack :  4096 bytes initially |  1936 bytes used (47%) |  2160 bytes free (52%)
[580474] 
[580477] [MAIN] Boot version: 31
[580479] [MAIN] Boot mode: 1
[580480] [MAIN] Last reset reason: Power on
[580482] [MAIN] Last reset info: flag: 0
[580484] 
[580485] [MAIN] Board: TUYA_GENERIC_DIMMER
[580487] [MAIN] Support: ALEXA API BROKER BUTTON DEBUG_TELNET DEBUG_WEB DOMOTICZ HOMEASSISTANT LED MDNS_SERVER MQTT NTP SCHEDULER TELNET TERMINAL THINGSPEAK WEB 
[580490] [MAIN] WebUI image: LIGHT
[580491] 
[580492] [MAIN] Firmware MD5: e9c4025aa1b736841ffbffe0e1bbfc07
[580496] [MAIN] Power: 3190 mV
[580498] [MAIN] Power saving delay value: 10 ms
[580499] 

---8<-------

[580500] +OK
[583887] [TYUA] OUT: 55aa00000000ff
[583912] [TYUA] IN: 55aa000000010101
[583914] [TYUA] OUT: 55aa000300010407
[583937] [TYUA] IN: 55aa0003000002
[592897] [TYUA] OUT: 55aa00000000ff
[592921] [TYUA] IN: 55aa000000010101
[592924] [TYUA] OUT: 55aa000300010407
[592947] [TYUA] IN: 55aa0003000002
[601899] [TYUA] OUT: 55aa00000000ff
[601923] [TYUA] IN: 55aa000000010101
[601926] [TYUA] OUT: 55aa000300010407
[601948] [TYUA] IN: 55aa0003000002
[610905] [TYUA] OUT: 55aa00000000ff
[610930] [TYUA] IN: 55aa000000010101
[610932] [TYUA] OUT: 55aa000300010407
[610955] [TYUA] IN: 55aa0003000002
[619908] [TYUA] OUT: 55aa00000000ff
[619932] [TYUA] IN: 55aa000000010101
[619935] [TYUA] OUT: 55aa000300010407
[619958] [TYUA] IN: 55aa0003000002
[628910] [TYUA] OUT: 55aa00000000ff
[628935] [TYUA] IN: 55aa000000010101
[628937] [TYUA] OUT: 55aa000300010407
[628960] [TYUA] IN: 55aa0003000002 
```
The status page looks like:

[image]

After changing brightness on the status page I get:

```
[321009] [WEBSOCKET] Requested action: brightness
[321027] [TYUA] OUT: 55aa0006000803020004000000ff15
[322369] [TYUA] OUT: 55aa00000000ff
[322393] [TYUA] IN: 55aa000000010101
[322396] [TYUA] OUT: 55aa000300010407
[322418] [TYUA] IN: 55aa0003000002
[331377] [TYUA] OUT: 55aa00000000ff
[331401] [TYUA] IN: 55aa000000010101
[331404] [TYUA] OUT: 55aa000300010407
[331426] [TYUA] IN: 55aa0003000002
``` 
but the light is not controllable. It will toggle on and off by pressing the button on the device giving:
```
[442577] [TYUA] IN: 55aa00070008020200040000001026
[442579] [TUYA] Unknown DP id=2 type=2
[442591] [TYUA] IN: 55aa0007000501010001010f
[442614] [TYUA] IN: 55aa0007000802020004000000ff15
[442615] [TUYA] Unknown DP id=2 type=2
[445528] [TYUA] IN: 55aa0007000501010001000e
[445542] [TYUA] IN: 55aa0007000802020004000000ff15
[445545] [TUYA] Unknown DP id=2 type=2
[445557] [TYUA] IN: 55aa0007000501010001000e
[445569] [TYUA] IN: 55aa00070008020200040000001026
[445570] [TUYA] Unknown DP id=2 type=2
[448472] [TYUA] OUT: 55aa00000000ff
```
but again, the light cannot a adjusted in brightness from the web GUI.

Hope that is of some help?

mcspr, 2019-06-16:
It is a strange AP mode issue. I did notice some code in tasmota modifying SDK wifi settings storage on wifi disconnect, maybe that what causes it. But as you have noted, it goes away after reset because we modify it again.

Brightness control must work after changing doing `set tuyaDimmerDP 2` and rebooting, I have missed setSetting call somewhere in the queries but I got some new idea for that anyways.
And one virtual relay, I'll update the hardware config with 1 dummy relay and RELAY_PROVIDER_LIGHT
It is strange, however, that it sends 16 (55aa0007000802020004000000 --> 10 <-- 26) as OFF brightness value. But it would not matter, as relay would turn the light off first.

Button does send anything different on double, triple or long presses?

I think I'll add tuya device description + discovered protocols to the `info` banners, so its not only displayed on boot.

davebuk, 2019-06-17:
I hadn't realised you had added that code. The following is single on/off, double, triple, off, 5 seconds held on, off, 10 seconds held on, off, 30 seconds held on, off.

For reference, the built in LED is off when off and RED when the light is on.
```
[616544] [WEBSOCKET] #8 connected, ip: 192.168.1.121, url: /ws
[623703] [TYUA] OUT: 55aa00000000ff
[623727] [TYUA] IN: 55aa000000010101
[623730] [TYUA] OUT: 55aa000300010407
[623752] [TYUA] IN: 55aa0003000002
[632704] [TYUA] OUT: 55aa00000000ff
[632728] [TYUA] IN: 55aa000000010101
[632731] [TYUA] OUT: 55aa000300010407
[632754] [TYUA] IN: 55aa0003000002
[641712] [TYUA] OUT: 55aa00000000ff
[641736] [TYUA] IN: 55aa000000010101
[641739] [TYUA] OUT: 55aa000300010407
[641762] [TYUA] IN: 55aa0003000002
[650722] [TYUA] OUT: 55aa00000000ff
[650746] [TYUA] IN: 55aa000000010101
[650749] [TYUA] OUT: 55aa000300010407
[650771] [TYUA] IN: 55aa0003000002
[659727] [TYUA] OUT: 55aa00000000ff
[659752] [TYUA] IN: 55aa000000010101
[659755] [TYUA] OUT: 55aa000300010407
[659778] [TYUA] IN: 55aa0003000002
[668022] [TYUA] IN: 55aa0007000501010001010f
[668046] [TYUA] IN: 55aa00070008020200040000001026
[668049] [TUYA] Unknown DP id=2 type=2
[668050] [TYUA] IN: 55aa0007000501010001010f
[668074] [TYUA] IN: 55aa0007000802020004000000ff15
[668076] [TUYA] Unknown DP id=2 type=2
[668731] [TYUA] OUT: 55aa00000000ff
[668755] [TYUA] IN: 55aa000000010101
[668758] [TYUA] OUT: 55aa000300010407
[668781] [TYUA] IN: 55aa0003000002
[669466] [TYUA] IN: 55aa0007000501010001000e
[669480] [TYUA] IN: 55aa0007000802020004000000ff15
[669482] [TUYA] Unknown DP id=2 type=2
[669495] [TYUA] IN: 55aa0007000501010001000e
[669509] [TYUA] IN: 55aa00070008020200040000001026
[669511] [TUYA] Unknown DP id=2 type=2
[670643] [TYUA] IN: 55aa0007000501010001010f
[670667] [TYUA] IN: 55aa00070008020200040000001026
[670670] [TUYA] Unknown DP id=2 type=2
[670672] [TYUA] IN: 55aa0007000501010001010f
[670695] [TYUA] IN: 55aa0007000802020004000000ff15
[670697] [TUYA] Unknown DP id=2 type=2
[670954] [TYUA] IN: 55aa0007000501010001000e
[670979] [TYUA] IN: 55aa0007000802020004000000ff15
[670981] [TUYA] Unknown DP id=2 type=2
[670983] [TYUA] IN: 55aa0007000501010001000e
[671006] [TYUA] IN: 55aa00070008020200040000001026
[671008] [TUYA] Unknown DP id=2 type=2
[672027] [TYUA] IN: 55aa0007000501010001010f
[672041] [TYUA] IN: 55aa00070008020200040000001026
[672044] [TUYA] Unknown DP id=2 type=2
[672056] [TYUA] IN: 55aa0007000501010001010f
[672070] [TYUA] IN: 55aa0007000802020004000000ff15
[672072] [TUYA] Unknown DP id=2 type=2
[672482] [TYUA] IN: 55aa0007000501010001000e
[672496] [TYUA] IN: 55aa0007000802020004000000ff15
[672499] [TUYA] Unknown DP id=2 type=2
[672511] [TYUA] IN: 55aa0007000501010001000e
[672524] [TYUA] IN: 55aa00070008020200040000001026
[672526] [TUYA] Unknown DP id=2 type=2
[672895] [TYUA] IN: 55aa0007000501010001010f
[672910] [TYUA] IN: 55aa00070008020200040000001026
[672913] [TUYA] Unknown DP id=2 type=2
[672925] [TYUA] IN: 55aa0007000501010001010f
[672939] [TYUA] IN: 55aa0007000802020004000000ff15
[... 59 lines cut ...]
[720268] [TUYA] Unknown DP id=2 type=2
[720282] [TYUA] IN: 55aa0007000501010001000e
[720294] [TYUA] IN: 55aa00070008020200040000001026
[720295] [TUYA] Unknown DP id=2 type=2
[722766] [TYUA] OUT: 55aa00000000ff
[722790] [TYUA] IN: 55aa000000010101
[722793] [TYUA] OUT: 55aa000300010407
[722816] [TYUA] IN: 55aa0003000002
[731772] [TYUA] OUT: 55aa00000000ff
[731796] [TYUA] IN: 55aa000000010101
[731799] [TYUA] OUT: 55aa000300010407
[731823] [TYUA] IN: 55aa0003000002
[740780] [TYUA] OUT: 55aa00000000ff
[740804] [TYUA] IN: 55aa000000010101
[740807] [TYUA] OUT: 55aa000300010407
[740830] [TYUA] IN: 55aa0003000002
[749784] [TYUA] OUT: 55aa00000000ff
[749808] [TYUA] IN: 55aa000000010101
[749811] [TYUA] OUT: 55aa000300010407
[749834] [TYUA] IN: 55aa0003000002
[758789] [TYUA] OUT: 55aa00000000ff
[758814] [TYUA] IN: 55aa000000010101
[758817] [TYUA] OUT: 55aa000300010407
[758840] [TYUA] IN: 55aa0003000002
```

davebuk, 2019-06-17:
On a side note, after flashing using the Tasmota minimal, then espurna 493KB .bin file, I only have 258048 bytes free for an OTA update. I guess using the two step method with Tasmota has re-sized the flash memory? Can I get back to two 512KB halves using OTA uploads or will I just need to erase/upload espurna using the serial ports?

mcspr, 2019-06-17:
Button does not seem to do anything other than sending brightness 0 to 255 and back, switch ON OFF.
What does the paper manual say about wifi modes? I did find one example: https://fccid.io/ZZH-WF500D/User-Manual/Users-Manual-3846958
Pressing 5 times in a row and holding 6th should trigger WiFi config? Maybe that will work?

That flash size issue is my bad, I copied the build flags string from the other wemos board which also had it wrong. It should be `${common.build_flags_1m0m}` instead of `${common.build_flags}`. With platformio defaults we reserve 256Kb for SPIFFS and can't use that space for firmware. No way to OTA upgrade I'm afraid, It will reject firmware based on size first, not looking at the new flash layout.

davebuk, 2019-06-17:
The manuals says that if the devices doesn't boot with a rapidly flashing LED, hold the button down for 15 seconds and confirm the LED is now flashing rapidly.

Pressing 5 times and holding on the sixth gives:

```
[421641] [TYUA] OUT: 55aa00000000ff
[421664] [TYUA] IN: 55aa000000010101
[421666] [TYUA] OUT: 55aa000300010407
[421689] [TYUA] IN: 55aa0003000002
[430649] [TYUA] OUT: 55aa00000000ff
[430674] [TYUA] IN: 55aa000000010101
[430678] [TYUA] OUT: 55aa000300010407
[430701] [TYUA] IN: 55aa0003000002
[437921] [TYUA] IN: 55aa0007000501010001010f
[437934] [TYUA] IN: 55aa00070008020200040000001026
[437936] [TUYA] Unknown DP id=2 type=2
[437947] [TYUA] IN: 55aa0007000501010001010f
[437970] [TYUA] IN: 55aa0007000802020004000000ff15
[437971] [TUYA] Unknown DP id=2 type=2
[438453] [TYUA] IN: 55aa0007000501010001000e
[438468] [TYUA] IN: 55aa0007000802020004000000ff15
[438470] [TUYA] Unknown DP id=2 type=2
[438483] [TYUA] IN: 55aa0007000501010001000e
[438495] [TYUA] IN: 55aa00070008020200040000001026
[438496] [TUYA] Unknown DP id=2 type=2
[438999] [TYUA] IN: 55aa0007000501010001010f
[439013] [TYUA] IN: 55aa00070008020200040000001026
[439014] [TUYA] Unknown DP id=2 type=2
[439026] [TYUA] IN: 55aa0007000501010001010f
[439041] [TYUA] IN: 55aa0007000802020004000000ff15
[439042] [TUYA] Unknown DP id=2 type=2
[439535] [TYUA] IN: 55aa0007000501010001000e
[439559] [TYUA] IN: 55aa0007000802020004000000ff15
[439562] [TUYA] Unknown DP id=2 type=2
[439563] [TYUA] IN: 55aa0007000501010001000e
[439588] [TYUA] IN: 55aa00070008020200040000001026
[439591] [TUYA] Unknown DP id=2 type=2
[439654] [TYUA] OUT: 55aa00000000ff
[439678] [TYUA] IN: 55aa000000010101
[439681] [TYUA] OUT: 55aa000300010407
[439704] [TYUA] IN: 55aa0003000002
[440105] [TYUA] IN: 55aa0007000501010001010f
[440118] [TYUA] IN: 55aa00070008020200040000001026
[440120] [TUYA] Unknown DP id=2 type=2
[440131] [TYUA] IN: 55aa0007000501010001010f
[440153] [TYUA] IN: 55aa0007000802020004000000ff15
[440157] [TUYA] Unknown DP id=2 type=2
[440640] [TYUA] IN: 55aa0007000501010001000e
[440654] [TYUA] IN: 55aa0007000802020004000000ff15
[440655] [TUYA] Unknown DP id=2 type=2
[440667] [TYUA] IN: 55aa0007000501010001000e
[440678] [TYUA] IN: 55aa00070008020200040000001026
[440679] [TUYA] Unknown DP id=2 type=2
[448658] [TYUA] OUT: 55aa00000000ff
[448682] [TYUA] IN: 55aa000000010101
[448683] [TYUA] OUT: 55aa000300010407
[448705] [TYUA] IN: 55aa0003000002
```
Does it matter in the output code it says [TYUA] rather than [TUYA]?
When you are ready, let me know when to try building the next test version.

mcspr, 2019-06-18:
I was expecting to see either 55aa0004... or 55aa0005... somewhere. Huh. Maybe the 15 second hold would send it. And TYUA thing is just a misspelling in the logging method 🤷‍♂ 

Will do. I want to try to replace static switch / light config with some dynamic detection.

*ot. researching some other tuya modules, they can be pretty expressive: https://blakadder.github.io/templates/oil_diffuser_550ml.html*

davebuk, 2019-07-02:
@mcspr Have you found any time to look any more into the code? The external button press for dimming is not something I need for my current application, it was just a nice to have.
Do you have any working code that will enable power toggle using the built in switch and on/off and dimming control using MQTT?

mcspr, 2019-07-02:
I'll try to come up with something mergeable this week.

Current tree is not working, but I did include the relay definition and flash layout fix 2 weeks ago:
https://github.com/mcspr/espurna/commit/bca19f4eab5c3702404b6a9b878911e4711b60dc
Have you tried setting additional params in the terminal for the previous running build (bc last message shows `Unknown DP id=2 type=2`, including reboot to apply)?
```
set tuyaDimmerDP 2
set useTransitions 0
```
First one will apply the correct message DP, so both In and Out messages are parsed by both sides (e.g. `55aa00070008020200040000001026`). Second one I think is optional, it disables transitions if they look strange.

davebuk, 2019-07-03:
Failed to build, /code/espurna/tuya.ino:23:27: fatal error: tuya_protocol.h: No such file or directory

Unless I'm looking in the wrong place, I can't find tuya_protocol.h

mcspr, 2019-07-03:
It's not commited yet, note "not working" part. :) I have not tested that thing yet, but uploaded part of it as a backup.

If you want just the fix mentioned above, `git checkout <commit hash>` (add -b ... to checkout into a separate branch)

davebuk, 2019-07-03:
Sorry, my misunderstanding ;-) I'll wait until you have the code ready to merge. I am currently running the tasmota firmware so no rush.

Thanks again.

mcspr, 2019-07-07:
I have updated the experimental branch with protocol discovery and better (hopefully!) relay / channel interactions.
Based on the logs above, we are using 1 light channel and 1 switch that controls it.

Discovery should work out configuration for both. If not:
```
set tuyaSwitch0 1
set tuyaChannel0 2
reset
```
I tried with what I think discovery looks like and the config above.

Sidenote: while there is no Serial logging here, it can be remedied by using:
```
-DDEBUG_PORT=Serial1 -DDEBUG_SERIAL_SUPPORT=1
```
It did not always work for me though. Earlier I tried testing with current Core 2.5.2, where it worked. It *does not* work with platformio env using 2.3.0 for some reason.
It would be interesting to see early boot logs.

davebuk, 2019-07-08:
Looking good so far. :-)
After initial wifi setup etc the connected light was on and the button toggle switch switched the light off but after a second or so, it came back on, repeatedly. Using the web GUI I had a toggle switch and after selecting On/Off, the software and hardware button worked. The slight problem is that when selecting ON, the brightness changes to '1', so it looks like the light is off. 
[image]

**Brightness**
From the GUI I can then change the brightness up and down so the light is actually on and changing values. Every time change the brightness of Channel #0, it then jumps up to that value shown plus another 16 (e.g. set brightness to 200, it shows 200 then jumps to 216).

**ON/OFF Toggle**
I then rebooted from the GUI to get boot debugging info. This time I didn't have the ON/OFF toggle, but moving the #0 slider turned the light on and adjusted the brightness. By using the three configuration commands above and rebooting brought the switch back.

**Post boot**
```
[012012] [TUYA] OUT: 55aa00000000ff
[012036] [TUYA] IN: 55aa000000010101
[013472] [NTP] UTC Time  : 2019-07-08 13:31:24
[013473] [NTP] Local Time: 2019-07-08 15:31:24
[021020] [TUYA] OUT: 55aa00000000ff
[021044] [TUYA] IN: 55aa000000010101
[030021] [TUYA] OUT: 55aa00000000ff
[030045] [TUYA] IN: 55aa000000010101
[039027] [TUYA] OUT: 55aa00000000ff
[039050] [TUYA] IN: 55aa000000010101
[048034] [TUYA] OUT: 55aa00000000ff
[048057] [TUYA] IN: 55aa000000010101
[057035] [TUYA] OUT: 55aa00000000ff
[057059] [TUYA] IN: 55aa000000010101
[060007] [MAIN] System OK
[066036] [TUYA] OUT: 55aa00000000ff
[066060] [TUYA] IN: 55aa000000010101
[075045] [TUYA] OUT: 55aa00000000ff
[075069] [TUYA] IN: 55aa000000010101
```
**Some ON/OFF commands**
```
[796602] [WEBSOCKET] Requested action: relay
[796604] [RELAY] #0 scheduled OFF in 0 ms
[796606] [RELAY] #0 set to OFF
[796609] [TUYA] OUT: 55aa0006000501010001000d
[796646] [TUYA] IN: 55aa0007000501010001000e
[796660] [TUYA] IN: 55aa0007000501010001000e
[796673] [TUYA] IN: 55aa00070008020200040000001026
[797608] [RELAY] Setting relay mask: 0
[799607] [WEBSOCKET] Requested action: relay
[799609] [RELAY] #0 scheduled ON in 0 ms
[799611] [RELAY] #0 set to ON
[799615] [TUYA] OUT: 55aa0006000501010001010e
[799627] [TUYA] OUT: 55aa00060008020200040000000116
[799651] [TUYA] IN: 55aa0007000501010001010f
[799663] [TUYA] IN: 55aa0007000501010001010f
[799675] [TUYA] IN: 55aa00070008020200040000001026
[799691] [TUYA] IN: 55aa00070008020200040000000117
[800613] [RELAY] Setting relay mask: 1
[801616] [WEBSOCKET] Requested action: relay
[801618] [RELAY] #0 scheduled OFF in 0 ms
[801624] [RELAY] #0 set to OFF
[801631] [TUYA] OUT: 55aa0006000501010001000d
[801664] [TUYA] IN: 55aa0007000501010001000e
[801676] [TUYA] IN: 55aa0007000501010001000e
[801688] [TUYA] IN: 55aa00070008020200040000001026
[802627] [RELAY] Setting relay mask: 0
[803421] [WEBSOCKET] Requested action: relay
[803424] [RELAY] #0 scheduled ON in 0 ms
[803425] [RELAY] #0 set to ON
[803429] [TUYA] OUT: 55aa0006000501010001010e
[803441] [TUYA] OUT: 55aa00060008020200040000000116
[803464] [TUYA] IN: 55aa0007000501010001010f
[803476] [TUYA] IN: 55aa0007000501010001010f
[803488] [TUYA] IN: 55aa00070008020200040000001026
[803501] [TUYA] IN: 55aa00070008020200040000000117
[804388] [TUYA] OUT: 55aa00000000ff
[804411] [TUYA] IN: 55aa000000010101
[804428] [RELAY] Setting relay mask: 1
[804542] [WEBSOCKET] Requested action: relay
[804545] [RELAY] #0 scheduled OFF in 0 ms
[804548] [RELAY] #0 set to OFF
[804554] [TUYA] OUT: 55aa0006000501010001000d
[804588] [TUYA] IN: 55aa0007000501010001000e
[804600] [TUYA] IN: 55aa0007000501010001000e
[804612] [TUYA] IN: 55aa00070008020200040000001026
[805553] [RELAY] Setting relay mask: 0
[813392] [TUYA] OUT: 55aa00000000ff
[813416] [TUYA] IN: 55aa000000010101
```
**Brightness changes**
```
[917941] [WEBSOCKET] Requested action: relay
[917943] [RELAY] #0 scheduled ON in 0 ms
[917946] [RELAY] #0 set to ON
[917953] [TUYA] OUT: 55aa0006000501010001010e
[917965] [TUYA] OUT: 55aa00060008020200040000000116
[917988] [TUYA] IN: 55aa0007000501010001010f
[918000] [TUYA] IN: 55aa0007000501010001010f
[918012] [TUYA] IN: 55aa00070008020200040000001026
[918027] [TUYA] IN: 55aa00070008020200040000000117
[918952] [RELAY] Setting relay mask: 1
[921009] [WEBSOCKET] Requested action: channel
[921116] [TUYA] OUT: 55aa00060008020200040000008499
[921161] [TUYA] IN: 55aa0007000802020004000000849a
[921162] [TUYA] IN: 55aa0007000501010001010f
[921185] [TUYA] IN: 55aa000700080202000400000094aa
[921458] [TUYA] OUT: 55aa00000000ff
[921481] [TUYA] IN: 55aa000000010101
[923591] [WEBSOCKET] Requested action: channel
[923698] [TUYA] OUT: 55aa0006000802020004000000bed3
[923741] [TUYA] IN: 55aa0007000802020004000000bed4
[923754] [TUYA] IN: 55aa0007000501010001010f
[923767] [TUYA] IN: 55aa0007000802020004000000cce2
[926536] [WEBSOCKET] Requested action: channel
[926643] [TUYA] OUT: 55aa00060008020200040000004055
[926687] [TUYA] IN: 55aa00070008020200040000004056
[926699] [TUYA] IN: 55aa0007000501010001010f
[926713] [TUYA] IN: 55aa00070008020200040000004e64
[930459] [TUYA] OUT: 55aa00000000ff
```
**tuya.show**
```
[093190] [WEBSOCKET] Requested action: dbgcmd
[093194] [TUYA] BOOL
[093195] tuyaSwitch0=1 value=ON
[093195] [TUYA] INT
[093196] tuyaChannel0=2 value=78
```
This corresponded with the values on the web 'Status' page.

mcspr, 2019-07-08:
> After initial wifi setup etc the connected light was on and the button toggle switch switched the light off but after a second or so, it came back on, repeatedly. Using the web GUI I had a toggle [...]

Thank you for checking that out!

Does the device remember the last state all by itself and web shown switch as ON initially? And without touching real button (I got that right?), UI / MQTT switching did work initially?
Regarding the switch loop, did it log "[RELAY] #0 scheduled ..." etc. while this was happening? I got a vague idea why it may desync because of the channel offset situation that you describe.

> Every time change the brightness of Channel #0, it then jumps up to that value shown plus another 16 (e.g. set brightness to 200, it shows 200 then jumps to 216).

This is real strange thing. Looking at logs, sending 1 (0x1) results in it returning 16 (0x10) and immediately reporting 1 (0x1) again. Or another example:
```
[419996] [TUYA] OUT: 55aa0006000802020004000000 -> 38 <- 4d // setting channel slider to 56
[420031] [TUYA] IN: 55aa0007000802020004000000 -> 38 <- 4e // MCU acknowledges (as documented)
[420043] [TUYA] IN: 55aa0007000501010001 -> 01 <- 0f // light is ON (so should be the switch toggle)
[420067] [TUYA] IN: 55aa0007000802020004000000 -> 48 <- 5e // MCU reports brightness 72 (what?)
```
But what happens with full channel brightness 255 (0xFF) or near it? Terminal has `channel 0 <value>`

> I then rebooted from the GUI to get boot debugging info. This time I didn't have the ON/OFF toggle, but moving the #0 slider turned the light on and adjusted the brightness.

This is strange too. It does slightly different things because other MCU is not powered down, but nothing specific to make switch go missing... Will test this more closely. You sure that the switch did not come up after page reload?
Did it loop because of hardware button toggle is this state?

mcspr, 2019-07-08:
https://github.com/xoseperez/espurna/issues/1729#issuecomment-496889121
> Pin locations if others need them. To flash, the fourth pin needs to be held at GND before power applied and kept at GND during flashing. I think it disables Tx/Rx communication to the MCU. You then need to boot with the GPIO 0 pin held to GND, then release a few seconds after boot. The Tx/Rx wires need to be connected directly to the TYWE3S pins. Flash using which ever method you prefer.

GPIO0 is up 2 pins, based on this https://docs.tuya.com/en/hardware/WiFi-module/wifi-e3s-module.html#21-dimensions
The N/C between 15 and 0 is the TX1 (Serial1) which can be used to debug really early boot
(or I can add a buffer for all debug output during that time... which may help to debug reboot situation too)

davebuk, 2019-07-08:
> Does the device remember the last state all by itself and web shown switch as ON initially? And without touching real button (I got that right?), UI / MQTT switching did work initially?

Yes, switch all works as you would expect and it tries to switch back on at the previous level. When selected to OFF, the channel#0 goes to 16, MQTT reports 0. When selected back ON, the light very briefly flicks to somewhere near the previous value e.g 120 or 255 etc but then jumps straight to 1 with MQTT reporting 1. The light is still represented as ON just extremely dim (can't tell if ON or OFF) but the red LED on the device is still on which follows the ON/OFF commands.   

> Regarding the switch loop, did it log "[RELAY] #0 scheduled ..." etc. while this was happening? I got a vague idea why it may desync because of the channel offset situation that you describe.

Unfortunately I didn't see those logs.

> This is real strange thing. Looking at logs, sending 1 (0x1) results in it returning 16 (0x10) and immediately reporting 1 (0x1) again. Or another example:

Whats even more strange, regardless of ON/OFF state, doing a full power cycle (not software reboot), a second switch and channel value are set. `tuya.show` gives:
```
[861777] [WEBSOCKET] Requested action: dbgcmd
[861780] [TUYA] BOOL
[861781] tuyaSwitch0=1 value=ON
[861782] tuyaSwitch1=1 value=ON
[861782] [TUYA] INT
[861783] tuyaChannel0=2 value=165
[861783] tuyaChannel1=2 value=150
```
 These are switch#1 and channel#1. s#0 is independent of s#1 but s#1 makes s#0 follow after a few seconds. Adjusting the brightness of c#1 gives a value which does not jump by 16, then a few seconds later c#0 follows c#1 with the value plus 16. To note, the **brightness** slider has no effect and never changes from 255. Power cycling again doesn't try to add a third switch or delete the second, only using the following resets back to one switch and channel.
```
set tuyaSwitch0 1
set tuyaChannel0 2
reset
```

[image]

[image]

> But what happens with full channel brightness 255 (0xFF) or near it? Terminal has `channel 0 <value>`

Any value above 241 stays at that value, upto and including 255 with MQTT reporting the same. Less than 239, a few seconds later the value jumps to plus 16. 

> This is strange too. It does slightly different things because other MCU is not powered down, but nothing specific to make switch go missing... Will test this more closely. You sure that the switch did not come up after page reload?
> Did it loop because of hardware button toggle is this state?

No, the switch never appeared in the web GUI after a refresh of the page.

mcspr, 2019-07-08:
I accidentally added a second element when it can't update the current state. Happens only on boot and only without any predefined configuration. That phantom second response is the cause.

Double response from MCU can also be some clever way to guide transitions in tuya software, showing next possible step... (but only upwards for some reason). And sending values divisible without remainders does not seem to have any difference - one of the channel change logs sends value of 64, immediately responding with 78. Order is also something of note, it does send 16 and 1 instead of the usual value value+16 when channel value is 1.
Probably it can be safely ignored, comparing with "value"+-16 before doing anything with it

Will check it out

davebuk, 2019-07-09:
Tasmota seemed to do the same. Tasmota would remember the previous value and switch back on at that level, but then jump up a bit. Every ON/OFF would increase the brightness every time.

I use openHab to control everything, so had a rule that sets the dimmer value to 50% after being turned ON. After a few seconds it would jump to 56%. I've changed my rule for espurna so it now sets 111 and after a few seconds goes to 126.

davebuk, 2019-07-30:
@mcspr Hi there. Have you had a chance to look at making any changes to the code to remove the second auto configured channel and the possibility of the light returning to the previous brightness state when switching back on?
Thanks.

mcspr, 2019-07-30:
I haven't got a chance to test it yet, but I have updated the tuya branch. Will try in the morning if just ignoring dimmer state is sufficient enough. If not, I will try to add value filtering.
Software tests show that discovery problem should be fixed.

davebuk, 2019-07-31:
That will probably be fine to give better usage. I am current having issues controlling via MQTT as I have set that up to work WRT channel 1. Initial switch on levels are being set by rules in openHab.

mcspr, 2019-07-31:
With the current build or the previous one?

So the flow right now is:
- boot the device
- tuya module checks for saved channel / switch config, creates them if needed
- relay & light modules restore values from settings (brightness should always be restored, switch state is restored only when relay boot mode set to something else than the default)
- tuya discovery starts if previous step had not created any relays / channels, but state is received from the MCU
- any further brightness messages from the MCU are ignored and we only know about ON / OFF events (todo: setting)

[43 more replies]

PR #1997 (2019-11-20): Tuya support
Test build to resolve #1729 

Need to:
- <s>fix broker interface</s><br>see broker.h, simplify brokerPublish signature. this breaks backwards compatibility, but I guess I could restore it with old "type(int)" dispatching to a proper TBroker
- <s>take a look again if this is really relay / light provider</s><br>(note: this could work outside of relay module completely, but wait until other providers are configured as such)
- <s>clean-up tuya classes</s>
- <s>avoid breaking existing modules</s>
closed 2019-11-20 by mcspr