xoseperez/espurna #1735: Compile Error homeassitant.ino

Wikibear, 2019-05-22:
That error happens on compile with Plattform.io and ATOM also VSC
`
E:/espurna-master/code/espurna/homeassitant.ino:13:6: error: redefinition of 'bool _haEnabled'
 bool _haEnabled = false;
      ^
E:/espurna-master/code/espurna/homeassistant.ino:13:6: error: 'bool _haEnabled' previously defined here
 bool _haEnabled = false;
      ^

E:/espurna-master/code/espurna/homeassitant.ino:14:6: error: redefinition of 'bool _haSendFlag'
 bool _haSendFlag = false;
      ^
E:/espurna-master/code/espurna/homeassistant.ino:14:6: error: 'bool _haSendFlag' previously defined here
 bool _haSendFlag = false;
      ^
E:/espurna-master/code/espurna/homeassitant.ino: In function 'void _haWebSocketOnSend(ArduinoJson::JsonObject&)':
E:/espurna-master/code/espurna/homeassitant.ino:18:6: error: redefinition of 'void _haWebSocketOnSend(ArduinoJson::JsonObject&)'
 void _haWebSocketOnSend(JsonObject& root) {
      ^
E:/espurna-master/code/espurna/homeassistant.ino:248:6: error: 'void _haWebSocketOnSend(ArduinoJson::JsonObject&)' previously defined here
 void _haWebSocketOnSend(JsonObject& root) {
      ^
E:/espurna-master/code/espurna/homeassitant.ino: In function 'void _haSendSwitches()':
E:/espurna-master/code/espurna/homeassitant.ino:130:6: error: redefinition of 'void _haSendSwitches()'
 void _haSendSwitches() {
      ^
E:/espurna-master/code/espurna/homeassistant.ino:117:6: error: 'void _haSendSwitches()' previously defined here
 void _haSendSwitches() {
      ^
E:/espurna-master/code/espurna/homeassitant.ino: In function 'void _haSend()':
E:/espurna-master/code/espurna/homeassitant.ino:136:6: error: redefinition of 'void _haSend()'
 void _haSend() {
      ^
E:/espurna-master/code/espurna/homeassistant.ino:215:6: error: 'void _haSend()' previously defined here
 void _haSend() {
      ^
E:/espurna-master/code/espurna/homeassitant.ino: In function 'void _haConfigure()':
E:/espurna-master/code/espurna/homeassitant.ino:156:6: error: redefinition of 'void _haConfigure()'
 void _haConfigure() {
      ^
E:/espurna-master/code/espurna/homeassistant.ino:235:6: error: 'void _haConfigure()' previously defined here
 void _haConfigure() {
      ^
E:/espurna-master/code/espurna/homeassistant.ino:235:6: error: 'void _haConfigure()' previously defined here
 void _haConfigure() {
      ^
E:/espurna-master/code/espurna/homeassitant.ino: In function 'void haSetup()':
E:/espurna-master/code/espurna/homeassitant.ino:165:6: error: redefinition of 'void haSetup()'
 void haSetup() {
      ^
E:/espurna-master/code/espurna/homeassistant.ino:299:6: error: 'void haSetup()' previously defined here
 void haSetup() {
      ^
E:/espurna-master/code/espurna/homeassitant.ino:171:44: error: 'wsOnAfterParseRegister' was not declared in this scope
         wsOnAfterParseRegister(_haConfigure);
                                            ^
`

mcspr, 2019-05-25:
> E:/espurna-master/code/espurna/homeassitant.ino

see #924 and please retry with dev version instead

closed 2019-05-25 by mcspr

wsw70, 2019-05-27:
Would it be possible to fix `master` too? Right now the build fails.