xoseperez/espurna #1738: Alexa devices getting mixed up and controlling other espurna devices

clueo8, 2019-05-25:
All of a sudden my sonoffs with espurna 1.13.3 stopped working reliably with Alexa (haven't updated or touched config in months).  I tried adding/removing from Alexa and rediscovered them but it appears they are crossing paths now.  For example, I have a few sonoff basics and a dual.  Alexa finds them all but when it discovered the basic, turning the basic on and off is actually controlling one of the dual switches!  It's like Alexa is mixing up the devices.  I updated the dual to 1.13.5 to see if that would help but it didn't.  HA is able to control them fine still, just the Alexa is messed up now...

Alferez, 2019-05-25:
I have the same problem since Wednesday when I saw that alexa lit her circle of light alone. It seemed like it was updating.
I have tried to delete the devices in alexa and rediscover them, but it keeps on turning on and off the ones you want without having anything to do. My version is 1.13.5 and my Echo Dot 3rd firmware version is 2382875524

clueo8, 2019-05-26:
I did notice my Echo Show updating last week as well...

jemarr, 2019-05-27:
https://github.com/Aircoookie/Espalexa/issues/72

jemarr, 2019-05-27:
https://github.com/arendst/Sonoff-Tasmota/issues/5849

jemarr, 2019-05-27:
Seems that all Hue emulators has the same issue....
Fauxmo, Espalexa, Tasmota,...

jemarr, 2019-05-27:
Same problem since five days ago... Amazon has changed something.

I tried Fauxmo, Espalexa libs as well with same issue.

I have three ESP8266 boards with 1, 2 and 4 devices each. The last discovered devices board is which Alexa takes as 1st, 2nd.... device on Routines and Groups, doesn't matter the device name.

When you call by voice all devices runs well, the problem is just with Groups and Routines.

I've 2 Echo Dot 3rt Gen.

rmcbc, 2019-05-28:
I have the same problem, probably a Alexa update, everything was working ok, now it's all mixed.
Does any one knows how to solve this?

anrada73, 2019-05-29:
in Germany too.(

rmcbc, 2019-05-29:
I install 1.13.2 and it seems to work, I have only one device active, but it works, I give more feedback when I flash all the other modules.

rmcbc, 2019-05-29:
After configuring the firmware for more than 30 devices all are working ok with firmware 1.13.2
The Wemo emulation works and for me it's better because you can chose if the device is a light or a plug.
@xoseperez if it's possible make it possible to compile Wemo or Philips emulator with latest version, I don't know if I'm asking too much but I think it's a good thing to make it possible to chose one emulation or the other.
Thanks for your great job.

jemarr, 2019-05-31:
With Tasmota they have found a solution. What they do to solve the issue is to assign a unique LightId for each device based on the iP adress or MAC adress of the esp8266 board.

arendst/Sonoff-Tasmota#5890 (comment)

Is it possible to do the same with Espurna o Fauxmo lib ???

PieBru, 2019-06-03:
+1 for a solution non involving the DHCP variable part of the IP address.
IMO, fixed definitions cause maintenance workload over time, I prefer plug & play things.
Let's make DHCP things and Alexa happy toghether :)
Piero

rmcbc, 2019-06-03:
Use MAC address is diferent for each device and don't change with DHCP.

PieBru, 2019-06-04:
Even the MAC address can change when the LAN isn't wired or 802.11s
In my experience, some WDS implementations are flawed and don't solve completely the variable-MAC problem.
Piero

jemarr, 2019-06-16:
Amazon already fixed the issue... now is working fine again.

clueo8, 2019-07-05:
Confirmed; I'm not having this issue anymore.

closed 2019-07-05 by clueo8