xoseperez/espurna #1739: Missing xtensa-lx106-elf-g++

wsw70, 2019-05-26:
I am trying to build a `Dockerfile` to better automatize the build of ESPurna binaries and I have a problem with `xtensa-lx106-elf-g++`.

The `Dockerfile` is simple: it downloads `bash`, `curl`, `git` and  `g++`, and `platformio` ; then clones the repository and informs that when teh container is ran, `pio run -e ...` will be launched.

```
FROM python:2-alpine
RUN apk add --no-cache bash curl git g++
RUN python -c "$(curl -fsSL https://raw.githubusercontent.com/platformio/platformio/develop/scripts/get-platformio.py)"
RUN git clone https://github.com/xoseperez/espurna.git
WORKDIR /espurna/code
CMD pio run -e itead-sonoff-basic -e itead-sonoff-rfbridge -e itead-sonoff-touch -e aithinker-ai-light
```

The build is successful, but when running the container I get:

```
root@srv /e/d/espurna# docker run -e PLATFORMIO_BUILD_FLAGS="'-DUSE_CUSTOM_H'" -v /etc/docker/bin:/espurna/code/.pioenv -v /etc/docker/custom.h:/espurna/code/espurna/config/custom.h espurna
********************************************************************************
If you like PlatformIO, please:
- follow us on Twitter to stay up-to-date on the latest project news > https://twitter.com/PlatformIO_Org
- star it on GitHub > https://github.com/platformio/platformio
- try PlatformIO IDE for IoT development > https://platformio.org/platformio-ide
- support us with PlatformIO Plus > https://pioplus.com
********************************************************************************

Processing itead-sonoff-basic (framework: arduino; platform: espressif8266@1.5.0; board: esp01_1m)
--------------------------------------------------------------------------------
LibraryManager: Installing id=64 @ 5.13.4
Downloading...

(... libraries are being installed ...) 

Unpacking
PackageManager: Installing framework-arduinoespressif8266 @ ~1.20300.1
Downloading
Downloading
Unpacking
Unpacking
CorePackageManager: Installing tool-scons @ ~2.20501.7
Downloading
Downloading
Unpacking
Unpacking
tool-scons @ 2.20501.7 has been successfully installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp01_1m.html
PLATFORM: Espressif 8266 > Espressif Generic ESP8266 ESP-01 1M
HARDWARE: ESP8266 80MHz 80KB RAM (1MB Flash)
Converting espurna.ino
sh: xtensa-lx106-elf-g++: not found
*** Error 127
AssertionError: :
File "/usr/local/lib/python2.7/site-packages/platformio/builder/main.py", line 169:
env.SConscript("$BUILD_SCRIPT")
File "/root/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 541:
return _SConscript(self.fs, *files, **subst_kw)
File "/root/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 250:
exec _file_ in call_stack[-1].globals
File "/root/.platformio/platforms/espressif8266/builder/main.py", line 375:
target_elf = env.BuildProgram()
File "/root/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "/usr/local/lib/python2.7/site-packages/platformio/builder/tools/platformio.py", line 107:
env.BuildFrameworks(env.get("PIOFRAMEWORK"))
File "/root/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "/usr/local/lib/python2.7/site-packages/platformio/builder/tools/platformio.py", line 316:
env.ConvertInoToCpp()
File "/root/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "/usr/local/lib/python2.7/site-packages/platformio/builder/tools/piomisc.py", line 196:
out_file = c.convert(ino_nodes)
File "/usr/local/lib/python2.7/site-packages/platformio/builder/tools/piomisc.py", line 55:
return self.process(contents)
File "/usr/local/lib/python2.7/site-packages/platformio/builder/tools/piomisc.py", line 78:
assert self._gcc_preprocess(contents, out_file)
========================== [ERROR] Took 61.08 seconds ==========================

(... same error for the other binaries ...)
```

**What is `xtensa-lx106-elf-g++` and where should it come from?**

wsw70, 2019-05-26:
The `python:2-alpine`image is too restricted. I changed to the default one (`python:2`) and it works great.

I will provide a complete description to add to the "contributed" section of the wiki once I can confirm everything works.

closed 2019-05-26 by wsw70

copyrights, 2019-06-04:
Just for completeness as I ran in to that same issuein #1754 . Alpine is missing libc which xtensa-lx106-elf-g++ is link against.