xoseperez/espurna #1743: What could prevent a firmware upgrade?

wsw70, 2019-05-27:
One of my Sonoff Basic was behaving erratically (sometimes available, sometimes not) and I decided to reflash it with the latest `dev` firmware.

I can reach it via the web interface, the API and telnet. I tried to reflash it with the `dev` (bit also `master`, and even (sorry - that was just for tests :)) Tasmota) via

- the web interface: I chose the firmware, pressed `Upgrade`, the blue rectangle below the form appeared (but did not show any progress) and nothing.
- the API: the connection was successful (`100 Continue`) but ultimately failed
- telnet: 

```
[661157] [TELNET] Client #0 connected
ota http://192.168.10.251:8000/firmware.bin
[667996] +OK
[674935]
[674938] [EEPROM] Reenabling EEPROM rotation
[674939] [OTA] Disconnected
```

Basically nothing happened, I am still at 1.13.5.

I also tried to factory reset before retrying all the flashes above.

**What could be the reason for the firmware not accepting to be reflashed?**

I will ultimately reflash it via a cable but wanted to understand what could go wrong. Beside the refusal to reflash, the other functionality is OK (switch, MQTT, ...)

mcspr, 2019-05-28:
#1631 ? 
`set loopDelay 10` and `reset` should do it

wsw70, 2019-06-02:
That was it, thank you!

closed 2019-06-02 by wsw70