xoseperez/espurna #1744: Use > 1MB on Sonoff POW R2?

schweini, 2019-05-27:
It seems that the Sonoff POW R2 actually has 4MB of space on its ESP8266.

Since the R2's Espurna image is quite big (easily over 0.5M), using the extra space for OTA would be awesome, in order to avoid the two-step OTA thing.

Is there a way to use the extra space? I quickly tried simply telling the Arduino UI to compile for 4MB (1M Flash), and that didn't seem to work.

mcspr, 2019-05-29:
Which flash size option did you use? All 4M options aside from `4M (3M SPIFFS)` would allow to upload more than 512KB

xoseperez, 2019-06-04:
@schweini have you tested this? All POW devices (v1 and v2) report different sizes for the chip and the SDK, check the `info` command output:

```
[760174] [MAIN] Flash size (CHIP)   :  4194304 bytes / 1024 sectors (   0 to 1023)
[760174] [MAIN] Flash size (SDK)    :  1048576 bytes /  256 sectors (   0 to  255)
```

schweini, 2019-06-11:
No idea what I changed, but after a git pull, I can now use my Sonoff POW R2 perfectly with the 3M+1M SPIFFS flash size option in Arduino, and hence do those sweet, sweet OTAs without a problem!

closed 2019-06-11 by schweini

schweini, 2019-06-11:
I have noticed that I can't seem to OTA a unit that by mistake has been flashed with a too big flash size, back to the correct (1M) image. I always get an error 8.