xoseperez/espurna #1745: Support for YSHP0101 COOSA Smart Wifi Plug

membiblio, 2019-05-28:
I'm attempting to add support for a COOSA YSHP0101 Smart Plug which is similar to a BLITZWOLF_BWSHPX_V23 which the espurna binary will control the relay and led 1 but the button, led2 and power monitor do not work. 

Here is the I/O config for the YSHP0101:
HLW8012 SEL is on ESP pin 12 (same as BLITZ)
HLW8012 CF1 is on ESP pin 5 (same as BLITZ)
HLW8012 CF is on ESP pin 4 (same as BLITZ)
Button is on ESP pin 1 (this is TX so pinmode will need to be called)
Relay is on ESP pin 14 (same as BLITZ)
LED1 (Blue) is on ESP pin 13 (same as BLITZ)
LED2 (Red) is on ESP pin 3 (this is RX so pinmode will need to be called)

I have setup and can compile BLITZWOLF_BWSHPX_V23 and have flashed it to the YSHP0101 COOSA plug - it works and I can do OTA. Only the relay and blue light work, nothing else.

Here is a image of the bottom of the ESP8266 which is an TYWE2S module:

[image]

Here is the exterior of the plug itself - I've seen COOSA on the front of other modules - on this one it just says YSHP0101 on the back:

[image]

Here is an image of the STATUS page when connected to 110VAC in the USA you can see it shows 53v instead :

[image]

And to conclude - when I modify hardware.h to copy the BLITZWOLF_BWSHPX #elif defined to another section at the end of the file but before UNSUPPORTED named 'YSHP0101' - when I compile under Arduino - the entire config directory is overwritten so that my changes are reverted back to what the originally were wiping out my new YSHP0101. I'm sure there is some magic incantation I'm missing and would like to ask for assistance please. 

Thank you for a great project and I'm looking forward to being able to contribute a new and working YSHP0101 config section to your code base.

membiblio, 2019-05-29:
I was able to get this to compile, finally. Not sure how just closed arduino ide and then edited in the config directory hardware.h to copy the BLITZWOLF device to a new device named YSHP0101 and then selected that device in arduino.h then opening the arduino ide to build the code. I uploaded it to a device using 2 step update and voila it is working. Thank you to Max (@mcspr) over on the chat who helped by providing some guidance and hand holding.  Next - once I get this device working how do I contribute my changes back to be included in the code base?

mcspr, 2019-05-29:
> Next - once I get this device working how do I contribute my changes back to be included in the code base?

Great! Relevant files are hardware.h, arduino.h and code/platformio.ini
WebUI allows to input expected measurement values and you also can tweak runtime settings directly: `pwrRatioC`, `pwrRatioV`, `pwrRatioP` for current, voltage and power ratios respectively. Defaults are from definitions like HLW8012_CURRENT_RATIO, etc., you get/set them either using telnet of Debug panel: https://github.com/xoseperez/espurna/wiki/Terminal#settings
Voltage resistor settings are build-time only.

And I just checked IDE settings, "save when verifying or uploading" can do the thing you are describing. Just make sure not to edit same file in both IDE and some other editor. Or just disable internal editor altogether.

membiblio, 2019-05-29:
Max - very close to a working device now. Read pwrRatioV, pwrRatioC & pwrRatioP using get from telnet & gui debug window - added those to the new YSHP0101 hardware.h entry. I also set the gpio to be as they should for this hardware which included Button on GPIO 1 (TX) and LED2 on GPIO 3 (RX) which has caused problems - the device does light the red led now but also constantly toggles the relay and never comes up on the wifi - I'm guessing it is resetting - I will have to reflash the stub and another version of the code which does not set GPIO 1 and GPIO 3 until I figure out where to put pinMode(1,FUNCTION_3) and pinMode(3,FUNCTION_3) - 

So my question now is:  Is pinMode the right thing to do? Is FUNCTION_3 the right arg for GPIO to be I/O vs tx and rx?  And where should these pinMode calls go in espurna to support this new YSHP0101 COOSA plug where it does not effect other plugs config/operation? 

Thank you for the advice regarding 'get pwrRatioV' - that made this almost painless :)

mcspr, 2019-05-30:
Try adding `#define DEBUG_SERIAL_SUPPORT 0`. Serial logging is ON by default

membiblio, 2019-05-30:
In arduino.h or in hardware.h by adding to the device specific configuration?

mcspr, 2019-05-30:
Device specific, it shouldn't have serial logging enabled

membiblio, 2019-05-31:
Max - everything except the button works using this config:

I am not sure why the button is not working - perhaps I need to add, somewhere,
pinMode(1, INPUT_PULLUP) - I either need to debug which I will need to think about how
or get advice about what might be awry - pin 1 is the RX pin - debug serial is off - and the converse 
pin 3 TX does faithfully light the red led - interesting note I broke the blue led when prying apart the 
sonically welded plastic case and when I replaced with a 220 ohm resistor the blue led then took
so much current the device reset constantly upping the resistor to 330 works well and the blue 
led intensity is tolerable.

I must leave this alone for a while to do real work to pay bills.

If anyone has any advice for me to try regarding the button please do.

Another question: How does 'findme' mode work? - I searched and did not find anything relevant perhaps my google fu is off today.

#elif defined(YSHP0101)

    // Info
    #define MANUFACTURER                "COOSA"
    #define DEVICE                      "YSHP0101"

    // Buttons
    #define BUTTON1_PIN                 1 // (aka: RX) 
    #define BUTTON1_MODE                BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
    #define BUTTON1_RELAY               1

    // Relays
    #define RELAY1_PIN                  14
    #define RELAY1_TYPE                 RELAY_TYPE_NORMAL

    // LEDs
    #define LED1_PIN                    13    // in GUI this is LED #0 (BLUE)
    #define LED1_PIN_INVERSE            1

    #define LED2_PIN                    3     // (aka: TX) in GUI this is LED #1 (RED)
    #define LED2_PIN_INVERSE            1
    #define LED2_MODE                   LED_MODE_FINDME
    #define LED2_RELAY                  1

    // HJL01 / BL0937
    #ifndef HLW8012_SUPPORT
    #define HLW8012_SUPPORT             1
    #endif

    #define HLW8012_SEL_PIN             12
    #define HLW8012_CF1_PIN             5
    #define HLW8012_CF_PIN              4

    #define HLW8012_SEL_CURRENT         LOW
    #define HLW8012_CURRENT_RATIO       11805
    #define HLW8012_VOLTAGE_RATIO       768228
    #define HLW8012_POWER_RATIO         9631355
    #define HLW8012_INTERRUPT_ON        FALLING

    #define DEBUG_SERIAL_SUPPORT   0
    // Plugs button is on 3 (TX) so no serial please
    // This will also free up 1 for LED2

#endif

mcspr, 2019-06-01:
See an example from a different hardware:
https://github.com/xoseperez/espurna/blob/9ceca44770cc7772afc020ce7bad2d4cf321eadb/code/espurna/config/hardware.h#L3392
BUTTON_SET_PULLUP will be translated into INPUT_PULLUP mode you want:
https://github.com/xoseperez/debounceevent/blob/fdc9c0868c4978ddb66b8775b22b1ec7b29fbbe1/src/DebounceEvent.cpp#L55
(slightly hidden by the fact that it is set by an external library)

Regarding LED modes, [see the wiki](https://github.com/xoseperez/espurna/wiki/Buttons-and-switches#leds)

membiblio, 2019-06-01:
Max - AOK that magic did it. The Coosa YSHP0101 plug is now fully working under Espurna. 

Here is the final configuration file for hardware.h 

#elif defined(YSHP0101)

    // Info
    #define MANUFACTURER                "COOSA"
    #define DEVICE                      "YSHP0101"

    // Buttons
    #define BUTTON1_PIN                 1 // (aka: RX) 
    #define BUTTON1_MODE                BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH | BUTTON_SET_PULLUP
    #define BUTTON1_RELAY               1

    // Relays
    #define RELAY1_PIN                  14
    #define RELAY1_TYPE                 RELAY_TYPE_NORMAL

    // LEDs
    #define LED1_PIN                    13    // in GUI this is LED #0 (BLUE)
    #define LED1_PIN_INVERSE            1

    #define LED2_PIN                    3     // (aka: TX) in GUI this is LED #1 (RED)
    #define LED2_PIN_INVERSE            1
    #define LED2_MODE                   LED_MODE_FINDME
    #define LED2_RELAY                  1

    // HJL01 / BL0937
    #ifndef HLW8012_SUPPORT
    #define HLW8012_SUPPORT             1
    #endif

    #define HLW8012_SEL_PIN             12
    #define HLW8012_CF1_PIN             5
    #define HLW8012_CF_PIN              4

    #define HLW8012_SEL_CURRENT         LOW
    #define HLW8012_CURRENT_RATIO       11805
    #define HLW8012_VOLTAGE_RATIO       768228
    #define HLW8012_POWER_RATIO         9631355
    #define HLW8012_INTERRUPT_ON        FALLING

    #define DEBUG_SERIAL_SUPPORT   0
    // Plugs button is on gpio 1 (RX) so no serial please
    // This will also free up gpio 3 for LED2

#endif

mcspr, 2019-06-01:
I think it is pretty self-explanatory. In case of this device, it will keep the red led on when relay is OFF, and will turn it off when relay is ON again. [The blue led will follow wifi status, blinking in short bursts when disconnected from AP](https://github.com/xoseperez/espurna/blob/9ceca44770cc7772afc020ce7bad2d4cf321eadb/code/espurna/config/defaults.h#L506)

mcspr, 2019-06-18:
Sorry for the delay. I wanted to add your changes to the dev tree, but I've noticed that we already have this exact hw profile as TECKIN_SP22_V14!
https://github.com/xoseperez/espurna/blob/0998110d869264a893f8a0df6eff833620c7a4cb/code/espurna/config/hardware.h#L2812-L2851
We did group these and SHPX (since they are mostly the same), there is even a mention of Coosa brand:
https://github.com/xoseperez/espurna/blob/0998110d869264a893f8a0df6eff833620c7a4cb/code/espurna/config/hardware.h#L2720
👀 So you could've used the teckin-sp22-v14 env all along.

closed 2019-08-25 by stale[bot]