xoseperez/espurna #1747: MQTT -> Publish button status

skyynet, 2019-05-31:
I publish from my Sonoff Dual R2 to my Mosquitto Broker on my Synology.

When I read /home/relay/0 any time I get the current state (0 or 1). This also applies to relay/1.

When I read /home/button/0 I get no result. I see its status is published but would like to read it any time using phpMQTT.

So generally I'd like to see button/0 and button/1 behave like relay/0 and relay/1.

Is this something I can configure somewhere, a bug or a feature request?

mcspr, 2019-05-31:
per https://github.com/xoseperez/espurna/issues/608 i think buttons need to be switches. only then we can reliably say about the current state and enable retained flag for mqtt messages

(also a reminder about? https://github.com/xoseperez/espurna/issues/619)

skyynet, 2019-06-01:
Thanks for the hint. Unfortunately this doesn't help. Subscribing to relay/0 or 1 gives me immediately status "0" indicating the relay is off. Subscribing to button/0 or 1 gives me nothing. When connecting the according pin with gnd it gives me 2,3 or 4.

I would really like to get state "0" for open buttons and something else for closed ones.

mcspr, 2019-06-02:
Right, but that can only happen when espurna sends out mqtt with retain flag. That way, mqtt broker saves the last published message and will send it out to every subscriber on initial connection, until espurna sends something different or deletes it.

Note that https://github.com/xoseperez/espurna/commit/3c2b2017c0f0980da0dcfeeb524b184309724991 did explicitly remove this behaviour:
https://github.com/xoseperez/espurna/blob/9ceca44770cc7772afc020ce7bad2d4cf321eadb/code/espurna/button.ino#L32
We can add a setting, maybe? `btnRetain`, `BUTTON_MQTT_RETAIN` define, 0 or 1, to control this

And idk if you'd want that, but based on config above buttons will still change relay state. When button action is set to NONE, to still receive button events `BUTTON_MQTT_SEND_ALL_EVENTS` needs to be 1 / true
https://github.com/xoseperez/espurna/blob/9ceca44770cc7772afc020ce7bad2d4cf321eadb/code/espurna/button.ino#L103-L105

skyynet, 2019-06-02:
Thanks for the info. Now I know that it's nothing i oversaw but "works as designed".
IMHO it's not intuitive to have the relays send as "retain" and the buttons not.
This might be a pro or con depending on ones projects.
I use a Sonoff Dual R2 to open an close slides on my grass mower's garage. I intend to use the two additional buttons to sense the mower in the garage and the slides being opened or closed.

Thus, inventing a configurable retain feature for the buttons (and relays) would be great.

Your hint regarding button.ino should give me the possibility to change this on my own.

Thanks for your big help and maybe a configurable "retain" might make it into one of the next versions.

mcspr, 2019-06-08:
*mini reminder: github will not send notifications for edits! i just accidentally noticed this in gitter window*

That should've worked.

With small code snippet in terminal setup():
```cpp
terminalRegisterCommand(F("TEST"), [](Embedis* e) {
    mqttSend(MQTT_TOPIC_BUTTON, 0, "1", false, true); // send 1 with retain flag
    delay(100);
    mqttSend(MQTT_TOPIC_BUTTON, 0, "2", false, false); // send 2 without retain
});
```

Running `test` command on the device while subscribed to the button topic:
```
$ mosquitto_sub -v -t '+/+/button/0'
ESPURNA-35A259/button/0 1
ESPURNA-35A259/button/0 2
^C
$ mosquitto_sub -v -t '+/+/button/0'
ESPURNA-35A259/button/0 1
```

skyynet, 2019-06-09:
Thanks. While I'm not that good in Terminal I checked with MQTT Box on MacOS which supports subscribe & publish to topics.

The code modification has worked so far that when I subscribe to the buttons, I receive an initial
`2 (qos : 1, retain : true, cmd : publish, dup : false, topic : /home/garden/mowgli/button/0, messageId : 14, length : 33, Raw payload : 50)`

When I connect button with gnd I always receive a 2 with retain:false which is ok.

Unfortunately there is no 0 sent when I disconnect the bridge. I assumed, with BUTTON_MQTT_SEND_ALL_EVENTS this would also be sent.

This is the only thing which prevents me from using MQTT to check the state of the button.
I don't think my usage so uncommon as a lot of people would like to control something and check its state before and afterwards. So implementing it easy configurable into ESPurna would be great.

mcspr, 2019-06-11:
https://github.com/xoseperez/espurna/blob/0998110d869264a893f8a0df6eff833620c7a4cb/code/espurna/config/types.h#L32-L39
If button is configured as PUSHBUTTON, we receive a bunch of events that depend on connection/disconnection count and their longevity, ref the header values above.
I think there is a slight misunderstanding of what 0 actually is. It only happens in this mode and when we do something [event mapper](https://github.com/xoseperez/espurna/blob/0998110d869264a893f8a0df6eff833620c7a4cb/code/espurna/button.ino#L80-L93) does not understand as any other event type
for example: if event 1 received 4 times in a row, we can't map it to anything and send 0 (NONE)
```
[077427] [BUTTON] Button #0 event 1
[077933] [BUTTON] Button #0 event 2
[079911] [BUTTON] Button #0 event 1
[080143] [BUTTON] Button #0 event 1
[080650] [BUTTON] Button #0 event 3
[081978] [BUTTON] Button #0 event 1
[082262] [BUTTON] Button #0 event 1
[082537] [BUTTON] Button #0 event 1
[082860] [BUTTON] Button #0 event 1
[083368] [BUTTON] Button #0 event 0
```

And if it is SWITCH, we receive event 2 (BUTTON_EVENT_RELEASED /  BUTTON_EVENT_CLICK) when contact is connected or disconnected. So retaining this is of no use.

I do hope I properly understood that the issue here is that we only receive event, not the actual status. One remedy is to have SWITCH timeout after some time, triggering 0 (NONE) event. Another option is to use EventSensor instead of button. See https://github.com/xoseperez/espurna/pull/1771 and [code/espurna/config/sensors.h](https://github.com/xoseperez/espurna/blob/0998110d869264a893f8a0df6eff833620c7a4cb/code/espurna/config/sensors.h#L362) for info about it's configuration

skyynet, 2019-06-16:
Thanks for your help. BUTTONS really don't work like I expected them to.
My aim is to have Buttons send their state like 0 for open and 1 for closed and the MQTT Broker retain the  last sent status.

I read #1771 and sensors.h but couldn't figure out how to configure this for my purpose.

mcspr, 2019-06-16:
EventSensor minimal config would be
```
#define EVENTS_SUPPORT 1
#define EVENTS_PIN <button GPIO>
#define EVENTS_PIN_MODE CHANGE
#define EVENTS_TRIGGER 1
```
But, as noted in the PR, it can be imprecise for this purpose (within interrupt, digitalRead can sometimes return surprising results. and due to Core limitations and compatibility with Arduino API we are kind of stuck with it)
Which is why I made the #1772, where already existing button handler now can ignore events and just report raw status. Naming might need to change though, to avoid mqtt-only meaning.

skyynet, 2019-06-22:
I compiled with the above mentioned EventSensor values with EVENTS_PIN set to 9 which should be BUTTON1 on the Sonoff Dual R2. When I subscribe to MQTT I see event being 0. When I connect BUTTON1 with GND nothing happens. Compiling with EVENTS_PIN set to 0 also nothing happens when connecting BUTTON0 with GND.

And idea? Also how do I configure to check for BUTTON1 and BUTTON0?

Thanks.

mcspr, 2019-06-22:
EVENTS_PIN_MODE needs to be INPUT_PULLUP with some pins (ref button modes that sets it as such)
Have you tried #1772?

skyynet, 2019-06-22:
That's my not working config:

```
#define EVENTS_SUPPORT                  1       // Do not build with counter support by default
#define EVENTS_TRIGGER                  1       // 1 to trigger callback on events,
#define EVENTS_PIN                      9       // GPIO to monitor
#define EVENTS_PIN_MODE                 INPUT_PULLUP   // INPUT, INPUT_PULLUP, CHANGE
#define EVENTS_INTERRUPT_MODE           CHANGE  // RISING, FALLING, CHANGE
#define EVENTS_DEBOUNCE                 50      // Do not register events within less than 50 millis
```

As I'm not good at all in patching source code from github I'll wait for #1772 to be merged into the code. So I havn't tried, yet.

mcspr, 2019-06-27:
The events sensor "event" is a logical read. So connecting GND would give 0 and 3v3 - 1.

#1772 configuration proposes additional "reporting mode", so that each button can be either sending events or pressed status. I do wonder it bitmasks are needed though and that can just be some global setting like BUTTON_MQTT_SEND = BUTTON_MQTT_SEND_PRESSED, as it was in the 515e6607914ac92ffcff27fc6caa7cd7522f559a

closed 2019-09-02 by stale[bot]