xoseperez/espurna #1748: Cant reconnect to router after random disconnect from router.

kiralikbeyin, 2019-05-31:
**Bug description**
Cant reconnect to router after random disconnect from router.
*A clear and concise description of what the bug is.*
RF bridge disconnects from router randomly (1-48 hours) but doesnt reconnect to router and begins AP mode.
**Steps to reproduce**
No idea
**Expected behavior**
Auto reconnect to router 
**Device information**

[252618] [MAIN] EVAK 1.13.205
[252619] [MAIN] info@evakilli.com
[252620] [MAIN] http://EvAkilli.com

[252621] [MAIN] CPU chip ID: 0xCACDE6
[252622] [MAIN] CPU frequency: 80 MHz
[252625] [MAIN] SDK version: 1.5.3(aec24ac9)
[252626] [MAIN] Core version: 2.3.0
[252627] [MAIN] Core revision: 159542381
[252630] 
[252632] [MAIN] Flash chip ID: 0x144051
[252640] 
[252647] [MAIN] Flash size (SDK)    :  1048576 bytes /  256 sectors (   0 to  255)
[252656] [MAIN] Reserved            :    16384 bytes /    4 sectors ( 252 to  255)
[252661] [MAIN] EEPROM sectors: 251, 250
[252667] [MAIN] EEPROM:  4096 bytes initially |  3538 bytes used (86%) |   558 bytes free (13%)
[252674] 
[252681] [MAIN] Last reset info: Fatal exception:4 flag:1 (WDT) epc1:0x40104ee8 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000
[252689] [MAIN] WebUI image: RFBRIDGE
[252697] [MAIN] Power: 3154 mV
[301645] [MAIN] Uptime: 301 seconds
[301649] [MAIN] Heap  : 33376 bytes initially | 17240 bytes used (51%) | 16136 bytes free (48%)
[301651] [MAIN] Power: 3155 mV
[301654] [MAIN] Time: 2019-05-31 18:52:52

**Tools used**
* *Desktop operating system*
Mac High Sierra 10.13.6 
* *Browser & version*
Chrome latest
* *IDE & version* 
Arduino 1.8.2
* *Compiler & version (if not embedded in IDE)*
2.3.0

mcspr, 2019-05-31:
Is it the same when wifi scanning is disabled? (`wifiScan` settings key)

kiralikbeyin, 2019-05-31:
Yes.. I am trying to solve this problem since May 15.

mcspr, 2019-05-31:
Does it reconnect after `wifi.reset` terminal command?
Do you use WebUI or telnet frequently with rfbridge? I do see that there is a check for connected clients in [wifiReconnectCheck()](https://github.com/xoseperez/espurna/blob/9ceca44770cc7772afc020ce7bad2d4cf321eadb/code/espurna/wifi.ino#L580-L589). Perhaps, it stalls reconnection even without active websocket or telnet clients.

*btw I did see the edit about the verision. lets mention that the version we are testing is 1.13.6-dev @ 820d8c4be8af5cb0be1b94187a9c0ed0f6bbbb86*

kiralikbeyin, 2019-06-01:
wifi.reset didint work.

````
[043408] [WEBSOCKET] Requested action: dbgcmd
[043411] [WIFI] RESET 1
[043412] [WIFI] RESET 11
[043414] [WIFI] RESET 111
[043415] [WIFI] RESET 112
````
code
````
void _wifiConfigure() {

  DEBUG_MSG_P(PSTR("[WIFI] RESET 11\n"));

  jw.setHostname(getSetting("hostname").c_str());
  DEBUG_MSG_P(PSTR("[WIFI] RESET 111\n"));

#if USE_PASSWORD
  jw.setSoftAP(getSetting("hostname").c_str(), getAdminPass().c_str());
  DEBUG_MSG_P(PSTR("[WIFI] RESET 112\n"));

#else
  jw.setSoftAP(getSetting("hostname").c_str());

  DEBUG_MSG_P(PSTR("[WIFI] RESET 113\n"));

#endif
  jw.setConnectTimeout(WIFI_CONNECT_TIMEOUT);
  DEBUG_MSG_P(PSTR("[WIFI] RESET 114\n"));

  wifiReconnectCheck();
  DEBUG_MSG_P(PSTR("[WIFI] RESET 115\n"));

  jw.enableAPFallback(WIFI_FALLBACK_APMODE);
  DEBUG_MSG_P(PSTR("[WIFI] RESET 116\n"));

  jw.cleanNetworks();
  DEBUG_MSG_P(PSTR("[WIFI] RESET 117\n"));

  DEBUG_MSG_P(PSTR("[WIFI] RESET 12\n"));

  _wifi_ap_mode = getSetting("apmode", WIFI_AP_FALLBACK).toInt();

  // If system is flagged unstable we do not init wifi networks
#if SYSTEM_CHECK_ENABLED
  if (!systemCheck()) return;
#endif

  DEBUG_MSG_P(PSTR("[WIFI] RESET 13\n"));

  // Clean settings
  _wifiClean(WIFI_MAX_NETWORKS);

  int i;
  for (i = 0; i < WIFI_MAX_NETWORKS; i++) {
    if (getSetting("ssid" + String(i)).length() == 0) break;
    if (getSetting("ip" + String(i)).length() == 0) {
      jw.addNetwork(
        getSetting("ssid" + String(i)).c_str(),
        getSetting("pass" + String(i)).c_str()
      );
    } else {
      jw.addNetwork(
        getSetting("ssid" + String(i)).c_str(),
        getSetting("pass" + String(i)).c_str(),
        getSetting("ip" + String(i)).c_str(),
        getSetting("gw" + String(i)).c_str(),
        getSetting("mask" + String(i)).c_str(),
        getSetting("dns" + String(i)).c_str()
      );
    }
  }
  DEBUG_MSG_P(PSTR("[WIFI] RESET 14\n"));

  jw.enableScan(getSetting("wifiScan", WIFI_SCAN_NETWORKS).toInt() == 1);

  unsigned char sleep_mode = getSetting("wifiSleep", WIFI_SLEEP_MODE).toInt();
  sleep_mode = constrain(sleep_mode, 0, 2);
  DEBUG_MSG_P(PSTR("[WIFI] RESET 15\n"));

  WiFi.setSleepMode(static_cast<WiFiSleepType_t>(sleep_mode));  
}
````

i think ```  jw.setConnectTimeout(WIFI_CONNECT_TIMEOUT); ``` is making trouble.

kiralikbeyin, 2019-06-01:
wifi reset when in AP mode

````
[118255] [WEBSOCKET] Requested action: dbgcmd
[118260] [WIFI] RESET 1
[118264] [WIFI] RESET 11
[118268] [WIFI] RESET 111
[118272] [WIFI] RESET 112
[118275] [WIFI] RESET 114
[118278] [WIFI] RESET 115
[118281] [WIFI] RESET 116
[118285] [WIFI] RESET 117
[118288] [WIFI] RESET 12
[118290] [WIFI] RESET 13
[118308] [WIFI] RESET 14
[118312] [WIFI] RESET 15
[118316] [WIFI] RESET 10
[118320] [WIFI] RESET 21
[118594] [WIFI] Disconnected
[118599] [WIFI] RESET 20
[118601] +OK
[118602] [WIFI] RESET 30
[118767] [WIFI] Scanning
[120896] [WIFI]     BSSID: 6C:F xxx 1 RSSI: -54 SEC: OPEN SSID:  WiSpotter
[120899] [WIFI]     BSSID: A2:5 xxx  1 RSSI: -73 SEC: WPA2 SSID: AyseBilge iPhone’u
[120902] [WIFI]     BSSID: FC:4A:E9:   1 RSSI: -91 SEC: AUTO SSID: VODAFONE_262C
....

[120932] [WIFI] No known networks found
```

kiralikbeyin, 2019-06-01:
Debug screen stops at 
````
[023595] [WEBSOCKET] Requested action: dbgcmd
[023604] [WIFI] RESET 1
[023606] [WIFI] RESET 11
[023608] [WIFI] RESET 111

`````

telnet stops at:
````
[113375] [WEBSOCKET] Requested action: dbgcmd
[113379] [WIFI] RESET 1
[113380] [WIFI] RESET 11
[113381] [WIFI] RESET 111
[113383] [WIFI] RESET 112
[113387] [WIFI] RESET 114
[113388] [WIFI] RESET 115
[113389] [WIFI] RESET 116
[113390] [WIFI] RESET 117
[113391] [WIFI] RESET 12
[113393] [WIFI] RESET 13
[113407] [WIFI] RESET 14
[113410] [WIFI] RESET 15
[113413] [WIFI] RESET 10
[113416] [WIFI] RESET 21
````

21 is:
````
void wifiDisconnect() {
  DEBUG_MSG_P(PSTR("[WIFI] RESET 21\n"));

  jw.disconnect();
}
````

why debug screen stops at ```[023608] [WIFI] RESET 111```
 i cant understand.

kiralikbeyin, 2019-06-01:
when it is connected to router and wifi.ap, it starts in AP mode and didnt connect to router

kiralikbeyin, 2019-06-01:
i think this will help 

````
void wifiReconnectCheck() {
  bool connected = false;

  if (WiFi.status() == WL_CONNECTED)
  {
    connected = true;
  }

  if (!WiFi.isConnected())
  {
    connected = false;
  }
  
  //#if WEB_SUPPORT
  //  if (wsConnected()) connected = true;
  //#endif
  //#if TELNET_SUPPORT
  //  if (telnetConnected()) connected = true;
  //#endif
  DEBUG_MSG_P(PSTR("[WIFI] wifiReconnectCheck() \n"));

  jw.setReconnectTimeout(connected ? 0 : WIFI_RECONNECT_INTERVAL);
  DEBUG_MSG_P(PSTR("[WIFI] %u \n"), connected);

}

````

I reconnects to router if connection drops and changes to AP mode..

mcspr, 2019-06-01:
@kiralikbeyin You missed my original question about the websockets or telnet, but I hope you had noticed their purpose. Why I am asking is possibility of improperly closed connections.

Note that it will reconnect eventually, if neither web or telnet is connected. That is the intended behaviour.
Whenever telnet client is disconnected:
https://github.com/xoseperez/espurna/blob/9ceca44770cc7772afc020ce7bad2d4cf321eadb/code/espurna/telnet.ino#L44
Whenever ws is disconnected:
https://github.com/xoseperez/espurna/blob/9ceca44770cc7772afc020ce7bad2d4cf321eadb/code/espurna/ws.ino#L442
Which may not ever happened, based on the bug description.

By your proposal, it will drop any connected clients immediately, because softAP will be brought down.

mcspr, 2019-06-01:
And if by `wifi.ap` you mean terminal command, it think it is missing a `wifi.sta` counterpart, because it disables STA mode completely within justwifi.
https://github.com/xoseperez/espurna/blob/9ceca44770cc7772afc020ce7bad2d4cf321eadb/code/espurna/wifi.ino#L551-L562

closed 2019-08-08 by stale[bot]