xoseperez/espurna #1760: Sonoff Basic R2, which binary?

darkoddio, 2019-06-05:
Which binary do I need to use to flash a Sonoff Basic R2 if I don't want to use any sensor?
[image]

davebuk, 2019-06-09:
I'm pretty sure the standard Sonoff-basic will work. I think the only difference is the GPIO pin assignment is different if you want to use the internal solder pads for an external switch (IO 2 instead of 14).

xoseperez, 2019-06-09:
@davebuk is right. The only difference is where the sensor is attached so in your case you can safely flash the standard sonoff basic image.

closed 2019-06-09 by xoseperez

darkoddio, 2019-06-09:
thanks guys, that's exactly what I did few days ago :)
I don't plan to connect any sensor/switch to it

clueo8, 2019-09-03:
I successfully got Espurna on one of these new Sonoff Basic units, but when I tried to connect a Dallas temp sensor to the IO 2 pad, it doesn't power on any more, it seems like its dead... Won't power on anymore... Any idea what could have happened or how to diagnose?

davebuk, 2019-09-03:
My guess is you've bridged the pad to another line causing a short? Or maybe the sensor is holding IO2 to ground and forcing boot mode.

You could take the sensor off and see if it starts working again.

clueo8, 2019-09-03:
I did remove the sensor, but it fails to power on anymore with mains and FTDI doesn't recognize it anymore.  Didn't know if it's stuck in a strange mode? Or dead...

davebuk, 2019-09-03:
I guess you have soldered header pins to the board. Make sure they haven't joined together with a stray bit of solder. 

If the LED is not coming on at all, powered either from your FTDI or mains power, it might unfortunately be dead. I don't recommended it but it if you have the technical knowledge you could check if the mains supply is being converted to 3v DC. SAFETY NOTICE! Obviously remember mains voltage can kill. 

I had one device once that powered up but wouldn't run or flash a firmware. I left it in a cupboard for a few weeks and when I came to try it again it worked and could flash it. 😁