xoseperez/espurna #1762: show config topic field blank when code uploaded from platformio

nilava, 2019-06-05:
The show config in HASS is not working properly when i am uploading code from platform io instead of using the prebuilt firmware for nodemcu lolin.

```
switch:
  - name: ESPURNA_1F7B48
    platform: mqtt
    state_topic: 
    command_topic: 
    payload_on: 1
    payload_off: 0
    availability_topic: 
    payload_available: 1
    payload_not_available: 0
 ```

This is what i am getting where as you can see state_topic, command_topic and availability_topic is blank

xoseperez, 2019-06-09:
What is your MQTT configuration?

nilava, 2019-06-12:
What is my MQTT Config as in?

xoseperez, 2019-06-12:
Can you copy paste here the output for the `keys` command in the debug console (removing private data)?

jyz0501, 2019-08-30:
payload_on: 1
payload_off: 0

It should be written like this

    payload_on: "ON"
    payload_off: "OFF"

The  show  config  is  wrong！

mcspr, 2019-08-30:
@jyz0501 please see #1883 
If you use current `dev` build and you do see `ha.config` outputting mqtt toic, I believe this is no longer a bug, as it should've been fixed by the #1755

PR #1885 (2019-09-03): Home Assistant updates
- remove "platform" key, see #1440. this implicitly sets schema to "basic". pending some other clean-up regarding json and mqtt queueing, other schema can be added down the line 
- updated ws module queue elem to capture callbacks list, allows to pass more than one callback (for example, when they are generated on the fly as lambdas, see ha wsPost usage)
- modified method to send ha config to use global ws queue, fix #1762 problem with empty topics and ensure json allocation is consistent.
- use existing defines to set mqtt payload options. amend https://github.com/xosep [... cut]
commit 13c1a193 (2019-09-03): Home Assistant updates (#1885)

- remove "platform" key, see #1440. this implicitly sets schema to "basic". pending some other clean-up regarding json and mqtt queueing, other schema can be added down the line 
- updated ws module queue elem to capture callbacks list, allows to pass more than one callback (for example, when they are generated on the fly as lambdas, see ha wsPost usage)
- modified method to send ha config to use global ws queue, fix #1762 problem with empty topics and ensure json allocation is consistent.
- use existing defines to set mqtt payload options. amend #1085, #1188, # [... cut]
closed 2019-09-03 by mcspr