xoseperez/espurna #1763: Domoticz H801 RGBWW

physke, 2019-06-07:
Hi,

I've managed to get espurna on a H801 working with domoticz thru MQTT for RGB and brightness settings, but not for warm/cold white LEDs. Looks like domoticz is sending cmode=2, but there is nothing in the espurna code for cmode=2. only 3 & 4 (ColorModeRGB & ColorModeCustom). Is this an issue with domoticz or espurna or the way I have it configured?

**domoticz setting for CW/WW:**
[image]

**received by espurna:**
[image]

**domoticz setting for RGB:**
[image]

**received (and processed) by espurna:**
[image]

Thanks
Jason

physke, 2019-08-08:
;(

mcspr, 2019-08-09:
As you have noted, domoticz module specifically filters out any other cmode:
https://github.com/xoseperez/espurna/blob/274f71d7d9047bb2bdf6b5c2cca7a6ff9d63142b/code/espurna/domoticz.ino#L61

While we do see the problem... what is the difference between cmodes? Should we care at all about them when receiving values?
I could not find anything about them
https://www.domoticz.com/wiki/MQTT
https://www.domoticz.com/wiki/Special:AllPages
...and google searches in general, since I do not use Domoticz

Only reference is by reading ColorSwitch.h, counting from 0 this is cmode=2:
https://github.com/domoticz/domoticz/blob/22ee3f8b37158f20a62e345d410055a5cd9bd250/hardware/ColorSwitch.h#L18-L28
> ColorModeTemp,     // White with color temperature. Valid fields: t

mcspr, 2019-08-27:
I see. So it needs to be mapped from 0...255 into kelvin / mired values. Ranges are 2000...6536 kelvin and 153...500 mired, based on the source here.

2nd link has range of 0...100 for kelvin instead of 0...255, but that does not apply here.

physke, 2019-08-27:
Found this:

br, = brightness 0..100
cw, = cold white Range:0..255, Cold white level
ww, = warm white Range:0..255, Warm white level
m, = mode
ColorModeNone = 0, // Illegal
ColorModeWhite = 1, // White. Valid fields: none
ColorModeTemp = 2, // White with color temperature. Valid fields: t
ColorModeRGB = 3, // Color. Valid fields: r, g, b.
ColorModeCustom = 4, // Custom (color + white). Valid fields: r, g, b, cw, ww, depending on device capabilities
ColorModeLast = ColorModeCustom,
t = temperature 0..255, Color temperature (warm / cold ratio, 0 is coldest, 255 is warmest) 

and https://www.domoticz.com/wiki/Domoticz_API/JSON_URL's#Set_an_RGB_CW_WW_or_CW_WW_light_to_a_certain_color_temperature

commit ec259767 (2019-08-30): domoticz: handle cmode=2

resolve #1763
closed 2019-08-30 by mcspr