xoseperez/espurna #1767: websocket port

kiralikbeyin, 2019-06-09:
I can get half of http server data but i cant get WS datas from remote. 

If i connect to my device which is forward to ...duckdns.org with the same wifi network everything is ok.

Maybe i have to open more port in my router for ws?
(using 80 as default)

mcspr, 2019-06-11:
Does this sound like #1610 ? If yes, see the lwip patch comments there and the attached PR. Or you can try using recent Core releases instead of the 2.3.0

In the future, please write out the version you are using (release bin, self-built) and if it is not from releases - which Core version was used and what build method was used (PlatformIO, ArduinoIDE, Makefile etc.). As it was specified in the template.
It was an annoying bug, so it was a simple guess to make.

kiralikbeyin, 2019-06-12:
I am now using 2.5.2 and ws works fine for remote. (before 2.3.0 which makes trouble) Thanks @mcspr 

Is there any disabilty using 2.5.2?

mcspr, 2019-06-13:
Not really, besides total firmware size.

You may experience some issues regarding active async webserver and doing anything with wificlient (per https://github.com/xoseperez/espurna/pull/1716#issuecomment-496698176 and #1620), as I've seen some weird timeout problems there. It is solved by scheduler patches in current esp8266/Arduino master, you can pick those patches individually if that becomes a problem.

closed 2019-06-13 by mcspr