xoseperez/espurna #1773: Adding two DHT sensor to Sonoff Pow R2

JavierAder, 2019-06-11:
Is that possible? I need keep track 2 temperatures (plus current, power, etc) and my  Sonoff Pow R2 v2.0  seems to have two free pins in j2 connector (these are named IO4 and IO5, I think they are pins GIPIO 14 and and GPIO 15, but I am not sure).

Can someone guide me? I think the procedure for only one sensor DHT would be to add a custom.h in the config folder; something like 

`#define` DHT_SUPPORT 1
`#define` DHT_PIN 14
`#define` DHT_TYPE   DHT_CHIP_DHT22

and then build for Sonoff pow r2.
Is it would be correct? What about add two sensor DHT in GIPIO 14 and 15?
[image]

JavierAder, 2019-06-12:
I answer myself (well, the code actually does).
sensor.ino

```
/*
    // Example on how to add a second DHT sensor
    // DHT2_PIN and DHT2_TYPE should be defined in sensors.h file
    #if DHT_SUPPORT
    {
        DHTSensor * sensor = new DHTSensor();
        sensor->setGPIO(DHT2_PIN);
        sensor->setType(DHT2_TYPE);
        _sensors.push_back(sensor);
    }
    #endif
*/
```

Ok, I think it's better to do this in custom.h.
I will try the following custom.h, BUT I don't know if pin number are right.... Someone knows? IO4 and IO5 in the previous image are connected to GPIO4 and GPIO5, but internally they are pins 16 and 24. What convention is right? 

```
//example of custom.h for support two DHT sensors (chip type 22) in Sonoff Pow R2 v2.0
//(It needs uncomment section in sensor.ino for support second sensor)

//general support
#define DHT_SUPPORT 1
//support first sensor in pin GPIO4 (PIN 16)
#define DHT_PIN 16
#define DHT_TYPE DHT_CHIP_DHT22
//support second DHT sensor in pin GPIO5 (PIN 24)
#define DHT2_PIN 24
#define DHT2_TYPE DHT_CHIP_DHT22

```

mcspr, 2019-06-17:
ESP8266 GPIO numbers are 0 to 16. Where did you get GPIO 4 == PIN 16 from? Mapping is pretty easy, GPIO 4 is just number 4
(technical reference .pdf, ["ESP8266EX has 17 GPIO pins"](https://www.espressif.com/sites/default/files/documentation/esp8266-technical_reference_en.pdf))

I would strongly advise against doing this though, because sonoff does not isolate IO pins. GND pin is linked with mains. Can't reference any issue here, but the here's the first search result from the Sonoff-Tasmota:
https://github.com/arendst/Sonoff-Tasmota/issues/4732

JavierAder, 2019-06-30:
Hi, sorry by delay; I din't see your answer.

> ESP8266 GPIO numbers are 0 to 16. Where did you get GPIO 4 == PIN 16 from? 

from the same document:

> GPIO4                            pin16                   GPIO4_U

Physically GPIO4 is pin 16; but, thinking this a bit better, obviously one has to use the logical number of the pin, not the phisycal one :) ....

> I would strongly advise against doing this though, because sonoff does not isolate IO pins. GND pin is linked with mains. Can't reference any issue here, but the here's the first search result from the Sonoff-Tasmota:

Good point...I already knew it but I had not meditated to much about that... In refrigerators ocurrs the same (that is where I use sonoff, to analyze its operation); they generally use non-isolated power source, and sensors (NTC in general) are at the same potential as the main lines; but in this case, users do not manipulate these.
GPIO probably do not represent a serious hazard (low current) but GND or 3.3V do....
What about use a isolated connection to  a independent powered (and isolated) DHT? I think in optocouplers and a little isolated 3.3v power source (I did see similar situation in refrigerators, but related to the display/touchs screen/buttons, which makes sense ), but well, it is a hardware/electronic problem, not directly related to Espurna; I am running of topic....

Anyway, I will try first get a version of Espurna supporting Pow R2 and two DHT and then look for a electronic solution to the isolation.

closed 2019-09-05 by stale[bot]