xoseperez/espurna #1776: Support for these remotes

TurboTronix, 2019-06-15:
Tried Tasmota and now Espurna but nothing seems to be working. I don't even know if these types of remotes are supported:

These are my 2 remotes (the 315Mhz is for a projector curtain, and the other is for RGB pool lights)
[image]
[image]

TurboTronix, 2019-06-15:
Funny thing I just noticed, I made Espurna learn my 433.92mhz remote codes (on/off is the same button on the remote so made the ON/OFF in Espurna the same code: C00101B31800557FC0). When I press the ON/OFF button on my remote it changes the switch status is Espurna from On to Off and vise versa but it doesn't change my light from ON/OFF only my actual remote does.

TurboTronix, 2019-06-15:
This is what happens when I press the buttons via Espurna (button switch0 pressed on and off, and button switch1 pressed on and off):
```
[924386] [WEBSOCKET] Requested action: relay
[924389] [RELAY] #0 scheduled ON in 0 ms
[924392] [RELAY] #0 set to ON
[924395] [RF] Enqueuing MESSAGE 'C00101B41800557FC0' 1 time(s)
[924399] [MQTT] Sending ESPURNA-CD02DC/relay/0 => 1 (PID 1)
[925402] [RELAY] Setting relay mask: 1
[925582] [WEBSOCKET] Requested action: relay
[925585] [RELAY] #0 scheduled OFF in 0 ms
[925589] [RELAY] #0 set to OFF
[925592] [RF] Enqueuing MESSAGE 'C00101B41800557FC0' 1 time(s)
[925596] [MQTT] Sending ESPURNA-CD02DC/relay/0 => 0 (PID 1)
[926601] [RELAY] Setting relay mask: 0
[926700] [WEBSOCKET] Requested action: relay
[926703] [RELAY] #1 scheduled ON in 0 ms
[926705] [RELAY] #1 set to ON
[926709] [RF] Enqueuing MESSAGE 'C00101B51800557F30' 1 time(s)
[926713] [MQTT] Sending ESPURNA-CD02DC/relay/1 => 1 (PID 1)
[927718] [RELAY] Setting relay mask: 2
[927740] [WEBSOCKET] Requested action: relay
[927742] [RELAY] #1 scheduled OFF in 0 ms
[927745] [RELAY] #1 set to OFF
[927748] [RF] Enqueuing MESSAGE 'C00101B51800557F30' 1 time(s)
[927752] [MQTT] Sending ESPURNA-CD02DC/relay/1 => 0 (PID 1)
[928758] [RELAY] Setting relay mask: 0
```

And this via the original remote(Button pressed On then Off, and button Mode pressed twice):
```
[973305] [RF] Received code: 00557FC0
[973308] [RF] Action 0xA4
[973309] [RF] Received message 'C00101B51800557FC0'
[973313] [RF] Trying to match code 557FC0
[973315] [RF] Match ON code for relay 0
[973318] [RF] Match OFF code for relay 0
[973320] [RF] Matched message 'C00101B41800557FC0'
[973322] [RELAY] #0 scheduled ON in 0 ms
[973326] [MQTT] Sending ESPURNA-CD02DC/rfin => C00101B41800557FC0 (PID 1)
[973333] [RELAY] #0 set to ON
[973337] [MQTT] Sending ESPURNA-CD02DC/relay/0 => 1 (PID 1)
[974342] [RELAY] Setting relay mask: 1
[979136] [RF] Received code: 00557F30
[979138] [RF] Action 0xA4
[979140] [RF] Received message 'C00101B41800557F30'
[979143] [RF] Trying to match code 557F30
[979147] [RF] Match ON code for relay 1
[979151] [RF] Match OFF code for relay 1
[979154] [RF] Matched message 'C00101B51800557F30'
[979158] [RELAY] #1 scheduled ON in 0 ms
[979162] [MQTT] Sending ESPURNA-CD02DC/rfin => C00101B51800557F30 (PID 1)
[979169] [RELAY] #1 set to ON
[979176] [MQTT] Sending ESPURNA-CD02DC/relay/1 => 1 (PID 1)
[980181] [RELAY] Setting relay mask: 3
```

TurboTronix, 2019-06-16:
How can i send rfout from command line?

TurboTronix, 2019-06-16:
Ok i managed to send an rfout command via hassio but it doesn't control my light. Can't figure out if the rf bridge transmit any rf signal, is there a way to see if the rf bridge send anything out?

sesam, 2019-06-23:
Using a cheap SDR (soft defined radio, $9 and up) you can monitor RF activity to record and replay RF remores and more.

pelson, 2019-07-16:
Similar situation here. I have the "Brennenstuhl Comfort-Line RCR CE1 1011 IP20 FR 3726" sockets and "Brennenstuhl Comfort-Line RCT CE1 CR 3726" remote. When I press a button I get one of 4 or 5 different codes, but each button press is definitely not unique (so not a rolling code FWICT). Based on no evidence at all, I suspect this is using the Intertechno v3 protocol - I'm not entirely certain how to confirm this though.

I've got a couple of sonos RF bridges, both with direct bypass ([RFB_DIRECT](https://github.com/xoseperez/espurna/wiki/Hardware-Itead-Sonoff-RF-Bridge#modifications)) and am happy to experiment with the bridges if there are suggestions for better understanding the devices.