xoseperez/espurna #1777: ANALOG_SUPPORT breaks relays

lblabr, 2019-06-16:
i'm using Arduino IDE to compile espurna firmware, using 1.13.6-dev.
Board Wemos mini / compatible to Wemos mini.

Enabling Analog_Support will break relay functions. Where do i have to start to figure out what happens ?

#elif defined(BEREGNUNG)

    // Info
    #define MANUFACTURER        "WEMOS"
    #define DEVICE              "BEREGNUNG"

    // Disable non-core modules
    #define ALEXA_SUPPORT           0
    #define BROKER_SUPPORT          0
    #define BUTTON_SUPPORT          0
    #define DOMOTICZ_SUPPORT        0
    #define HOMEASSISTANT_SUPPORT   0
    #define I2C_SUPPORT             0
    #define MDNS_SERVER_SUPPORT     0
    #define MQTT_SUPPORT            1
    #define NTP_SUPPORT             1
    #define SCHEDULER_SUPPORT       1
    #define THINGSPEAK_SUPPORT      0
    #define WEB_SUPPORT             1

    // Extra light-weight image
    #define DEBUG_SERIAL_SUPPORT    0
    #define DEBUG_TELNET_SUPPORT    0
    #define DEBUG_WEB_SUPPORT       1
    #define LED_SUPPORT             0
    #define TELNET_SUPPORT          0
    #define TERMINAL_SUPPORT        1	

//    #define SENSOR_SUPPORT          1
	#define ANALOG_SUPPORT			1
	
    #define RELAY1_PIN          2
    #define RELAY1_TYPE         RELAY_TYPE_INVERSE
    #define RELAY2_PIN          0
    #define RELAY2_TYPE         RELAY_TYPE_INVERSE
    #define RELAY3_PIN          4
    #define RELAY3_TYPE         RELAY_TYPE_INVERSE
    #define RELAY4_PIN          5
    #define RELAY4_TYPE         RELAY_TYPE_NORMAL
	
//    #define I2C_SDA_PIN         12  // D2
//    #define I2C_SCL_PIN         14  // D1
	
#endif

mcspr, 2019-06-23:
Note that I don't use ANALOG_SUPPORT, just an observation based on code.

https://github.com/xoseperez/espurna/blob/0998110d869264a893f8a0df6eff833620c7a4cb/code/espurna/sensors/AnalogSensor.h#L72
As you can see, it hard-codes pin 0 as INPUT. While relays would want to write to it.
TBH, based on documentation, I don't quite understand how analogRead works on GPIO0 instead of ADC pin

mcspr, 2019-08-22:
Fixed the comment above in the https://github.com/xoseperez/espurna/commit/c15d87a517e56c6f85d24ca5d94310f8ee7fe4a3, closing.

closed 2019-08-22 by mcspr

lblabr, 2019-10-26:
with dev works like a charme, thanks!!