xoseperez/espurna #1782: Can I use custom.h in PlatformIO terminal/command line?

davebuk, 2019-06-17:
I have been using a custom.h file for some of my Sonoff devices and have modified platformio.ini with the `-DUSE_CUSTOM_H` and built using the environment selection in the GUI. If I update the code to pull in espurna updates, I have to revert the platformio.ini changes, update and then modify the .ini file again.

Using `pio run -e itead-sonoff-basic` from the terminal works, but should I be able to run something like: `pio run -e itead-sonoff-basic -DUSE_CUSTOM_H` so I don't have to modify the .ini file?

mcspr, 2019-06-19:
See [`PLATFORMIO_BUILD_FLAGS`](https://docs.platformio.org/en/latest/envvars.html#building)
And you might want to check out [`git-stash`](https://git-scm.com/docs/git-stash) feature too.

davebuk, 2019-06-21:
Thanks, I was trying to work out the git stash feature but some of the documentation is a bit daunting for a noob! I'll give it a go over the next few weeks.

davebuk, 2019-06-22:
Got it working. I had to register as a user first using `stash config` but I was then able to do what I wanted.

closed 2019-06-22 by davebuk