xoseperez/espurna #1783: Sensor developement for Oregon Scientific 433MHz sensors

oscarsan1, 2019-06-18:
Hi, I'm programing a sensor for Oregon Scientific Sensors with a 433MHz receiver. 
The problem is that it doesn't know how many sensors there are and which type until the signal from the sensor is received, and this may take several seconds, until 30 segons.

Then, when the "void _sensorInit() {..." is called for discover the magnitudes at the begining after reset the number of magnitudes is unkown. 
Is there any way to update the magnitudes numbers and types on the fly as it was detected?

Also if any one can help me to upload the code of this sensors in developement branch in github I would appreciate it

mcspr, 2019-06-23:
You can check out https://github.com/xoseperez/espurna/pull/1304 implementation for PZEM, where it defines each sensor reading with an offset whenever there are more than one device it offsets magnitude index by COUNT constant.
So you mark sensor as not-ready (...->ready()) before appending magnitudes.
You can also add sensor method to set how many devices are there (or set addresses list and count that?). getSetting() / setSetting() can be used in _sensorLoad() for this specific sensor

oscarsan1, 2019-06-23:
I've try to use ->ready() for waiting to detect sensors, but as there is no magnitudes the tick() is not called in sensor loop "     if (_magnitudes.size() == 0) return;", then the sensors without tick() are not detected. 

Finaly I've developed a static sensor id's, not detected online.  The user must program this sensors address in the code.

`#ifndef OOK_DECODER_SENSORS_ADDR
#define OOK_DECODER_SENSORS_ADDR            {0xEA4C3502, 0x1A2D4701, 0x1A2D1F03}  // 0xMMMMIICC
#endif                                                      // MMMM->model II->Id(change every battery replacement) CC->channel
`

mcspr, 2019-06-23:
You don't need magnitudes in tick directly. Just wait up with some internal representation of them and add them all at once, just when all of them are discovered.
Paired with get/setSetting API, sensor finds addresses and updates _count so each magnitude can be represented sequentially. BAT1, T1, HR1, BAT2, T2, HR2 ... and so on

oscarsan1, 2019-06-24:
I need some processing in tick() for avoiding overload time in interrupt function. 

Finaly I modify the sensor loop for calling tick in case of this sensor is active:

`    #ifndef OOK_DECODER_SUPPORT
    if (_magnitudes.size() == 0) return;
    #endif`
Then, there is a initial time for detecting sensors. All sensors detected in this time are registered. After that no more sensors are registered. Also user can add static sensors in the code. For this static sensors you are sure about order you defined it and then which sensor is. 

In my case, I detect a lot of sensors from my neighbours.

[image]

Please some instructions to upload the code in the developement branch?

mcspr, 2019-06-30:
My intent there was to separate things a bit. It is clear that we want to set magnitudes, it's just you modify "internal" part of the sensor module going around the intended public methods.

But, I would still like to see the code before we continue this :) If you did not use git to store your changes, consider starting a clean git-clone and copying your files there. Then, just add sensor files and your modifications to the espruna files

oscarsan1, 2019-07-02:
I have some troubles with github. Is the first time I use it. I've clone repository to local github desktop windows, make changes in local, but try to comit?. How can sicronize my changes to the public?

mcspr, 2019-07-02:
@oscarsan1 when you open the https://github.com/oscarsan1/espurna, there's a "button new pull request"
Now we can see difference between the base `dev` and yours `oscarsan1/espurna:dev` . If you decide to update some files, Github Desktop will mark them as changed and auto-add them to the commit and allow to push to yours `dev`. PR will track specific branch until it either merged or closed.

BTW, I think I finally understand the problem with magnitudes and tick. What can happen instead is every sensor gets a personal init timeout (see BaseSensor attribute methods, e.g. decimals), then sensorInit can be called based on that and not global 10seconds.

oscarsan1, 2019-07-02:
The code I've upload is working properly now. I've missed the sensor definition. Is uploaded now. 

`//------------------------------------------------------------------------------ 
//  Weather Station sensor with a 433Mhz reveiver. 
//  Manchester code (like Oregon Scientific)
//------------------------------------------------------------------------------

#ifndef OOK_DECODER_SUPPORT
#define OOK_DECODER_SUPPORT                  0       // Do not build with ook decoder support by default
#endif

#ifndef OOK_DECODER_PIN
#define OOK_DECODER_PIN                     3 //3//1 4       // GPIO to monitor
#endif

#ifndef OOK_DECODER_MODE
#define OOK_DECODER_MODE                    INPUT   // INPUT, INPUT_PULLUP
#endif

#ifndef OOK_DECODER_INTERRUPT_MODE
#define OOK_DECODER_INTERRUPT_MODE          CHANGE  // RISING, FALLING, CHANGE
#endif

#ifndef OOK_DECODER_SENSORS_ADDR 
#define OOK_DECODER_SENSORS_ADDR            {0xEA4C3502, 0x1A2D4701, 0x1A2D1F03} // 0xMMMMIICC 
#endif                        // MMMM->model II->Id(change every battery replacement) CC->channel 
 
#ifndef OOK_DECODER_AUTODISCOVER_ADDR
#define OOK_DECODER_AUTODISCOVER_ADDR       0       // if true registrer detected sensor by order of detection
#endif

#ifndef OOK_DECODER_SENSORS_MAX_NUMBER
#define OOK_DECODER_SENSORS_MAX_NUMBER      15       // Max sensors-magnitudes to register
#endif

#ifndef OOK_DECODER_AUTODISCOVER_WAIT_TIME
#define OOK_DECODER_AUTODISCOVER_WAIT_TIME  150     // seconds for waiting to detect sensors.
#endif
`

If OOK_DECODER_AUTODISCOVER_ADDR is true, the sensor wait OOK_DECODER_AUTODISCOVER_WAIT_TIME secons for detecting new sensors. If it is false only the sensors in the OOK_DECODER_SENSORS_ADDR list will be displayed and with the same order that in the list. 

I've implemented it like a sensor, I think that it must be adapt to work with RF_SUPORT toguether and SONOFF RF without hardware modification.

mcspr, 2019-07-04:
Thanks!

Do you want to do any other modifications or can I do some clean-up myself?
*(Split classes into files, slightly different naming for definitions to be more consistent with other sensors, small logic fixes and the detection change mentioned above)*
Do I use yours repo for that, copy into my own or do you want to do it through PR & comments? (https://github.com/xoseperez/espurna/compare/dev...oscarsan1:dev)

I do believe decoder class is MIT licensed code, but we should probably reference it's origin anyways
https://github.com/jeelabs/jeelib/blob/026bb6ea843abafff1619ae272a4ea754e0b016e/examples/RF12/ookRelay2/decoders.h
http://jeelabs.net/projects/cafe/wiki/Decoding_the_Oregon_Scientific_V2_protocol

oscarsan1, 2019-07-04:
No more modifications for my side. Only for a future integration with sonoff RF and RF_SUPORT would be interesting that this sensors works toguether with this devices. At the moment this sensor is not compatible. I don't have Sonoff RF so I don't kwow how it works. 
Please clean-up yourself as you consider.
I'm a beginner with github, I think that is more easy that you use my repo as I've never done a PR, but I can try it.
Yes we shoud reference it's origin.