xoseperez/espurna #1800: how to work with uart<=>mqtt?

connected201, 2019-06-28:
i git espurna firmware, open in vscode, comment #define UART_MQTT_SUPPORT, push compile, push upload, so i don't know if i do correctly to apply #define UART_MQTT_SUPPORT to wemos d1 mini configuration.
what i need to do to read and wride data using mqtt? i connected to my wemos d1 mini rdm6300 rfid card reader, directly to wemos, if i connect rdm6300 directly to uart on computer,  all work fine!
connection diagram:
[image]
computer code:
[image]
what i missing? in debug console nothing! this device works on with 9600 baud rate, but i don't know where i need change or if i need to change this baud rate in firmware

mcspr, 2019-06-28:
Please reference the general.h, all UART_MQTT settings are there

connected201, 2019-06-28:
i configured in this mode:
[image]
on Wemo d1 mini RX pin is GPIO3
[image]
all correct i configured? because after firmware information was not received by UART,

mcspr, 2019-06-28:
Ah! Sorry, I think you may need to try without USB or 5V power here.  5V powers up onboard serial converter, so RX becomes unusable. Just use the 3V3 pin with some appropriate power supply.

connected201, 2019-06-28:
as i understand, i need to give 3.3v directly to esp8266, i don't need to use integrated usb power?

mcspr, 2019-06-28:
Yep, just use the bottom left pin ref. picture above and Ground
I tried this briefly with a knock-off D1, but I think it also applies for the real one. If powered via 5V pin RX does not work. Same with USB power

connected201, 2019-06-28:
so, i connected wemos to 3.3v, rfid reader to 5v, ground is common for esp and rfid reader, but debug console is empty, 
wiring photo:

[image]

connected201, 2019-07-02:
i connected directly to usb ttl  same results, i cannot receive data over UART, maybe need to additional firmware configuration and after upload to esp?
[image]

mcspr, 2019-07-02:
- UART_MQTT_TERMINATION is looking for new line symbol, when there's none
http://zjembedded.blogspot.com/2012/06/rdm6300-rfid-with-pic18-dev-board.html
You need to disable that when doing tests with a real device
- I had a bunch of these laying around
https://www.aliexpress.com/item/32827321415.htm
And there is data coming through to the ESP. Maybe it is power issue? I do connect ESP to a separate power supply, not from the usbttl 3v3 pin (which does not work, just like yours)
```
[008395] [MQTT] Received ESPURNA-35A259/uartout/set => check
[008399] [UART_MQTT] Send data over UART: check
[010009] [MQTT] Sending ESPURNA-35A259/status => 1 (PID 7)
[010013] [MQTT] Publish ACK for PID 7
[013943] [UART_MQTT] Send data over MQTT: check
[013945] [MQTT] Sending ESPURNA-35A259/uartin => check (PID 8)
[013948] [MQTT] Publish ACK for PID 8
[020013] [MQTT] Sending ESPURNA-35A259/status => 1 (PID 9)
[020017] [MQTT] Publish ACK for PID 9
```

connected201, 2019-07-02:
so how to explain this?
why i cannot send only plain text data to send it to MQTT?
[image]
[image]

> https://www.aliexpress.com/item/32827321415.htm

if i open, i received blank page

connected201, 2019-07-02:
all done guys! i found good serial terminal, also i found issue about my device 
https://github.com/xoseperez/espurna/issues/878
[image]

closed 2019-07-02 by connected201