xoseperez/espurna #1802: Occasional reboots when toggling BWSHP-6

m-kozlowski, 2019-06-29:
**Question**
I'm experiencing random reboots when switching on or off Blitzwolf shp-6 plugs.
It doesn't matter if I'm switching via MQTT, Domoticz, Web interface, or a hardware button.
No crash info is being recorded. No suspicious stuff is being logged in websocket debug terminal either.

**Device information**
```
[046550] [MAIN] ESPURNA 1.13.6-dev (75c6a75b)
[046554] [MAIN] xose.perez@gmail.com
[046555] [MAIN] http://tinkerman.cat

[046556] [MAIN] CPU chip ID: 0x3435A0
[046557] [MAIN] CPU frequency: 80 MHz
[046560] [MAIN] SDK version: 1.5.3(aec24ac9)
[046562] [MAIN] Core version: 2.3.0
[046564] [MAIN] Core revision: 159542381
[046569] [MAIN] Build time: 1560205223
[046574] 
[046576] [MAIN] Flash chip ID: 0x1640EF
[046582] [MAIN] Flash speed: 40000000 Hz
[046583] [MAIN] Flash mode: DOUT
[046584] 
[046586] [MAIN] Flash size (CHIP)   :  4194304 bytes / 1024 sectors (   0 to 1023)
[046595] [MAIN] Flash size (SDK)    :  4194304 bytes / 1024 sectors (   0 to 1023)
[046600] [MAIN] Reserved            :     4096 bytes /    1 sectors (   0 to    0)
[046607] [MAIN] Firmware size       :   501184 bytes /  123 sectors (   1 to  123)
[046615] [MAIN] Max OTA size        :  2637824 bytes /  644 sectors ( 124 to  767)
[046622] [MAIN] SPIFFS size         :  1015808 bytes /  248 sectors ( 768 to 1015)
[046629] [MAIN] EEPROM size         :    16384 bytes /    4 sectors (1016 to 1019)
[046636] [MAIN] Reserved            :    16384 bytes /    4 sectors (1020 to 1023)
[046643] 
[046645] [MAIN] EEPROM sectors: 1019, 1018, 1017, 1016
[046646] [MAIN] EEPROM current: 1017
[046652] 
[046653] [MAIN] EEPROM:  4096 bytes initially |  1374 bytes used (33%) |  2722 bytes free (66%)
[046663] [MAIN] Heap  : 34984 bytes initially | 19488 bytes used (55%) | 15496 bytes free (44%)
[046670] [MAIN] Stack :  4096 bytes initially |  1348 bytes used (32%) |  2748 bytes free (67%)
[046677] 
[046679] [MAIN] Boot version: 31
[046681] [MAIN] Boot mode: 1
[046682] [MAIN] Last reset reason: External System
[046687] [MAIN] Last reset info: Fatal exception:0 flag:6 (EXT_SYS_RST) epc1:0x00000000 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000
[046702] 
[046704] [MAIN] Board: BLITZWOLF_BWSHPX
[046705] [MAIN] Support: ALEXA API BROKER BUTTON DEBUG_SERIAL DEBUG_TELNET DEBUG_WEB DOMOTICZ LED MDNS_SERVER MQTT NOFUSS NTP SCHEDULER SENSOR TELNET TERMINAL WEB 
[046720] [MAIN] Sensors: HLW8012 
[046722] [MAIN] WebUI image: SENSOR
[046724] 
[046725] [MAIN] Firmware MD5: d74fcdd87007bdae5bbd19496f96d626
[046733] [MAIN] Power: 3205 mV
[046737] [MAIN] Power saving delay value: 10 ms
```

Local build is based on commit 0998110d869264a893f8a0df6eff833620c7a4cb
Differences from upstream:
- wifi/mqtt/domoticz/ntp configuration defaults
- NOFUSS_SUPPORT 1
- HOMEASSISTANT_SUPPORT, THINGSPEAK_SUPPORT 0

You may also notice that the device is slightly hacked (increased flash size) it it also occured on unmodified hardware.

**Additional context**
Things i've tried so far:
- Disabling mqtt/domoticz integration, ntp, nofuss 
- using core 2_5_2 / sdk 2.2.1 
- changing between PIO_FRAMEWORK_ARDUINO_LWIP_HIGHER_BANDWIDTH and PIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY

And some hardware stuff:
1. Since i had 8 plugs stacked quite tight and suspected interference, I've isolated one and plugged external antenna. No joy.
2. Following EXT_SYS_RST track i thought that maybe reset is floating, but 10K pullup is definitely there. Soldering joints looks ok.

m-kozlowski, 2019-06-29:
It looks that when connected via serial everythings works fine. Bad power supply, then? 
Ripped the device apart again and found that it's powered with MP150 rated at 200mA, so it might be bit too low :(

mcspr, 2019-06-29:
Just a random though - can you try disabling lngclick action?
https://github.com/xoseperez/espurna/blob/e445cf8c7003827680d63f33b391bc95faec5de3/code/espurna/config/defaults.h#L137

afaik EXT_SYS_RST  is a normal reset result, nothing suspicious
does disabling alexa do anything?

m-kozlowski, 2019-06-30:
Disabling alexa (in runtime config) doesn't help.
No success with `BUTTON1_LNGCLICK    BUTTON_MODE_NONE` either.

mcspr, 2019-06-30:
Hm. I retract. RST pin is the reason for the reset, i.e. EXTERNAL part. One of the issues from the esp8266/Arduino:
https://github.com/esp8266/Arduino/issues/3266
However, we did already see something similar with T1 when trying to debug it while relay board is connected (and why I thought of button), resulting in insufficient power:
https://github.com/xoseperez/espurna/issues/1284

m-kozlowski, 2019-06-30:
It was electromagnetic interference apparently.
Good hint was that during recent tests toggling one plug caused neighboring one to reboot few times.
So i started googling for interference related reboots of esp8266 and found this:
https://www.esp8266.com/viewtopic.php?p=23441#p23441
Leaving RST pin floating is bit counterintuitive for me, but so far it seems to work (not a single reboot over few hundred switches).

closed 2019-06-30 by m-kozlowski