xoseperez/espurna #1813: Received code -> RF Action 0xA4

Thoky81, 2019-07-08:
Hello,

is there a way to get mqtt published code before they are converted? Below is my my debug status with RF Received code -> which is then converted by Action 0xA4 -> to longer code. I have a door bell which sends these "rolling" codes but in fact last 8 number are always the same. Can pls someone suggest how to do it ? Thanks

[589669] [RF] Received code: **00025067**
[589672] [RF] Action 0xA4
[589673] [RF] Received message 'C00201CF12**00025067**'
[589675] [RF] Trying to match code 025067
[589786] [MQTT] Sending ESPURNA-01/data => {"rfin":"C00201CF1200025067","time":"2019-07-08

Niek, 2019-07-11:
You can add a (virtual) relay and set your RF match code 025067 (the first numbers are used for protocol-encoding). Then you can report this relay's action to a custom MQTT group.

Thoky81, 2019-07-12:
Thanks for reply. Can I ask you for detailed instructions how to do it? I have search the wiki but could not find the relevant info.

Niek, 2019-07-15:
Go into the web ui (I assume you use the RF Bridge image), under the RF section you can change which relay to match the code with.

mcspr, 2019-09-13:
Just for future reference. Home Assistant has a pretty extensive templating system.
https://www.home-assistant.io/components/template/
https://jinja.palletsprojects.com/en/master/templates/

We can modify incoming data to cut everything but the last 6 chars:
```yaml
- platform: mqtt
  name: "ESPURNA 123456 RFIN"
  state_topic: "ESPURNA-123456/rfin"
  availability_topic: "ESPURNA-123456/status"
  payload_available: "1"
  payload_not_available: "0"
  force_update: true
  expire_after: 1
  value_template: >
       {% if not value %}
           "No data"
       {% endif %}
       {{ value[-6:] }}
```
Will create a `sensor.espurna_123456_rfin`, which state we can later use for some automation

closed 2019-11-12 by mcspr