xoseperez/espurna #1815: Using Wemos D1 mini with multiple GPIO inputs

davebuk, 2019-07-09:
I have a D1 mini and plan to use it with MQTT and openHab as the main hardware interface for a house alarm system. The idea is to have the currently installed hard wired PIRs feed their N/C outputs into the GPIO pins and report the 0/1 status for triggering the alarm once armed.

I have used cushom.h file with `DIGITAL_PIN` definitions but only got GPIO 14 to show in the status page. Another post says I should use `EVENTS_PIN` instead as this will give a real time MQTT response. Either way, my configuration does not give any more than one reading on the status page, just GPIO 14.

```
#elif defined(WEMOS_D1_MINI_RELAYSHIELD)

    // Info
    #define MANUFACTURER        "WEMOS"
    #define DEVICE              "D1_MINI_RELAYSHIELD"

    #define ALEXA_SUPPORT           0
    #define DOMOTICZ_SUPPORT        0
    #define HOMEASSISTANT_SUPPORT   0
    #define SCHEDULER_SUPPORT       0
    #define THINGSPEAK_SUPPORT      0
    #define DEBUG_SERIAL_SUPPORT    0

    // Buttons
    // No buttons on the D1 MINI alone, but defining it without adding a button doen't create problems
    #define BUTTON1_PIN          0   // Connect a pushbutton between D3 and GND,
                                     // it's the same as using a Wemos one button shield
    #define BUTTON1_MODE         BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
    #define BUTTON1_RELAY        1
     
    // Relays
    #define RELAY1_PIN           5  // D1
    #define RELAY1_TYPE          RELAY_TYPE_INVERSE

    // LED
    #define LED1_PIN             2  // D4
    #define LED1_PIN_INVERSE     1

    //Digital Event support
    
    #define EVENTS_SUPPORT       1
    #define EVENTS_TRIGGER      1
    #define EVENTS_PIN           3  // RX
    #define EVENTS_PIN_MODE      INPUT_PULLUP
    #define EVENTS_INTERRUPT_MODE	CHANGE
    #define EVENTS_PIN           4  // D2
    #define EVENTS_PIN_MODE      INPUT_PULLUP
    #define EVENTS_INTERRUPT_MODE	CHANGE
    #define EVENTS_PIN           12  // D6
    #define EVENTS_PIN_MODE      INPUT_PULLUP
    #define EVENTS_INTERRUPT_MODE	CHANGE
    #define EVENTS_PIN           13  // D7
    #define EVENTS_PIN_MODE      INPUT_PULLUP
    #define EVENTS_INTERRUPT_MODE	CHANGE    
    #define EVENTS_PIN           14  // D5
    #define EVENTS_PIN_MODE      INPUT_PULLUP
    #define EVENTS_INTERRUPT_MODE	CHANGE
```

ColinShorts, 2019-07-09:
Each iteration of the events pin definition overwrites the last, but that would be the expected behaviour of the code.

I.e:

A = 1
A = 2
A = 3
print A

> 3

davebuk, 2019-07-09:
So I can't use digital or events pin definitions for multiple GPIOs?

I guess I could try multiple toggle switch definitions  and make sure web GUI or mqtt doesn't change the switch state, just the 0/1 on the GPIO pins?

ColinShorts, 2019-07-09:
I don't think you'd have too much of a latency issue using normal button/switch defines over mqtt, the only time I see any lag is from the sonoff pir/door switches connecting via the RF bridge - I usually make it to the bottom of the stairs before the light comes on. The T1's that I use in mqtt groups (additional units as slaves) all switch in around half a second or so.

BUTTON_SWITCH will send an event on opening and closing, BUTTON_PUSHBUTTON will only send an event when it releases (so it might not work as desired). Pushbutton might also give you problems on certain gpio's when using n/c (I use a sonoff dual that's buried in the wall, and so occasionally I forget to switch off the light switches before restoring power to the light circuit. I've got connected to it's gpio... And it goes into programming mode every time 😆

Alternatively, there are a few Arduino based setups that might be more reliable, particularly if you can get it hard wired over Ethernet to your LAN. It should be feasible to use a really cheap spi based module, albeit at a fairly slow speed, which should still be plenty for mqtt. I've used them on rpi's in the past, and they worked first time as soon as spi was enabled ofc.

I wish I had a hard wired alarm system.

davebuk, 2019-07-10:
Thanks for the info. I'd planned to use BUTTON_SWITCH so I didn't have any issues with the long click etc. It would be nice to have the actual open/close state via mqtt but know SWITCH can't do that. For my application state toggle should be fine. I'll see how I get on and look at Arduino in the future if I want more capability.

davebuk, 2019-07-10:
@mcspr @xoseperez 
Can multiple digital sensors be added to the sensor.h file? Possibly their status could be used WRT #1747 with an mqtt 'sync' option to read the HIGH/LOW state of a GPIO associated with a button/switch and update everything to match the switch position.

davebuk, 2019-07-10:
I have tried with BUTTON_PUSHBUTTON for now, but the following config only gives one switch (Switch#0 at GPIO5). I didn't want to bind the other switches to the relay as the relay is going to activate the alarm sounder based on openHab rules. Do I have to bind a dummy relay?

```
elif defined(WEMOS_D1_MINI_RELAYSHIELD)

    // Info
    #define MANUFACTURER        "WEMOS"
    #define DEVICE              "D1_MINI_RELAYSHIELD"
    #define BUTTON_EVENT_NONE           0
    #define BUTTON_EVENT_PRESSED        1
    #define BUTTON_EVENT_RELEASED       2

    // Buttons
    // No buttons on the D1 MINI alone, but defining it without adding a button doen't create problems
    #define BUTTON1_PIN          0   // Connect a pushbutton between D3 and GND,
                                     // it's the same as using a Wemos one button shield
    #define BUTTON1_MODE         BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH
    #define BUTTON1_RELAY        1
    
    #define DEBUG_SERIAL_SUPPORT 0
    
    // Relays
    #define RELAY1_PIN           5  // D1
    #define RELAY1_TYPE          RELAY_TYPE_INVERSE

    // LED
    #define LED1_PIN             2  // D4
    #define LED1_PIN_INVERSE     1

    // GPIO Buttons
    
    #define BUTTON2_PIN         3  //RX
    #define BUTTON2_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH | BUTTON_SET_PULLUP
    #define BUTTON2_PRESS       BUTTON_MODE_OFF
    #define BUTTON2_CLICK       BUTTON_MODE_ON

    #define BUTTON3_PIN         4  //D2
    #define BUTTON3_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH | BUTTON_SET_PULLUP
    #define BUTTON3_PRESS       BUTTON_MODE_OFF
    #define BUTTON3_CLICK       BUTTON_MODE_ON

    #define BUTTON4_PIN         12  //D6
    #define BUTTON4_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH | BUTTON_SET_PULLUP
    #define BUTTON4_PRESS       BUTTON_MODE_OFF
    #define BUTTON4_CLICK       BUTTON_MODE_ON

    #define BUTTON5_PIN         13  //D7
    #define BUTTON5_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH | BUTTON_SET_PULLUP 
    #define BUTTON5_PRESS       BUTTON_MODE_OFF
    #define BUTTON5_CLICK       BUTTON_MODE_ON

    #define BUTTON6_PIN         14  //D5
    #define BUTTON6_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH | BUTTON_SET_PULLUP
    #define BUTTON6_PRESS       BUTTON_MODE_OFF
    #define BUTTON6_CLICK       BUTTON_MODE_ON
```

mcspr, 2019-07-10:
> I have tried with BUTTON_PUSHBUTTON for now, but the following config only gives one switch (Switch#0 at GPIO5).

Based on config, 5 buttons are linked to the single switch and expected behaviour is to release the switch when no buttons are pressed?
As I understand it, you do run into the same issue as was discussed in the #1747 - buttons handle events, not status. Since you disabled handlers for events 4 and 5 (long clicks), you don't receive them (see BUTTON_MQTT_SEND_ALL_EVENTS) and also don't trigger the relay switch, because button stays pressed for much longer than the short press timeout.

Other issue has a PR linked that proposes sending pressed status instead of event, but not sure how that would work better - as a configuration for type of the thing that gets reported (event number or boolean pressed status), or additional action. Configuration naming too...
DigitalSensor is another option, but would require sensor.ino modification to have multiple sensors created instead of just one
EventsSensor is more of event counter than something for triggers. It can sort-of do that, but that imo is an overkill for buttons

davebuk, 2019-07-11:
> Based on config, 5 buttons are linked to the single switch and expected behaviour is to release the switch when no buttons are pressed?

Not quite. I want 5 GPIOs to monitor the state of 5 PIRs (which are NC contacts when no movement detected) and report OPEN/CLOSED using MQTT (and ideally show their status it the web GUI). [OpenHAB](https://www.openhab.org/) would then use the state change of a virtual switch when the virtual alarm system is set as 'armed'. If the system is triggered, OpenHAB then switches Switch#0 on, activating the low level trigger relay and applying power to the sounder. After a time period or deactivation of Switch#0, the sounder goes off.
  
> As I understand it, you do run into the same issue as was discussed in the #1747 - buttons handle events, not status. Since you disabled handlers for events 4 and 5 (long clicks), you don't receive [...]

Playing around I saw that and made adjustments. All events now report their status to MQTT. Long and Longlong give 4 and 5 and when open gives 1. OpenHAB can then use, IF switch is NOT 1, no movement (OPEN or OFF), IF 1, movement (CLOSED or ON). I did have the odd event '6' appear. Not sure what state that represents? I also tried `BUTTON_SWITCH` but of course that only reports event 2, regardless of state.

> Other issue has a PR linked that proposes sending pressed status instead of event, but not sure how that would work better - as a configuration for type of the thing that gets reported (event number [...]

Just looking at the Wiki, I assumed DigitalSensor would allow multiple sensors and report 0/1 for each one, but it only allows the one GPIO. I'll look at sensor.ino, but my coding knowledge is pretty limited. :-)

davebuk, 2019-07-11:
Next step is to add RFID arm/disarm #1800!

mcspr, 2019-07-11:
> I did have the odd event '6' appear. Not sure what state that represents?
Sorry to reference code again, but that was triple click:
https://github.com/xoseperez/espurna/blob/0998110d869264a893f8a0df6eff833620c7a4cb/code/espurna/config/types.h#L32-L39
(which should probably be printed in the debug log, at the very least. maybe optionally, send this translation instead of evt number to mqtt)

For multiple sensors, this needs to be duplicated per sensor (including curly braces):
https://github.com/xoseperez/espurna/blob/1d133bee16b4c953d43959cfceede12eec389e0c/code/espurna/sensor.ino#L575-L582
https://github.com/xoseperez/espurna/blob/1d133bee16b4c953d43959cfceede12eec389e0c/code/espurna/sensor.ino#L469-L476
Which, again, *could* be configured via settings at runtime whenever DIGITAL_SUPPORT is on. Each header config option could get an index (a-la BMX280 config in the same method)

davebuk, 2019-07-11:
@ColinShorts @mcspr 
Thanks for the info. The Wiki is missing a few things, like event 6 stated above, so your knowledge is invaluable!

I'll try and find some time at the weekend to modify sensor.ino and build using digital sensors again. It seems the best solution for my project.

davebuk, 2019-07-12:
> For multiple sensors, this needs to be duplicated per sensor (including curly braces):
> https://github.com/xoseperez/espurna/blob/1d133bee16b4c953d43959cfceede12eec389e0c/code/espurna/sensor.ino#L576-L582

And multiple entries in sensor.h file?

```
#ifndef DIGITAL_PIN2
#define DIGITAL_PIN2                     3
#endif

#ifndef DIGITAL_PIN_MODE2
#define DIGITAL_PIN_MODE2                INPUT_PULLUP
#endif

#ifndef DIGITAL_DEFAULT_STATE2
#define DIGITAL_DEFAULT_STATE2 
```
etc...

davebuk, 2019-07-12:
Getting there!
[image]

Last piece of the puzzle. MQTT will only show the digital states with respect to the refresh rate set in the sensor tab of the web GUI. If a sensor goes 1 to 0 and back to 1, MQTT dosen't post that state unless the state is at 0 during its refresh cycle. Is there another setting is sensor.ino to report to MQTT on state change like the `EVENT_PIN` setup?

davebuk, 2019-07-12:
I wonder if `DIGITAL_PIN` is not supported. WRT #1186, this does seem to be the case. I have added a set of extra `EVENT_PIN` entries to my sensor.ino and sensor.h files. I now have five Count# entries on the status page. This works well as MQTT reports the change of state as 1 and 0 and will periodically send the current MQTT states as well.

[image]

The only strange issue is that Count#3 says its @GPIO3, but is set to 13 in custom.h?

```
    #define EVENTS_PIN4           13  // D7
    #define EVENTS_PIN_MODE4      INPUT_PULLUP
    #define EVENTS_INTERRUPT_MODE4	CHANGE
```

mcspr, 2019-07-13:
DigitalSensor *is* reading every sensor read interval, unlike EventsSensor. Note that with recent change I tried to speed up events processing by removing immediate callback call and caching it for a bit. Hope that works ok (including debounce time, my main worry are missing values)
Digital one can have the same trigger mechanics though, imitating the buttons pin reader

ColinShorts, 2019-07-13:
@davebuk, how long do the pir's hold the "switch" open? I wouldn't imagine that the state would change that rapidly. @mcspr's changes might be good enough for your use case already.

davebuk, 2019-07-13:
Probably a maximum of 1 second. Doing a 'walk test', three out of the five show MQTT state changes on every trigger (1) and then 0 when not triggered. I think I need external pull up resistors for the other two PIRs. I'll look to try that this weekend.

davebuk, 2019-07-13:
All five PIR's are now working with pull up resistors. A few lines from debug are below, but all Sensors#0-4 give the correct outputs when triggered. Thanks again for all your help.

```
[708744] [SENSOR] Sensor #4 callback, type 14, payload: '1.00'
[708747] [MQTT] Sending D1mini001/event/4 => 1 (PID 1)
[713047] [SENSOR] Sensor #4 callback, type 14, payload: '0.00'
[713050] [MQTT] Sending D1mini001/event/4 => 0 (PID 1)
[719297] [SENSOR] Sensor #4 callback, type 14, payload: '1.00'
[719299] [MQTT] Sending D1mini001/event/4 => 1 (PID 1)
[720121] [MQTT] Sending D1mini001/count/0 => 0 (PID 1)
[720123] [MQTT] Sending D1mini001/event/0 => 0 (PID 1)
[720126] [MQTT] Sending D1mini001/count/1 => 0 (PID 1)
[720127] [MQTT] Sending D1mini001/event/1 => 0 (PID 1)
[720129] [MQTT] Sending D1mini001/count/2 => 0 (PID 1)
[720132] [MQTT] Sending D1mini001/event/2 => 0 (PID 1)
[720134] [MQTT] Sending D1mini001/count/3 => 0 (PID 1)
[720136] [MQTT] Sending D1mini001/event/3 => 0 (PID 1)
[720140] [MQTT] Sending D1mini001/count/4 => 3 (PID 1)
[720142] [MQTT] Sending D1mini001/event/4 => 0 (PID 1)
[739058] [SENSOR] Sensor #4 callback, type 14, payload: '0.00'
[739061] [MQTT] Sending D1mini001/event/4 => 0 (PID 1)
[748657] [SENSOR] Sensor #0 callback, type 14, payload: '1.00'
[748660] [MQTT] Sending D1mini001/event/0 => 1 (PID 1)
[750123] [MQTT] Sending D1mini001/count/0 => 1 (PID 1)
[750126] [MQTT] Sending D1mini001/event/0 => 0 (PID 1)
```

closed 2019-07-15 by davebuk

ruimarinho, 2019-08-09:
@davebuk I'm trying to get the [Wemos PIR shield](https://wiki.wemos.cc/products:d1_mini_shields:pir_shield) working on a Wemos D1 Mini with the following config:

```
ESPURNA_FLAGS='-DEVENTS_SUPPORT=1 -DEVENTS1_PIN=0 -DBUTTON_MQTT_SEND_ALL_EVENTS=1'
```

The PIR is HIGH on GPI0 (D3) when motion is detected. However, the interrupt is not getting triggered with motion:

```
[001005] [SENSOR] INTERRUPT @ GPIO0 - count: 0
```

Any ideas?

davebuk, 2019-08-09:
I don't know of a specific reason, the only thing I have set differently is to enable `INPUT_PULLUP` to use the pull up resistors.

davebuk, 2019-08-09:
I don't build using the flags, I use platformio with a custom.h file and build using `-DUSE_CUSTOM_H`. I then set all the #define options in the custom.h file.

ruimarinho, 2019-08-09:
What is the expected message I should be seeing? A button sensor log?

davebuk, 2019-08-09:
AFAIK the button events flag you have set is for any configured buttons, not the event sensor you have added. 

Looking at my output i the debug screen (https://github.com/xoseperez/espurna/issues/1815#issuecomment-511153476)
its the 'sensor call back' that that triggers an instantaneous output and MQTT output. The sensor output you have referred to, I believe, its output depends on the Read/Report interval set in the 'SESNORS' tab in the webUI.

You could test the wemos GPIO pin by leaving the PIR disconnected for now and using a piece of wire to pulse your GPIO to the GND pin or the 3V3 pin and see what happens.  Do you have a status page similar to mine: https://github.com/xoseperez/espurna/issues/1815#issuecomment-511028088 When my PIRs go from LOW to HIGH (NC or open circuit) the count goes up on the status page for that configured GPIO.

davebuk, 2019-08-09:
The other thing, https://github.com/xoseperez/espurna/wiki/Sensors#counter-sensor status the default trigger is 'rising'. I changed mine in custom.h to use 'change' (not 'both', think wiki is wrong) so it reports all triggers.

ruimarinho, 2019-08-10:
I've removed the PIR shield and pulsing GPIO0 to GND gives me some interesting results. Is it normal to crash the Wemos if I keep the "pulse" long enough (e.g. 2 seconds)?

Sometimes only a `[BUTTON]` event is dispatched, other times I see a `[SENSOR]` callback as well.

```
[027140] [SENSOR] INTERRUPT @ GPIO0 - count: 0
[027140] [SENSOR] INTERRUPT @ GPIO0 - event: 0
[028145] [SENSOR] INTERRUPT @ GPIO0 - count: 0
[028145] [SENSOR] INTERRUPT @ GPIO0 - event: 0
[028875] [BUTTON] Button #0 event 1
[029149] [SENSOR] INTERRUPT @ GPIO0 - count: 0
[029149] [SENSOR] INTERRUPT @ GPIO0 - event: 0
[030153] [SENSOR] INTERRUPT @ GPIO0 - count: 0
[030154] [SENSOR] INTERRUPT @ GPIO0 - event: 0
[031158] [SENSOR] INTERRUPT @ GPIO0 - count: 0
[031159] [SENSOR] INTERRUPT @ GPIO0 - event: 0
[032162] [SENSOR] INTERRUPT @ GPIO0 - count: 0
[032163] [SENSOR] INTERRUPT @ GPIO0 - event: 0
[032233] [BUTTON] Button #0 event 4
[032234] [SENSOR] Sensor #0 callback, type 14, payload: '1.00' -- pulse too long here

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
[000371]
```

With the PIR shield installed, every time there is motion, the Wemos crashes (cc @mcspr):

```
[001004] [SENSOR] INTERRUPT @ GPIO0 - count: 0
[001004] [SENSOR] INTERRUPT @ GPIO0 - event: 0
[001907] [BUTTON] Button #0 event 1
[002009] [SENSOR] INTERRUPT @ GPIO0 - count: 0
[002009] [SENSOR] INTERRUPT @ GPIO0 - event: 0
[003013] [SENSOR] INTERRUPT @ GPIO0 - count: 0
[003013] [SENSOR] INTERRUPT @ GPIO0 - event: 0
[004017] [SENSOR] INTERRUPT @ GPIO0 - count: 0
[004017] [SENSOR] INTERRUPT @ GPIO0 - event: 0
[005021] [SENSOR] INTERRUPT @ GPIO0 - count: 0
[005022] [SENSOR] INTERRUPT @ GPIO0 - event: 0
[005882] [BUTTON] Button #0 event 4
[005882] [SENSOR] Sensor #0 callback, type 14, payload: '1.00'

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
[000371] [MAIN] System UNSTABLE
[000373]

---8<-------
```

davebuk, 2019-08-10:
See: https://github.com/xoseperez/espurna/blob/274f71d7d9047bb2bdf6b5c2cca7a6ff9d63142b/code/espurna/config/hardware.h#L98
All default builds of wemos devices set GPIO/D3 as a button1. Long press will trigger [button events.](https://github.com/xoseperez/espurna/wiki/Buttons-and-switches#buttons-and-switches)

I would change the GPIO pin used on your PIR board to another one and configure wemos accordingly.

bluexp1, 2021-07-19:
> Getting there!
> [image]
>
> Last piece of the puzzle. MQTT will only show the digital states with respect to the refresh rate set in the sensor tab of the web GUI. If a sensor goes 1 to 0 and back to 1, [...]

I have problem activating digital input/event in 1.14.1 version. I have edited arduino.h and sensor.h but digital inputs are not shown in web interface. Is there any steps I missed?
arduino.h
`//--------------------------------------------------------------------------------
// These settings are normally provided by PlatformIO
// Uncomment the appropiate line(s) to build from the Arduino IDE
//--------------------------------------------------------------------------------

//--------------------------------------------------------------------------------
// Hardware
//--------------------------------------------------------------------------------

//#define AITHINKER_AI_LIGHT
//#define ALLNET_4DUINO_IOT_WLAN_RELAIS
//#define ALLTERCO_SHELLY1
//#define ALLTERCO_SHELLY1PM
//#define ALLTERCO_SHELLY2
//#define ALLTERCO_SHELLY25
//#define ARILUX_AL_LC01
//#define ARILUX_AL_LC02
//#define ARILUX_AL_LC02_V14
//#define ARILUX_AL_LC06
//#define ARILUX_AL_LC11
//#define ARILUX_E27
//#define ARNIEX_SWIFITCH
//#define AUTHOMETION_LYT8266
//#define BESTEK_MRJ1011
//#define BH_ONOFRE
//#define BLITZWOLF_BWSHPX
//#define BLITZWOLF_BWSHPX_V23
//#define BLITZWOLF_BWSHP5
//#define DIGOO_NX_SP202
//#define EHOMEDIY_WT02
//#define EHOMEDIY_WT03
//#define ELECTRODRAGON_WIFI_IOT
//#define ESPURNA_BASE
//#define ESPURNA_CORE
//#define ETEKCITY_ESW01_USA
//#define EUROMATE_WIFI_STECKER_SCHUKO
//#define EUROMATE_WIFI_STECKER_SCHUKO_V2
//#define EXS_WIFI_RELAY_V31
//#define EXS_WIFI_RELAY_V50
//#define FORNORM_ZLD_34EU
//#define FOXEL_LIGHTFOX_DUAL
//#define FS_UAP1
//#define GBLIFE_RGBW_SOCKET
//#define GENERIC_8CH
//#define GENERIC_AG_L4
//#define GENERIC_ECH1560
//#define GENERIC_ESP01S_DHT11_V10
//#define GENERIC_ESP01S_DS18B20_V10
//#define GENERIC_ESP01S_RELAY_V40
//#define GENERIC_ESP01S_RGBLED_V10
//#define GENERIC_GU10
//#define GENERIC_E14
//#define GENERIC_V9261F
//#define GIZWITS_WITTY_CLOUD
//#define GOSUND_WS1
//#define GREEN_ESP8266RELAY
//#define HAMA_WIFI_STECKDOSE_00176533
//#define HELTEC_TOUCHRELAY
//#define HEYGO_HY02
//#define HOMECUBE_16A
//#define HUACANXING_H801
//#define HUACANXING_H802
//#define HYKKER_SMART_HOME_POWER_PLUG
[... 1600 lines cut ...]
    #include "../sensors/T6613Sensor.h"
#endif

#if TMP3X_SUPPORT
    #include "../sensors/TMP3XSensor.h"
#endif

#if V9261F_SUPPORT
    #include "../sensors/V9261FSensor.h"
#endif

#if VEML6075_SUPPORT
    #include "../sensors/VEML6075Sensor.h"
#endif

#if VL53L1X_SUPPORT
    #include "../sensors/VL53L1XSensor.h"
#endif

#if ADE7953_SUPPORT
    #include "../sensors/ADE7953Sensor.h"
#endif

#endif // SENSOR_SUPPORT
`

davebuk, 2021-07-19:
From those config files above, I would say it looks like you have enabled the following functions.
```
#define DALLAS_SUPPORT         1
#define DHT_SUPPORT            1
#define DIGITAL_SUPPORT        1
#define EVENTS_SUPPORT         1
```
I think the problem is the `sensors.h` has `#define DIGITAL1_PIN    14` then later `#define EVENTS1_PIN    14` which I believe overwrites the `DIGITAL` GPIO setting.

Try either editing `sensor.h` for the GPIOs you are using or look in the wiki about using `custom.h` file to make your own definitions.

davebuk, 2021-07-19:
I would tend to use just the `EVENT` function and not the `DIGITAL` one, so you get an immediate response rather than having to wait for the sensor update routine to run.

mcspr, 2021-07-20:
> I would tend to use just the `EVENT` function and not the `DIGITAL` one, so you get an immediate response rather than having to wait for the sensor update routine to run.

Note of the 1.15.0 function of `${magnitude}MinDelta${num}` and `${magnitude}MaxDelta${num}` settings. e.g. setting `digitalMinDelta0` to `0` will cause MQTT to report every reading interval, when the value changes and without waiting for the specified report time. Max delta works similarly, but the other way around - it will cause the report only when value actually changes for the specified amount. In case of digital, it is always `1`

`EVENT` was also changed since 1.14.1, since I removed the event trigger function in favour of the min & max delta triggers above. Right now it is expected to be used to count things e.g. water, wind, optical sensors that trigger the pin many times a second

bluexp1, 2021-07-22:
> From those config files above, I would say it looks like you have enabled the following functions.
>
> ```
> #define DALLAS_SUPPORT         1
> #define DHT_SUPPORT            1
> #define [...]

I tried with "#define EVENTS_SUPPORT      0" and even changed  EVENTS1_PIN but still Digital pin state is not shown on web interface!

mcspr, 2021-07-22:
@bluexp1 if you go to the DEBUG panel, and type `magnitudes` then click send / press enter, do you see digital sensor(s) in the output?

bluexp1, 2021-07-23:
@mcspr I received nothing in debug panel, but in serial terminal I got a "+OK"

mcspr, 2021-07-23:
@bluexp1 should be `[MAIN] Sensors: DIGITAL` at the very least, check the config flags in the .h

bluexp1, 2021-07-24:
@mcspr it is really strange, cause I already have "#define DIGITAL_SUPPORT        1" in both arduino.h and hardware.h!!!

mcspr, 2021-07-24:
@bluexp1 my only other guess is to use `#define DIGITAL1_PIN 4`, `#define DIGITAL2_PIN 5`, etc. in the arduino.h b/c you never end up reaching statements wrapped with `#ifndef DIGITAL1_PIN` b/c they *are* defined by later headers

bluexp1, 2021-07-28:
@mcspr Thanks, I added it to hardware.h and  it worked!