xoseperez/espurna #1819: Devices don't record last state after reboot or recovering from power lost

rmcbc, 2019-07-13:
**Bug description**
I facing a problem with dev and don't have the same problem with 1.13.2
In SWITCHES, I put Boot mode: Same as before.
And it always restart off.
It appens with SonOff BASIC and Shelly 1.

**Steps to reproduce**
Build with default_envs = itead-sonoff-basic or default_envs = allterco-shelly1

**Expected behavior**
If I put same as before it should be same as before.

**Tools used**
Visual Code with Platform IO core 4.0.0

[image]

[image]
[image]

rmcbc, 2019-07-13:
ITEAD_SONOFF_BASIC
==================

    // Info
    #define MANUFACTURER        "ITEAD"
    #define DEVICE              "SONOFF_BASIC"

    // Buttons
    #define BUTTON1_PIN         0
    #define BUTTON1_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH | BUTTON_SET_PULLUP
    #define BUTTON1_RELAY       1
    #define BUTTON1_PRESS       BUTTON_MODE_TOGGLE
    #define BUTTON1_CLICK       BUTTON_MODE_NONE
    #define BUTTON1_DBLCLICK    BUTTON_MODE_NONE
    #define BUTTON1_LNGCLICK    BUTTON_MODE_NONE
    #define BUTTON1_LNGLNGCLICK BUTTON_MODE_NONE   

	#define BUTTON2_PIN         3
	#define BUTTON2_MODE        BUTTON_PUSHBUTTON | BUTTON_DEFAULT_HIGH | BUTTON_SET_PULLUP
	#define BUTTON2_RELAY       1   
	#define BUTTON2_PRESS       BUTTON_MODE_TOGGLE
	#define BUTTON2_CLICK       BUTTON_MODE_NONE
	#define BUTTON2_DBLCLICK    BUTTON_MODE_NONE
	#define BUTTON2_LNGCLICK    BUTTON_MODE_NONE
	#define BUTTON2_LNGLNGCLICK BUTTON_MODE_NONE

    // Relays
    #define RELAY1_PIN          12
    #define RELAY1_TYPE         RELAY_TYPE_NORMAL

    // LEDs
    #define LED1_PIN            13
    #define LED1_PIN_INVERSE    1

    // Non-core modules
    #define ALEXA_SUPPORT               1
    #define DOMOTICZ_SUPPORT            1
    #define HOMEASSISTANT_SUPPORT       1
    #define I2C_SUPPORT                 0
    #define SCHEDULER_SUPPORT           1
    #define SENSOR_SUPPORT              0
    #define THINGSPEAK_SUPPORT          0
	
ALLTERCO_SHELLY1
================

	// Info
	#define MANUFACTURER        "ALLTERCO"
	#define DEVICE              "SHELLY1"

	// Buttons
	#define BUTTON1_PIN         5
	#define BUTTON1_MODE        BUTTON_PUSHBUTTON // | BUTTON_DEFAULT_HIGH | BUTTON_SET_PULLUP
	#define BUTTON1_RELAY       1

	#define BUTTON1_PRESS       BUTTON_MODE_TOGGLE
	#define BUTTON1_CLICK       BUTTON_MODE_NONE
	#define BUTTON1_DBLCLICK    BUTTON_MODE_NONE
	#define BUTTON1_LNGCLICK    BUTTON_MODE_NONE
	#define BUTTON1_LNGLNGCLICK BUTTON_MODE_NONE

	// Relays
	#define RELAY1_PIN          4
	#define RELAY1_TYPE         RELAY_TYPE_NORMAL

	// Non-core modules
	#define ALEXA_SUPPORT               1
	#define DOMOTICZ_SUPPORT            1
	#define HOMEASSISTANT_SUPPORT       1
	#define I2C_SUPPORT                 0
	#define SCHEDULER_SUPPORT           1
	#define SENSOR_SUPPORT              0
	#define THINGSPEAK_SUPPORT          0

rmcbc, 2019-07-13:
Debug when I turn the relay on:

[image]

After reboot:
[image]
[image]

mcspr, 2019-07-13:
Between reboots, what does the `get relayBoot0`, `rtcmem.dump` show and `info` part say about the last boot status?

mcspr, 2019-07-13:
...and can you please try with DEBUG_SERIAL_SUPPORT 0 in defines for Basic?

rmcbc, 2019-07-14:
@mcspr one thing I can say the modules that doesn't work ok with "Same as Before", became ok if I put "Toggle Before".
I'm going to DEBUG and them show the results.

mcspr, 2019-07-14:
Thanks! If you can, can you try recording boot log with one of the units? We can then see what boot status it uses or if anything else breaks it. Only-3v3 powered, ofc.

rmcbc, 2019-07-15:
@mcspr I can't reproduce the problem with the module that I describe this problem, because I erase the flash and write the new one with debug active, and now it's working ok.
There is another with the same problem I'm going to try to get it out and upgrade firmware using webpage.
Meanwhile I send you the current log file.
[log.txt](https://github.com/xoseperez/espurna/files/3393633/log.txt)

rmcbc, 2019-07-15:
Sorry, it was my problem, how could I miss it. The press button was always pressed and when we release it when we pressed. So it was inverted.
Going to close this one.

closed 2019-07-15 by rmcbc